Anime4000/RTL960x #2: DLINK DPN-100 Ver:C1 use RTL9601C1

ptitemimidu86, 2021-03-28:
<p>A D-LINK DPN-100 also has an RTL9601C1, I put the pictures above and below the SFP module board for you.</p>
<p>I wanted to try to flash with firmware V2801F but squashfs-tools (extract / repack rootfs) and&nbsp;<span style="font-size: 0.9em;">qemu-user-static (run MIPS VM) does not exist on the DLINK DPN-100 shell.</span></p>
<p>Is it possible to serial flash or the firmware is PCB dependent ?</p>
<p>For info the DLINK DPN-100 does not have webui, everything is in telnet ip 192.168.100.1</p>
<p>[image]</p>
<p>[image]</p>

Anime4000, 2021-03-29:
I afraid when you flash it, GPIO Laser not working:
```
V2801F Firmware to TWCGPON657 SFP = Works
TWCGPON657 Firmware to V2801F SFP = Not Working
```

I dont have `vm.img` file for V2801F firmware,
But you can try, buy V2801F and dump the flash and flash back to your D-LINK DPN-100

Or, save money way...

Flash with TWCGPON657 B13 Firmware `vm.img` to give you WebGUI, then from there you can flash V2801F firmware and fix `VS_AUTH_KEY` after that.

[Download TWCGPON657 Flash Image Binary](https://github.com/Anime4000/Hacking_RTL9601C1/files/6219530/vm.zip)

Don't forget to dump your D-LINK DPN-100 first just in-case

tarou79, 2021-03-29:
<p>Thank you for your comeback.<br />Indeed I had not thought about the laser calibration, I am going to dump the D-LINK DPN-100 and see what happens next.</p>
<p>While digging through my SFP ONTs, I found one that might be able to accept your firmware as it has an interface that looks like the TWCGPON657 (closer to Hisense LTA3415-SCA + though).</p>
<p>Screen :</p>
<p>[image]</p>
<p>It also has "flash set" commands from telnet. The Realtek chipset is an RTL9601B however and the OMCI_FAKE_OK variable does not exist.</p>
<p>Pictures :</p>
<p>[image]</p>
<p>[image]</p>
<pre id="tw-target-text" class="tw-data-text tw-text-large XcVN5d tw-ta" dir="ltr" data-placeholder="Traduction"><span lang="en">I stop polluting your github now.</span></pre>

Anime4000, 2021-03-30:
# TWCGPON657_B13
The reasons that you may need to downgrade firmware `TWCGPON657_B13` because older firmware `<= B13` can accept `V2801F_V1.9.0-201104.tar` firmware from [WebGUI](http://192.168.1.1/upgrade.asp).

I have tried using `B15 >=`, can't accept any version of V2801F due to mismatch `kernel` mtd

I just had luck with `B13`

# D-LINK DPN-100
About D-LINK DPN-100, the partition way different than standard RTL9601C1 SDK,

Yes, you need to dump and flash `B13` binary file, test if D-Link stick can boot with `TWCGPON657` firmware

You may try first, see Laser got working, and have RX reading.

Then use WebGUI to flash `V2801F_V1.9.0-201104.tar`

# RTL9601B
Maybe same as RTL9601C1, since it's under RTL960x SoC Family,

I think, since both are MIPS CPU, it can boot, but Laser driver are not?

I not sure about this

tarou79, 2021-03-30:
Ok everything is clear to me regarding the DPN-100. Whenever I have time I try this.

Regarding the other SFP GPON module with the RTL9601B (ref: afm0002tim) did not refuse the firmware update V2801F_V1.9.0-201104.tar from WebGUI, but once finished it did not restart, I think is brick now.

It was interesting to take this test on SFP because it's costs only 10/15 € on Ebay. Too bad it doesn't work ;)

Anime4000, 2021-03-30:
I see, it seem have different MIPS arch and driver between `RTL9601B` and `RTL9601C1` SoC.

`B13_20200507.tar` is not binary file like `*.bin; *.img`

You cannot simply flash `*.tar` into NAND memory,

`*.tar` firmware use on WebGUI
`*.img` image file for full NAND flash

What we know now, `RTL9601C1` firmware are not interchangeable between `RTL9601B`

tarou79, 2021-03-30:
<p>Yes, well summarized.</p>
<p>Screen of a kernel panic with firmware V2801F_V1.9.0-191015.tar updated from the WebGUI.</p>
<p>[image]</p>
<p>And restore process :</p>
<p>[image]</p>
<p>In any case, thank you for your answers and your very inspiring Github work :)</p>
<p>I would have to find an LTE3415 (Hisense) firmware now.</p>

frezbo, 2021-04-14:
I have flashed the `V2801F` on an Ubiquiti Ufiber instant module. Same RTL9601C1 chip. You'd need to run `flash_eraseall /dev/mtd3` or whatever the partition that contains the `config` as shown in `cat /proc/mtd`. If this is not done, the laser info is reported incorrect. The `/var/config` partition needs to be unmounted first. There might be some processes accessing the filesystem, kill them and unmount, then erase that partition and reboot. Laser info is then reported properly

obenturk, 2021-04-15:
Hi,

I have UF Instant and I am trying to do the same.

How do you know that laser info is reported incorrectly?

You can also find ufinstant files under my github.
https://github.com/obenturk/ufinstant

frezbo, 2021-04-15:
@obenturk inside `diag` `pon get transceiver <setting>` seems to report everything. But I can't get to any state other than *O1* with the V2801F firmware. The stock firmware goes upto *O3* (it would not get to *O5* since the ubiquiti `omci_app` seems to have the mac id hardcoded or something) As you said the laser info could be bogus. Let's keep updating this issue with future findings and see if we can get the UFiber module working.

obenturk, 2021-04-16:
@frezbo yes I also tried TWCGPON657 and V2801F with deleting /dev/mtd3 but no success. O1 only. and original firmware is going up to O5 but due to negotiation problem dropping to O1 again.

I will re install original firmware to check if lazer will work again or not.

May be lazer settings was deleted hope not.

Additionally I could not receive current firmware from Ubiquiti even I ask it from their support.

frezbo, 2021-04-16:
@obenturk interesting. I don't think ubiquiti would provide the firmware. I switch back to stock by setting `sw_commit` and `sw_active` to `0`. The V2801F is always flashed to the `1` partition. 

new update: at times the laser info is reported correctly in the UI, and after some reboots it starts reporting garbage data. Not sure why.

How did you get to `O5` in stock? I meant setting the correct mac id/serial number?

obenturk, 2021-04-16:
TWCGPON657 Firmware
[image]

`RTK.0> pon get transceiver
bias-current                                     - bias-current
part-number                                      - part-number
rx-power                                         - rx-power
temperature                                      - temperature
tx-power                                         - tx-power
vendor-name                                      - vendor-name
voltage                                          - voltage

RTK.0>  pon get transceiver temperature
Error (0x10): The chip is not found
RTK.0> pon get transceiver part-number
Error (0x10): The chip is not found
RTK.0> pon get transceiver voltage
Error (0x10): The chip is not found
RTK.0> pon get transceiver rx-power
Error (0x10): The chip is not found
`

obenturk, 2021-04-16:
@frezbo I have Ericsson T63G device and I think it uses different omci protocol due to that reason It can connect O5 but does not allow internet to pass.

I re upload original unifi firmware [https://github.com/obenturk/ufinstant/blob/main/flash/ufinstant/bin/firmware.tar](url) but this time I can not telnet with ubnt:ubnt or admin:stdONU101 or admin:system

I check the original flash all command output and realized following ;

`  <Value Name="SUSER_NAME" Value="ubnt"/>
  <Value Name="SUSER_PASSWORD" Value="$1$TW$XXieY8EAV1LpRn5izFeLo."/>`

frezbo, 2021-04-16:
@obenturk I did the same thing and got logged out. Anytime before switching firmware `/dev/mtd3` needs to be erased. The only way to get back is solder wires to the board (The serial ports are exposed and have pads). And when booting you can get into `u-boot` prompt. There we can set it to boot the existing custom firmware, boot into it and erase `mtd3` and go back to stock. I'll be little busy over the weekend, but let me know if you need any help. I'll try to help. This is a brief summary. If you need more details I can help after this weekend.

obenturk, 2021-04-17:
@frezbo Thanks for the info. I was just thinking to do this. Is there any pinout of uart pins to sfp pin side? I would like to have always connected uart pins but if the stick in sfp socket I am not able to route wirings.

frezbo, 2021-04-17:
@obenturk The UART pins are not exposed on the SFP pin side unfortunately. Ypu'd have to di-assemble and solder wires

obenturk, 2021-04-17:
@frezbo returned back to original firmware

transceiver values are ok now

`RTK.0> pon get transceiver bias-current
Bias Current: 10.936000 mA
RTK.0> pon get transceiver part-number
Part Number: RTL8290
RTK.0> pon get transceiver rx-power
Rx Power: -24.559320  dBm
RTK.0> pon get transceiver temperature
Temperature: 85.000000 C
RTK.0> pon get transceiver tx-power
Tx Power: -inf  dBm
RTK.0> pon get transceiver vendor-name
Vendor Name: REALTEK
RTK.0> pon get transceiver voltage
Voltage: 3.019400 V`

additionally I can ask for state and onuid and they are ok !!!!! but due to different mac and serial# onuid is 65 instead of 5.

`# omcicli get onuid
ONU ID: 65
# omcicli get state
ONU state: 5
`

for correct transceiver configuration I figureout following.

thereis europa.data at hw section and configures rtl8290

`RTL8290 driver version [0.2.21]

   Parameter: I2C_PORT=1 INTR_PIN=0 TXDIS_PIN=13 TXPWR_PIN=15 PON_MODE=1

Open file: europa.data.
Loaded data from europa.data
Init Europa module.
config TX_DIS pin...
Config GPIO 13 to OUTPUT mode, value = 0
config TxPwr pin...
Config GPIO 15 to OUTPUT mode, value = 0
Init I2C port ...
rtl8290_i2c_init  enable = 1 width=0  clock = 1008
disable DW ...
[ init ddmi A0&A2, A4 ]
Update DDMI A0...., flash_update = 1, addr = 0xc00d24a8
  addr = 0  02 04 0B FF FF FF FF FF    FF FF FF 03 0C 00 14 C8
 addr = 16  00 00 00 00 52 45 41 4C    54 45 4B 20 20 20 20 20
 addr = 32  20 20 20 20 00 00 00 00    52 54 4C 38 32 39 30 20
 addr = 48  20 20 20 20 20 20 20 20    30 30 30 31 05 1E 00 FF
 addr = 64  00 06 00 00 76 65 6E 64    6F 72 70 61 72 74 6E 75
 addr = 80  6D 62 65 72 32 30 31 34    30 31 32 33 68 80 02 FF
 addr = 96  FF FF FF FF FF FF FF FF    FF FF FF FF FF FF FF FF
 addr = 112 FF FF FF FF FF FF FF FF    FF FF FF FF FF FF FF FF
Update DDMI A2...., flash_update = 1, addr = 0xc00d25a8
  addr = 0  7F FF FF FF 7F FF FF FF    8E 94 6D 60 8C A0 75 30
 addr = 16  75 30 05 DC 61 A8 07 D0    00 00 0F 8D 00 0A 0C 5A
 addr = 32  00 0C 00 00 00 00 00 00    00 00 00 00 00 00 00 00
 addr = 48  00 00 00 00 00 00 00 00    00 00 00 00 00 00 3F 80
 addr = 64  00 00 00 00 00 00 01 00    00 00 01 00 00 00 01 00
 addr = 80  00 00 01 00 00 00 00 00    00 FF FF FF FF FF FF FF
 addr = 96  FF FF FF FF 00 00 00 00    FF 00 FF 00 00 00 FF 00
 addr = 112 00 00 00 00 00 00 00 00    00 00 00 00 00 00 00 00
Update A4...., flash_update = 1, length =0x7d, addr = 0xc00d26a9
  addr = 0  72 89 E1 E2 8E 32 9B 90    00 49 9F FF 23 04 78 7F
 addr = 16  FF 00 02 05 00 00 01 F6    CE 90 C0 00 00 38 24 40
 addr = 32  40 00 01 13 BC 1E 33 10    FB 18 E4 80 E0 01 B0 44
 addr = 48  40 AC 03 00 F4 00 20 40    00 7C 20 40 FC 30 10 00
 addr = 64  00 00 00 00 FF FC 12 0A    8B 04 01 33 74 A5 A0 E0
 addr = 80  00 00 00 00 81 30 00 15    00 FF 1F 01 00 FF 00 32
 addr = 96  FF 01 FF FF 05 00 60 80    00 08 10 A0 EC B0 BF FE
 addr = 112 00 00 00 00 FF 00 10 00    00 00 FF A0 C0
Update P0 = 0x6e, P1 = 0xb5, Pavg = 0xa8., flash_update = 1,  addr = 0xc00d29ea
 Read RSSI_V0 from flash...
SDADC RSSI V0 = 1652670...
<6>br0: port 1(eth0) entering forwarding state
[ APC Enable flow ]
## Enable LDX prior OffK...
OffK:     2     7     0     5   -12
 MPD0[0] = 0x7fb72c
 MPD0[1] = 0x7fc40c
 MPD0[2] = 0x7fb8ec
 MPD0[3] = 0x7fb978
 MPD0[4] = 0x7fb8fc
 MPD0[5] = 0x7fb454
 MPD0[6] = 0x7fb160
 MPD0[7] = 0x7fb0fc
 MPD0[8] = 0x7fbf78
 MPD0[9] = 0x7fb75c
 MPD0 = 0x7fb869
[ Tx Enable flow ]
[ Rx Enable flow ]
[ Update APD     ]
[ Enable TX SD   ]
[xml_INFO]: Update cs configuration success.
init user parameters ...
[ Enable RX_SD   ]
[ Enable up      ]
Init Europa Finish.
 DDMI Polling Task Add Success.
 OffKd Task Add Success.
 LoopMon Task Add Success.
europa_intrInit`

rcS file 

`mount -t jffs2 -o ro /dev/mtdblock8 /var/hw
if [[ -f /var/hw/europa.data ]]; then
        if ! cmp /var/config/europa.data /var/hw/europa.data -s; then
                echo "Copy /var/hw/europa.data /var/config/europa.data"
                cp -f /var/hw/europa.data /var/config/europa.data
        fi
elif [[ ! -f  /var/config/europa.data ]]; then
    echo "Warning: File /var/hw/europa.data nor /var/config/europa.data doesn't exist!!!. Generating."
    /bin/europacli open default
fi`

frezbo, 2021-04-17:
@obenturk glad you were able to revert to stock. So if I understand right, if I modify the the V2801F to mount `mtd8` and update the `rcS` script, the laser will work?

obenturk, 2021-04-17:
@frezbo I don't think it will work. Because V2801F does not have /bin/europacli and RTL8290 driver version [0.2.21] with Parameter: I2C_PORT=1 INTR_PIN=0 TXDIS_PIN=13 TXPWR_PIN=15 PON_MODE=1

I am not sure will it work or not that if we extract ufinstant firmware and add following to V2801F firmware

-  /lib/modules/europa_drv.ko
-  /bin/europacli
- ### rc32
- insmod /lib/modules/europa_drv.ko PON_MODE=1 I2C_PORT=1 INTR_PIN=0 TXDIS_PIN=13 TXPWR_PIN=15

This section is configuring lazer I believe

Update DDMI A0...., flash_update = 1, addr = 0xc00d24a8
addr = 0 02 04 0B FF FF FF FF FF FF FF FF 03 0C 00 14 C8
addr = 16 00 00 00 00 52 45 41 4C 54 45 4B 20 20 20 20 20
addr = 32 20 20 20 20 00 00 00 00 52 54 4C 38 32 39 30 20
addr = 48 20 20 20 20 20 20 20 20 30 30 30 31 05 1E 00 FF
addr = 64 00 06 00 00 76 65 6E 64 6F 72 70 61 72 74 6E 75
addr = 80 6D 62 65 72 32 30 31 34 30 31 32 33 68 80 02 FF
addr = 96 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
addr = 112 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
Update DDMI A2...., flash_update = 1, addr = 0xc00d25a8
addr = 0 7F FF FF FF 7F FF FF FF 8E 94 6D 60 8C A0 75 30
addr = 16 75 30 05 DC 61 A8 07 D0 00 00 0F 8D 00 0A 0C 5A
addr = 32 00 0C 00 00 00 00 00 00 00 00 00 00 00 00 00 00
addr = 48 00 00 00 00 00 00 00 00 00 00 00 00 00 00 3F 80
addr = 64 00 00 00 00 00 00 01 00 00 00 01 00 00 00 01 00
addr = 80 00 00 01 00 00 00 00 00 00 FF FF FF FF FF FF FF
addr = 96 FF FF FF FF 00 00 00 00 FF 00 FF 00 00 00 FF 00
addr = 112 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Update A4...., flash_update = 1, length =0x7d, addr = 0xc00d26a9
addr = 0 72 89 E1 E2 8E 32 9B 90 00 49 9F FF 23 04 78 7F
addr = 16 FF 00 02 05 00 00 01 F6 CE 90 C0 00 00 38 24 40
addr = 32 40 00 01 13 BC 1E 33 10 FB 18 E4 80 E0 01 B0 44
addr = 48 40 AC 03 00 F4 00 20 40 00 7C 20 40 FC 30 10 00
addr = 64 00 00 00 00 FF FC 12 0A 8B 04 01 33 74 A5 A0 E0
addr = 80 00 00 00 00 81 30 00 15 00 FF 1F 01 00 FF 00 32
addr = 96 FF 01 FF FF 05 00 60 80 00 08 10 A0 EC B0 BF FE
addr = 112 00 00 00 00 FF 00 10 00 00 00 FF A0 C0
Update P0 = 0x6e, P1 = 0xb5, Pavg = 0xa8., flash_update = 1, addr = 0xc00d29ea
Read RSSI_V0 from flash...
SDADC RSSI V0 = 1652670...

frezbo, 2021-04-17:
yeh, I was thinking of adding the `europa` stuff to the V2801F firmrware, but I wonder if it's worth the additional time and testing. I wished the Ubiquiti ones had an option to change the serial number easily. Maybe copy over the `omci_app` might work

obenturk, 2021-04-17:
I am not an expert for linux programs and kernel. Can we just drag and drop omci_app or europacli maybe busybox for vi implementation to their folders and will they work?

on the other hand how can we start from the scratch by obtaining Toolkit:rsdk-1.5.6-5281-EB-2.6.30-0.9.30.3-131105 ? I searched but I could not find realtek toolkit.

obenturk, 2021-04-18:
@frezbo 

I did followings

###replaced files

/bin/diag
/bin/europacli
/bin/busybox

chown 1013
chmod 775

###added files
/lib/modules/europa_drv.ko

chown 1013
chmod 664

###changed files

####rc2

mkdir /var/hw

mount -t jffs2 -o ro /dev/mtdblock8 /var/hw
if [[ -f /var/hw/europa.data ]]; then
	if ! cmp /var/config/europa.data /var/hw/europa.data -s; then
                echo "Copy /var/hw/europa.data /var/config/europa.data"
                cp -f /var/hw/europa.data /var/config/europa.data
	fi
elif [[ ! -f  /var/config/europa.data ]]; then
    echo "Warning: File /var/hw/europa.data nor /var/config/europa.data doesn't exist!!!. Generating."
    /bin/europacli open default
fi

####rc32

insmod /lib/modules/europa_drv.ko PON_MODE=1 I2C_PORT=1 INTR_PIN=0 TXDIS_PIN=13 TXPWR_PIN=15

frezbo, 2021-04-18:
@obenturk did replacing the files, fixed the laser issue?

obenturk, 2021-04-18:
No I am receiving following error.

----- do_rc [/etc/init.d/rc32] -----
SFP APP
europa_drv: module license 'Proprietary' taints kernel.
Disabling lock debugging due to kernel taint
europa_drv: Unknown symbol rtk_lddMapper_attach
europa_drv: Unknown symbol rtk_lddMapper_deattach
insmod: cannot insert '/lib/modules/europa_drv.ko': unknown symbol in module

:(

frezbo, 2021-04-19:
@obenturk it seems the Ufiber kernel is locked down and they might have done some proprietary stuff. Regarding the Realtek toolkit you might need to be a customer to obtain that or even worse sign an NDA. That's how usually these things are. Maybe it's worth a try to reach out to realtek via email.

obenturk, 2021-04-23:
@frezbo I tried original uImage + V2801F rootfs and omci start to work with new mac, serial + other things but there was an error like WAN not configured I did not try to configure WAN stuff. You can give a try for that.

frezbo, 2021-04-23:
@obenturk that's great news, I'll try to test these out over the weekend or the course of next week. Just wanted to thank you for all the time spared and tinkering you'd been able to do :muscle: :heart:

obenturk, 2021-04-24:
Thanks for the good words but I did not mean that it is fully functional. What I know is, I can connect the OLT side with ONT parameters buuuuuut!! ISP kicked me out. I did not face with this kind of result previously because SFP stick was identified by OLT as different ONU ID due to different SN and MAC.

After I upload ufinstant with original kernel + V2801F rootfs with modded files I believe I triggered an alarm from OLT side and it blocked ISP ONT as well.

ISP gave me different ONT which is Sercomm FG1000E very very small footprint. Now I am afraid I will not connect ufinstant anymore may be for sometime :)

frezbo, 2021-04-27:
@obenturk with the original `uImage` and the modded `rootfs` from*V2801F*. I am able to get upto `O3` state. The orginal firmware also foes upto `O3` state. Even after setting the right auth parameters, serial number, model number and mac id, I'm still stuck at `O3`. Meanwhile I have ordered a `CPGOS03-0490` from Carlitoxxx Pro. I'll wait until it arrives and test again.

obenturk, 2021-04-27:
Then I think you need LOID and password

frezbo, 2021-04-27:
[image]
the ISP provided ONT uses password auth and I have set `GPON_PLOAM_PASSWD`. I'm not sure if LOID also needs to be set

obenturk, 2021-04-28:
Could you please check with top command to see if omciapp fired up with correct elan mac id ?

obenturk, 2021-05-01:
@frezbo When you receive the CPGOS03-0490 Could you please dump whole flash from uboot? May be we can use its kernel and rootfs.

frezbo, 2021-05-02:
@obenturk Sure I will do take a backup before making any changes. It might take a while though for it to arrive and clear the customs.

robkegeenen, 2021-05-27:
For others struggling with the Ubiquiti UFiber Instant (UF-Instant), I thought I'd dump whatever little information I found here.
So here goes, Ubiquiti have indeed locked down changing of the GPON SN, and as determined by others, the hex-digit part of the SN is taken from the last 4 bytes of the MAC address as stored on the `sec` partition, with the characters `UBNT` prepended by `ubnthal.ko`:
> user=Firezone said:
> > UF-instant has very good cooling: thermal contact to the case through the thermally conductive layer on both sides of the board, shorter case, connector to the optics SC/APC (many people won't need adapters). Telnet to 192.168.1.1, login/password ubnt/ubnt. There is ssh (disabled by default). It won't let me in via http. In UF-instant the omci_app is customised so that the SerialNumber is taken from the /proc/ubnthal/sn file which is exposed by the driver ubnthal.ko. The driver generates SN by concatenating the letters UBNT and the last 4 bytes of the MAC address. The MAC address itself is taken from the mtd9 "sec" partition at offset 0xA022. So, to change SN you will have to patch ubnthal.ko and match the MAC address to "sec" partition, or modify omci_app.

Source: [https://forum.mikrotik.com/viewtopic.php?t=116364&start=300#p842585](https://forum.mikrotik.com/viewtopic.php?t=116364&start=300#p842585)
Originally from (Google translated): [http://forum.ru-board.com/topic.cgi?forum=8&topic=80480&start=1089](https://translate.google.com/translate?hl=&sl=ru&tl=en&u=http%3A%2F%2Fforum.ru-board.com%2Ftopic.cgi%3Fforum%3D8%26topic%3D80480%26start%3D1089)

I have been able to patch `ubnthal.ko` by:
* unsquashfs the rootfs (`/dev/mtd13`)
* hexedit `/lib/modules/ubnthal.ko`, at offsets 0x13E0 & 0x1404 (SW version v4.1.0.359-b87.191101.0940) there are two printf format strings `%02x%02x%02x%02x%02x%02x%02x%02x` & `%c%c%c%c%02x%02x%02x%02x`, for `/proc/ubnthal/sn_raw` & `/proc/ubnthal/sn` respectively.
  These can be patched by entering the desired SN in both formats and padding the rest with 0x00.
  [image]
  [image]
* `mksquashfs squashfs-root rootfs -b 131072 -comp lzma` and write it back to `/dev/mtd13` (don't forget to `flash_eraseall` first)

With this patch I have been able to get both proc files and `diag` => `gpon get serialnumber` to present the correct SN.
However, even with this patch I have not been able to get a link to my provider, the operational state keeps dropping back from O5 to O1/O3, indicating (AFAICT) there is still something wrong with the authentication to the OLT.

An alternative patch might be to swap `/proc/ubnthal/sn_raw` for a path to a custom file with the desired SN in `/bin/omci_app` (offset 0x2B2D4).
I did not investigate this approach before deciding to try and flash one of the V2801F firmwares on the stick, which has not gotten me far yet as I have yet to find a way to actually telnet to it.
Seems that with the V2801F firmware the stick is not able to establish network connection over the SGMII Ethernet interface, even though the stick seems to boot fine when watching the UART output...
Investigation ongoing, although not with high priority :)

Anyone wanting to communicate with the UART on this stick, see the pinout below and connect with 115200 8N1 (e.g. `screen <dev> 115200`).
I have succesfully used a [TUMPA](https://www.tiaowiki.com/w/TIAO_USB_Multi_Protocol_Adapter_User%27s_Manual) as USB-UART adapter, powering the SFP from it as well.
[image]
Initial letters are printed on the other side of the PCB as well.

robkegeenen, 2021-05-28:
@obenturk I just compared my flash dumps (https://github.com/Anime4000/RTL9601C1/pull/6) to yours;
* `boot`
* `kernel1`
* `rootfs1`
* `kernel2`
* `rootfs2`

are all identical, so I'm assuming we're running the same SW version.

Below are the differences between my `env2` and your `env2` (mainly MAC addrs) and between my `hw` and yours (few differences, but not ones I can easily explain).
However, your `env` and `env2` differ by quite a bit, while for me they're identical.
Lastly `sec` and `config` have a lot of differences, so I'll not post those here.

`diff <(hexdump -C RTL9601C1/UF-Instant/Firmware/dev_mtd2) <(hexdump -C ufinstant/flash/ufinstant/bin/env2.bin)`:
```diff
1c1
< 00000000  35 4c b7 3f 00 62 30 3d  73 65 74 65 6e 76 20 62  |5L.?.b0=setenv b|
---
> 00000000  57 f2 30 6f 00 62 30 3d  73 65 74 65 6e 76 20 62  |W.0o.b0=setenv b|
53,54c53,54
< 00000340  72 3d 31 38 3a 45 38 3a  32 39 3a 46 35 3a 43 39  |r=18:E8:29:F5:C9|
< 00000350  3a 45 44 00 66 69 6c 65  61 64 64 72 3d 38 30 30  |:ED.fileaddr=800|
---
> 00000340  72 3d 31 38 3a 45 38 3a  32 39 3a 46 35 3a 44 44  |r=18:E8:29:F5:DD|
> 00000350  3a 44 31 00 66 69 6c 65  61 64 64 72 3d 38 30 30  |:D1.fileaddr=800|
124,126c124,126
< 000007b0  6d 69 74 3d 30 00 73 77  5f 63 72 63 30 3d 38 65  |mit=0.sw_crc0=8e|
< 000007c0  32 33 65 66 38 63 00 73  77 5f 63 72 63 31 3d 38  |23ef8c.sw_crc1=8|
< 000007d0  65 32 33 65 66 38 63 00  73 77 5f 74 72 79 61 63  |e23ef8c.sw_tryac|
---
> 000007b0  6d 69 74 3d 30 00 73 77  5f 63 72 63 30 3d 36 37  |mit=0.sw_crc0=67|
> 000007c0  62 35 36 37 66 32 00 73  77 5f 63 72 63 31 3d 36  |b567f2.sw_crc1=6|
> 000007d0  37 62 35 36 37 66 32 00  73 77 5f 74 72 79 61 63  |7b567f2.sw_tryac|
```
`diff <(hexdump -C RTL9601C1/UF-Instant/Firmware/dev_mtd8) <(hexdump -C ufinstant/flash/ufinstant/bin/hw.bin)`:
```diff
6,8c6,8
< 00001020  00 00 81 a4 00 00 00 00  00 00 00 00 00 00 00 b0  |................|
< 00001030  00 00 00 b0 00 00 00 b0  00 00 00 00 00 00 00 00  |................|
< 00001040  00 00 00 00 00 00 00 00  00 00 00 00 40 20 40 14  |............@ @.|
---
> 00001020  00 00 81 a4 00 00 00 00  00 00 00 00 00 00 00 ad  |................|
> 00001030  00 00 00 ad 00 00 00 ad  00 00 00 00 00 00 00 00  |................|
> 00001040  00 00 00 00 00 00 00 00  00 00 00 00 8c d0 88 e8  |................|
10,11c10,11
< 00001060  00 00 00 01 00 00 00 02  00 00 00 b0 0b 08 00 00  |................|
< 00001070  f8 9b 34 9f 3b fb 9c 2c  65 75 72 6f 70 61 2e 64  |..4.;..,europa.d|
---
> 00001060  00 00 00 01 00 00 00 02  00 00 00 ad 0b 08 00 00  |................|
> 00001070  60 eb 67 ac 3b fb 9c 2c  65 75 72 6f 70 61 2e 64  |`.g.;..,europa.d|
14c14
< 000010a0  00 00 06 40 00 00 00 b0  00 00 00 b0 00 00 00 b0  |...@............|
---
> 000010a0  00 00 06 40 00 00 00 ad  00 00 00 ad 00 00 00 ad  |...@............|
16c16
< 000010c0  85 c7 52 f3 35 5c bf be  78 5e 62 62 e1 fe 0f 05  |..R.5\..x^bb....|
---
> 000010c0  da d9 b9 d0 f9 ac 77 42  78 5e 62 62 e1 fe 0f 05  |......wBx^bb....|
38,40c38,40
< 00001220  a2 16 e0 c1 f0 7f af f5  14 06 06 93 a7 ff ff 7f  |................|
< 00001230  c9 67 90 d4 dc 8a 45 1d  a3 80 03 dd 22 95 c6 16  |.g....E....."...|
< 00001240  01 04 10 b2 f1 00 01 06  00 42 33 70 51 ff ff ff  |.........B3pQ...|
---
> 00001220  a2 16 e0 c1 f0 ff 64 70  22 03 83 b6 f1 ff ff 5f  |......dp"......_|
> 00001230  4e 32 48 9a ef c3 a2 8e  51 c0 81 6e 91 4a 63 8b  |N2H.....Q..n.Jc.|
> 00001240  00 02 08 d9 78 80 00 03  00 eb 82 6f f8 ff ff ff  |....x......o....|
43,45c43,45
< 00001270  00 00 00 b0 00 00 00 ce  00 00 00 ce 00 00 00 00  |................|
< 00001280  00 00 00 00 00 00 00 00  00 00 c0 00 00 00 00 00  |................|
< 00001290  5e 16 f0 c9 19 85 e0 02  00 00 01 d2 a4 98 25 0e  |^.............%.|
---
> 00001270  00 00 00 ad 00 00 00 d2  00 00 00 d2 00 00 00 00  |................|
> 00001280  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
> 00001290  41 60 a7 a6 19 85 e0 02  00 00 01 d2 a4 98 25 0e  |A`............%.|
47,49c47,49
< 000012b0  00 00 06 40 00 00 00 ce  00 00 00 ce 00 00 00 ce  |...@............|
< 000012c0  00 00 00 00 00 00 01 8e  00 00 06 40 06 00 c0 00  |...........@....|
< 000012d0  64 09 f6 bc 05 75 3c b0  78 5e 62 62 e1 fe 0f 05  |d....u<.x^bb....|
---
> 000012b0  00 00 06 40 00 00 00 d2  00 00 00 d2 00 00 00 d2  |...@............|
> 000012c0  00 00 00 00 00 00 01 8e  00 00 06 40 06 00 00 00  |...........@....|
> 000012d0  c7 d7 4c 3e bf 88 fb d1  78 5e 62 62 e1 fe 0f 05  |..L>....x^bb....|
71,74c71,74
< 00001430  8b 41 86 d8 a2 c5 c8 cf  93 e1 ff 5e eb 29 0c 0c  |.A.........^.)..|
< 00001440  26 4f ff ff ff 92 cf 20  a9 b9 15 a6 8e 75 2a 5c  |&O..... .....u*\|
< 00001450  07 a3 80 03 dd 22 95 c6  16 01 04 10 b2 f1 00 01  |....."..........|
< 00001460  06 00 5a f0 6f 48 ff ff  ff ff ff ff ff ff ff ff  |..Z.oH..........|
---
> 00001430  8b 41 86 d8 a2 25 6f eb  0a 86 ff 27 83 13 19 18  |.A...%o....'....|
> 00001440  b4 8d ff ff ff 72 92 41  d2 7c 1f 4c 1d d3 47 b8  |.....r.A.|.L..G.|
> 00001450  0e 46 01 07 ba 45 2a 8d  2d 02 08 20 64 e3 01 02  |.F...E*.-.. d...|
> 00001460  0c 00 51 6a 70 4a ff ff  ff ff ff ff ff ff ff ff  |..QjpJ..........|
```

obenturk, 2021-05-30:
@robkegeenen 
I think serial and mac address for ubnthal.ko is in sec partition hw partiton has europa.data and it is required for laser configuration. If I did not use europa with V2801F firmware then there is no laser info + no o5 connection.

you already said that boot kernel and rootfs partitions are all same so I believe that it should be some basic flashing change to be able to fit in production cycle time.

I would like to try changing sec, hw, cfgfs, env and env2 one by one with yours to learn which partition changes SN and MAC then we can work on specificly that partition.

I don't know  can I find a suitable time for this. I appreciate if some one also can do that.

pmgcosta, 2021-06-01:
Hello i have an udm-pro and i want to use an SFP (AFM0002TIM) on MEO - Portugal FTTH Internet Provider

I'm have replase original router of my provider with an ONT HG8012h, i only have to change the serialnumber and password to " "
The issue is that on my sfp AFM0002TIM the serialnumber have only 12digits and my original ont have 16.

I have made that to change serial and password

ifconfig eth9:2 192.168.2.2 netmask 255.255.255.252 up
telnet 192.168.2.1

ssh admin@192.168.2.1 "cat /var/config/lastgood.xml" > lastgood.xml
ssh admin@192.168.2.1 "cat /var/config/lastgood_hs.xml" > lastgood_hs.xml

cat lastgood.xml | ssh admin@192.168.2.1 -T "cat > /var/config/lastgood.xml"
cat lastgood_hs.xml | ssh admin@192.168.2.1 -T "cat > /var/config/lastgood_hs.xml"

lastgood
<Value Name="GPON_PLOAM_PASSWD" Value="1234567890"/>
<Value Name="GPON_PLOAM_PASSWD" Value=" "/>

lastgood_hs
<Value Name="GPON_SN" Value="TNBB7BAB2A78"/>
<Value Name="GPON_SN" Value="5054494E91184A08"/>

pmgcosta, 2021-06-02:
The issue is:

# omcicli set sn PTIN, **7**1194A08; omcicli get sn;
SerialNumber: PTIN**7**1194a08

# omcicli set sn PTIN, **9**1194A08; omcicli get sn;
SerialNumber: PTIN**ffffff9**1194a08

Is that normal how can i fix that ?

pmgcosta, 2021-06-10:
I have one UF-Instant i can try to flash something, since my ISP don't kick me when i try ONT, SFP with different settings.

But i can't understand how disassemble the module without break, can some one put a picture ?

Anime4000, 2021-06-11:
@pmgcosta if it's works, then no need to change anything, your ISP kind enough allow universal ONU on their OLT

Den795, 2021-07-21:
Hello!
Please tell me how to roll back the firmware to the original one, there is no such possibility in the GUI, but it is very necessary, because after the flash V2801F_V1.9.0-201104.tar ONU STATE O1 :(

zambelly, 2021-08-18:
Has anyone done more work on Ubiquiti's UF-Instant?
@robkegeenen @obenturk @frezbo
I bought this ONU stick, hoping it could replace my ISP provided ONU, but I see I chose wrong :frowning_face: 

I wrote a script that takes the current firmware for it from Ubiquiti, and automatically patches ubnthal.ko with my serial number and MAC address, but after doing that and setting all the variables in the configuration using the `flash` utility, I still only get to `O3`

@obenturk I see you were looking for the original firmware earlier for this stick.
In normal operation, when you use it with Ubiquiti's OLT, it gets updated by the OLT.
All you have to do is download the OLT firmware (https://www.ui.com/download/ufiber)
From the tarball extract `squashfs.tmp` and in this SquashFS image you can find the ONU firmware under `usr/share/onufw`

Here is my script in case anyone wants it: https://gist.github.com/zambelly/bf9d27cc872b807ef5e6faa408501748
It runs ok on my machine running Manjaro Linux, I can't guarantee it's portable

robkegeenen, 2021-09-08:
No breakthroughs yet, but some small updates:

@obenturk, I have confirmed that the MAC is stored in the "sec" partition (at offset 0xA022, just as Firezone said), and the SN is generated from that by `ubnthal.ko`. However, patching `ubnthal.ko` is still required to change the `UBNT` that is put in front of the last part of the MAC to generate the SN. This `UBNT` is the same for all UF-Instants, and so does not need to be on the "sec" partition to facilitate changing at production time.

I noted in my last post that I was able to power the SFP from the TUMPA, and that was indeed the case. However, lately I have been observing some weird behavior doing this (e.g. sudden power-cycling of the SFP). Therefore, I now recommend leaving the 3v3 from the USB-TTL adapter unconnected and powering the SFP from an actual SFP slot or powering it with a dedicated 3v3 power supply. Especially when programming the flash (e.g. via U-Boot). Anyway, just wanted to note this as YMMV.

Lastly, I have found a marginally useful manual on the Realtek `diag` utility, see: https://github.com/Anime4000/RTL9601C1/pull/11 (source: https://usermanual.wiki/Pdf/RTKPONDiagShellUserGuideV001.291982779/html).

frezbo, 2021-09-09:
@robkegeenen have you tried https://github.com/Unifi-Tools/UFiber.Configurator

robkegeenen, 2021-09-09:
@frezbo, I have now. Unfortunately I get a stack trace, both on FW v4.1.0.359 and on FW v4.3.1.3909:
```
Unhandled exception: System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation.
 ---> System.InvalidOperationException: Invalid data, CRC doesn't match
   at UFiber.Configurator.NVRAM..ctor(ReadOnlySpan`1 data, UInt32 offset, UInt32 length)
   at <Program>$.<>c.<<Main>$>b__0_2(String host, String user, String pw, Int32 port, Boolean dryRun, String slid, String vendor, String serial, String mac, String fwToRestore)
   --- End of inner exception stack trace ---
   at System.RuntimeMethodHandle.InvokeMethod(Object target, Object[] arguments, Signature sig, Boolean constructor, Boolean wrapExceptions)
   at System.Reflection.RuntimeMethodInfo.Invoke(Object obj, BindingFlags invokeAttr, Binder binder, Object[] parameters, CultureInfo culture)
   at System.Delegate.DynamicInvokeImpl(Object[] args)
   at System.Delegate.DynamicInvoke(Object[] args)
   at System.CommandLine.Invocation.ModelBindingCommandHandler.InvokeAsync(InvocationContext context)
   at System.CommandLine.Invocation.InvocationPipeline.<>c__DisplayClass4_0.<<BuildInvocationChain>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass23_0.<<UseParseErrorReporting>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass16_0.<<UseHelp>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass27_0.<<UseVersionOption>b__1>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass25_0.<<UseTypoCorrections>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c.<<UseSuggestDirective>b__24_0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass22_0.<<UseParseDirective>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass11_0.<<UseDebugDirective>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c.<<RegisterWithDotnetSuggest>b__10_0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass14_0.<<UseExceptionHandler>b__0>d.MoveNext()
```

I don't have a Windows machine available and don't have any .net stuff setup on my Linux boxes, so debugging the app is a bit difficult for me now. However, I have attached the bin dump as generated by the tool in the `dumps/` directory, to hopefully help you debug. :grin: (Had to rename it to `.txt`, because GitHub...) I have checked, it is indeed equal to when I manually extract `/dev/mtd3` (the `config` partition).
[fw-09092021-044802.bin.txt](https://github.com/Anime4000/RTL9601C1/files/7138419/fw-09092021-044802.bin.txt)

mauimauer, 2021-10-09:
@zambelly I was able to use your script. However I also had to patch omci_app as it was hardcoded to send UBNT as VID (Vendor Id):

`sed -i "s/UBNT/HWTC/" squashfs-root/bin/omci_app`

I get up to O5, but am now facing an issue with omci_app simply dying

manias21, 2021-10-19:
20211016_094611_MX25L12835F.bin
https://mega.nz/file/qCYF3IrC#ZvGBFXucYpIGUJJ1GfT4Ni7vXDD0frZRHCHGf5fT_IA

I am the same, a full backup, if it can help, I collaborate with the tests.
I have the materials to read and write

manias21, 2021-10-30:
firmware latinq alcatel.
https://mega.nz/file/iG5m3a7S#jvvFgzygTVLx1NuBVsJlq2h9UfUwV8zzVIPCV1WeD2s

Danno88-88, 2021-11-13:
any possibile guide to flash any other firmware onto the uf-instant?

manias21, 2021-11-14:
yes,..
Yes, I changed it, but I don't remember which of all, because I was testing them one by one, but all the ones I have tested shown here, only one admits 20 hex digits to move but does not link with the alcatel olt, however with the huawei yes.

robkegeenen, 2021-12-14:
It took roughly 30 messages spread over multiple support tickets and roughly 3 and a half months, but I finally did get some sort of GPL source code package for the UF-Instant from Ubiquiti. However, it only contains the source code for these components:
* kernel
* busybox
* bridge-utils
* iptables
* ebtables
* iproute2
* route
* libnet
* memtester
* mtd-utils
* dropbear

Absent are the juicy bits, such as omci_app, diag and ubnthal. And when I asked about those specifically, the response was that they are under other licenses such as MIT or are proprietary packages, and that they won't be providing the source code to those. Therefore, the package still doesn't provide a way to build a full firmware image (for debugging purposes) or to study the Ubiquiti modifications to omci_app.

The package contains the expected versions of the GPL projects in the v4.1.0 version of the firmware. The fact that I requested the source code to two different versions of the firmware probably got lost somewhere in the message stream. On the other hand, I suspect few, if any, of the GPL projects will have bumped to a different version in the newer firmware version.

I have attached the provided package for archival and for inspection by curious others. :slightly_smiling_face: 
[UF-Instant gpl.part0.tar.gz](https://github.com/Anime4000/RTL9601C1/files/7715080/UF-Instant.gpl.part0.tar.gz)
[UF-Instant gpl.part1.tar.gz](https://github.com/Anime4000/RTL9601C1/files/7715081/UF-Instant.gpl.part1.tar.gz)
[UF-Instant gpl.part2.tar.gz](https://github.com/Anime4000/RTL9601C1/files/7715082/UF-Instant.gpl.part2.tar.gz)
[UF-Instant gpl.part3.tar.gz](https://github.com/Anime4000/RTL9601C1/files/7715083/UF-Instant.gpl.part3.tar.gz)

JamonesAlVapor, 2021-12-23:
Hola @manias21 - were you able to get any firmware to work with an Alcatel ONT? (I guess Movistar). Which firmware version did you found that let you set the entire hex PLOAM?

Some useful things I'm finding out, on those modules are two different kernels (k0, k1) and rootfs (r0, r1) that work in an A/B fashion.

You can look at the uboot settings (nvram) with `nv getenv` and `nv printenv`, the default command that runs is `bootcmd`:

```
boot_by_commit=if itest.s ${sw_commit} == 0;then run set_act0;run b0;else run set_act1;run b1;fi
boot_by_tryactive=if itest.s ${sw_tryactive} == 0;then setenv sw_tryactive 2;setenv sw_active 0;saveenv;run en_wdt;run b0;else setenv sw_tryactive 2;setenv sw_active 1;saveenv;run en_wdt;run b1;fi
bootargs_base=console=ttyS0,115200
bootcmd=if itest.s ${sw_tryactive} == 2; then run boot_by_commit;else run boot_by_tryactive;fi
bootdelay=5
...
sw_active=1
sw_commit=1
sw_tryactive=2
```

So:
- changing `sw_commit` changes the default kernel+rootfs to boot (0 or 1)
- useful for testing: changing `sw_tryactive` boots the chosen option **once** and resets the parameter for the next reboot. This allows you to test firmwares on the other slot without fear of bricking as the default slot will be reset on the following reboot. So you can, if you are in `sw_commit=1`:
  - flash the firmware to the other slot (0: uimage to k0, rootfs to r0)
  - test if it works by setting `sw_tryactive=0`
  - if it works, make it permanent by setting `sw_commit=0`
  - if it won't boot, just reboot and 1 will be booted again

You can firmware update by running:

```
#--- on telnet/ssh on the sfp
flash_eraseall /dev/mtdX

#--- from your computer
cat uimage | ssh -oHostKeyAlgorithms=+ssh-rsa admin@192.168.2.1 "cat >/dev/mtdX"
```
where mtdX is the correct partition for the slot you want to use (k0/r0 or k1/l1, in my case mtd4/mtd5 for slot 0, mtd6/mtd7 slot 1)

It seems you may also be able to gain uboot netconsole and tftp to reflash/access the boot console without a serial interface, I haven't been able to make it work yet though.

JamonesAlVapor, 2021-12-24:
Update on my last post.... if you do this be careful with switching firmwares between different versions/brands as they share the `config` partition and you may get locked out due to the default firmware trying to use an incompatible config.. as I just did. 

Oh well, I'll now have to solder a serial console one of these days...

panchititin, 2021-12-24:
I have checked about 16 firmware versions, in 30% it supports ploam 20 hex characters, including Carlitos, and in the rest only 10 in hex, but some admit to assign it in diag the 20 characters, but none of the firmwares gives positive result in Alcatel, for this reason I have requested support with an external company, for work reasons I have extended the tests to February, the manufacturer through a negotiation is motivated to process a solution, when possible I indicate the data and provide the progress.
But at the moment no one known has a solution against the issue of alcatel in movistar, something else is the huawei ... that the omci is made for it.

manias21, 2021-12-24:
I have checked about 16 firmware versions, in 30% it supports ploam 20 hex characters, including Carlitos, and in the rest only 10 in hex, but some admit to assign it in diag the 20 characters, but none of the firmwares gives positive result in Alcatel, for this reason I have requested support with an external company, for work reasons I have extended the tests to February, the manufacturer through a negotiation is motivated to process a solution, when possible I indicate the data and provide the progress.
But at the moment no one known has a solution against the issue of alcatel in movistar, something else is the huawei ... that the omci is made for it.

Danno88-88, 2022-01-12:
Hi Manias, would you share me the procedure to flash the UF-Instant with a random firmware allowing to change PLOAM and Serial and OMCI app as well?

My ploam is 12 characters, so no need to trouble with a 20 characters one.

Anime4000, 2022-03-01:
Hi everyone...

I have update this repo for more proper documentation and troubleshoot

I made [Setup_Stick.md](https://github.com/Anime4000/RTL9601C1/blob/main/Docs/Setup_Stick.md) document for how to and information why. I made this because my OLT only allow handful Universal ONU model, illegal ONU will not provision even having `O5` ONU Status

Currently trying support 2.5GbE on V2801F stick, see #17
*Because RLT9601C1 can support HiSGMII, TWCGPON657 can run 2.5GbE*

If you want to try 2.5GbE, I have update info at [FLASH_GETSET_INFO.md @ 2.5GbE](https://github.com/Anime4000/RTL9601C1/blob/main/Docs/FLASH_GETSET_INFO.md#lan_sds_mode-min-1-max-5)

and, I update a [qemu-test.sh](https://github.com/Anime4000/RTL9601C1/tree/main/Tools/emulator), you can modify `rootfs` without UART or NAND programmer

EnfermeraSexy, 2022-03-05:
So, after reading almost all the thread, the Ubiquiti uf-instant has problems with alcatel OLTs. But is there any fix to make it work with the V2801F firmware for other OLTs like huawei ones?
I'm going to receive one unit in the coming days and i would like to test it with Orange/Jazztel (Spain).

manias21, 2022-03-05:
I give you an easy answer, after having tested more than 15 sfp, about 20 firmwares, in a percentage it admits 20 hex of ploam, but why?, I only get 05, the pppoe does not connect me, be it the V2801F as it is carlitoxx v1 and v2, with alcatel nothing at all... there is no omci profile that is bounded for her. As much Chinese as Spanish as you want, I can say that I have invested two months. There is currently nothing compatible in sfp for the alcatel olt.
Now let's talk about the huawei, because for some specific models yes, you can get the 05 and dhcp or ppeo and it usually works, but it all depends on the huawei olt model that is in your area.
Because now I am testing a fiber that only needs the mac to link to 05 and the pppoe does not connect me and that is huawei too, be it carlito's or V2801F, the same thing happens to me as with the alcatel.
Ami Maria's plants grow without being able to see them... But it smells like Maria, I don't know if you follow me....

[52 more replies]