Anime4000/RTL960x #3: hi,A new brand gpon stick

bluehj777, 2021-03-31:
hi,TWCGPON657 is hard to find in taobao,They are all out of stock.According to the seller, it takes three months or more to re-produce.but after a hard search, I found a new similar brand in Singapore.It seems that it should be very niche
Although it is more expensive, I have checked with the seller and it also uses Realtek chips,I think it's supposed to be the same thing as TWCGPON657.The package I ordered is on the way.I'll run some tests on it when I get it.Anyway,thanks for your more information in here.
BTW since lowyat.net is no longer accepting new user registrations, I'm leaving a message here.

https://dimiks.store/product/universal-gpon-epon-onu-ont-sfp-stick-module/

Anime4000, 2021-03-31:
@bluehj777 

Hi,

I didn't know TWCGPON657 is out of stock 😶 ... Maybe many users find out that TWCGPON657 can use V2801F firmware on the cheap?

The stick you bought is V2801F, and Dimiks are not from Singapore, it was from Ukraine, because I bought my first stick.

# Build Quality
## V2801F
I have explore inside `V2801F` hardware, found out using same/recycle case as `DFP-34X-2C2` and very few thermal pad, in prolong use in hot weather, my `V2801F` died due to excessive heat.

You need buy a fan to control V2801F overheating

V2801F firmware are very stable and highly compatible many OLT vendor, if didn't work, simply use `OMCI_FAKE_OK 1` and will work!

## TWCGPON657
I also explore inside of `TWCGPON657`, thermal well build and less heat compared with `V2801F`

`TWCGPON657` firmware can cause lock up, not stable, this very reasons I update firmware `V2801F` into `TWCGPON657` hardware.

bluehj777, 2021-03-31:
Is it a Ukrainian company registered in Singapore？
https://dimiks.store/about-us/
Anyway, this is the only place I can get it at the moment. Many normal electronic goods in China are now out of stock because of chips.

Anime4000, 2021-03-31:
I got a parcel from Ukraine,

I live in Malaysia, delivery from Singapore to Malaysia just take 1 - 2 days only, but mine take 3 weeks to arrive,
[image]

I see...

closed 2022-03-24 by Anime4000