Anime4000/RTL960x #5: UART Pintout for this model

jjeziorny, 2021-05-14:
I locked myself out of an SFP which has the RTL chipset by missing a " in the lastgood.xml, so need to get UART access to it to try and recover it.
Any ideas what pins should I use?

[image]
[image]
[image]

Anime4000, 2021-05-16:
Hi,

You have same stick with me, unfortunately V2801F stick did not UART pinout.

Best effort to fix your issue is by reading and writing NAND Flash directly

jjeziorny, 2021-05-17:
Thanks @Anime4000,
Do you have a working one? Can you check if it creates a second interface/IP?  Some of the other SFPs I have do create a second management interface on VLAN2, which can be used to access it in case of issues with the main admin IP

Like this one:
```
# ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 16436 qdisc noqueue state UNKNOWN 
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UNKNOWN qlen 1000
    link/ether ac:4a:fe:98:0a:f2 brd ff:ff:ff:ff:ff:ff
3: eth0.2: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN qlen 1000
    link/ether 00:00:00:01:00:02 brd ff:ff:ff:ff:ff:ff
4: nas0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UNKNOWN qlen 1000
    link/ether 00:00:00:01:00:02 brd ff:ff:ff:ff:ff:ff
5: pon0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UNKNOWN qlen 1000
    link/ether 00:00:00:01:00:02 brd ff:ff:ff:ff:ff:ff
6: br0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UNKNOWN 
    link/ether ac:4a:fe:98:0a:f2 brd ff:ff:ff:ff:ff:ff
    inet 192.168.2.1/30 brd 192.168.2.3 scope global br0
7: nas0_0: <BROADCAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UNKNOWN qlen 10
    link/ether ac:4a:fe:98:0a:f5 brd ff:ff:ff:ff:ff:ff
9: br0.2@br0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP 
    link/ether ac:4a:fe:98:0a:f2 brd ff:ff:ff:ff:ff:ff
    inet 192.168.10.1/24 brd 192.168.10.255 scope global br0.2
```

Note the last entry:
```
9: br0.2@br0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP 
    link/ether ac:4a:fe:98:0a:f2 brd ff:ff:ff:ff:ff:ff
    inet 192.168.10.1/24 brd 192.168.10.255 scope global br0.2
```

Never played with writing NAND Flash, so might take a while to achieve that.
I have also bought a SFPTotal Whistle to try and reprogram SFPs, but so far couldn't get it to read any of the SFPs I have

Anime4000, 2021-05-17:
> Do you have a working one? Can you check if it creates a second interface/IP? Some of the other SFPs I have do create a second management interface on VLAN2, which can be used to access it in case of issues with the main admin IP

most of GPON Stick have `192.168.1.1` for WebGUI/Managemen

If you want to access, simply add `sfp @ 192.168.1.1` into NAT group.

I add `sfp` in `WAN` group on Mikrotik Router, from Mikrotik LAN `192.168.88.0/24` can access `192.168.1.1` of `sfp`

No need add VLAN or Bridge.

With this, you can increase `sfp` MTU to `1508` then PPPoE can use `1500` MTU in Mikrotik

jjeziorny, 2021-05-17:
Yes I know, however whilst configuring a different IP on the SFP I missed a " in the lastgood.xml, now the SFP lost the IP and I can't access.
Some SFPs have a secondary IP configured by default, on a VLAN or not, hence I was asking if you could boot your SFP and do an `ip addr` so see if it has any other IP configured.

Anime4000, 2021-05-20:
> Yes I know, however whilst configuring a different IP on the SFP I missed a " in the lastgood.xml, now the SFP lost the IP and I can't access.
> Some SFPs have a secondary IP configured by default, on [...]

sorry, `ip addr` only show 192.168.1.1 on `br0`, no other ip address

you can reflash the firmware to fix missing file?

jjeziorny, 2021-05-20:
I would love to, never done it. I guess I just a new thing to learn. 
Any tips on how to do it?

Anime4000, 2021-05-25:
I haven't go that route yet, so, I not sure how to read and write NAND flash

closed 2022-03-24 by Anime4000