Anime4000/RTL960x #10: help me please with generating new VS_AUTH_KEY

bullgod-mort, 2021-08-31:
Thank you in advance.
Since lowyat.net is no longer accepting new user registrations so creating new issue is the only way to contact you.
Could you please help me with generating new VS_AUTH_KEY?
ELAN_MAC_ADDR: 88:86:03:F6:5D:(91-9D)
HW_HWVER: 323.A

Anime4000, 2021-08-31:
```
88:86:03:F6:5D:91
88:86:03:F6:5D:92
88:86:03:F6:5D:93
88:86:03:F6:5D:94
88:86:03:F6:5D:95
88:86:03:F6:5D:96
88:86:03:F6:5D:97
88:86:03:F6:5D:98
88:86:03:F6:5D:99
88:86:03:F6:5D:9A
88:86:03:F6:5D:9B
88:86:03:F6:5D:9C
88:86:03:F6:5D:9D
```
```
323.A
```

VS_AUTH_KEY, Works within range: `88:86:03:F6:5D:00` - `88:86:03:F6:5D:FF`
```
B25322F12DBD1EE07B3C17EFEBBC3E26
```

[image]

bullgod-mort, 2021-09-01:
thanks a lot!

bullgod-mort, 2021-09-02:
sorry to bother you again, but could you suggest what could be wrong while updating firmware?
at the moment I have firmware C00R657V00B12, when I try to update the firmware version via the web interface to firmware V2801F_V1.9.0-201104, after confirmation I get a message that "The site <module IP address> did not send data."
[image]

Anime4000, 2021-09-02:
You need to downgrade the firmware first, since boot image having different checksum, that's why it fail/no changes

bullgod-mort, 2021-09-02:
Any b12 should work? But as i see in details i already have b12 on board.
Another question is: how to run au3 script?

bullgod-mort, 2021-09-03:
thanks. firmware updated to V2801F_V1.9.0-201104.tar
but during cloning ONU information i can't change Software Version:
[image]
[image]

Anime4000, 2021-09-03:
this known bug, so far don't have a way to make it permanent

bullgod-mort, 2021-09-03:
After trying to activate manual tagging I've lost access to module. Is there any way to reset it to factory or retrieve access to it?