Anime4000/RTL960x #13: HG8240 VS_AUTH_KEY Help

mauimauer, 2021-10-01:
Hey there,

could you help me out for the correct VS_AUTH_KEY so I can set the correct ELAN MAC? Your help is very much appreciated:

VS_AUTH_KEY?
ELAN_MAC_ADDR: 80D4A5AB97B3-BB (9)
HW_HWVER: 140DC601

Many thanks.

kswong319, 2021-10-06:
VSAUTHKEYGEN.EXE 80:D4:A5:AB:97:00 140DC601
VS_AUTH_KEY = 8D6668143DCE7FE0D7178C4D7F933B1C

kswong319, 2021-10-06:
> 
> 
> Hey there,
> 
> could you help me out for the correct VS_AUTH_KEY so I can set the correct ELAN MAC? Your help is very much appreciated:
> 
> VS_AUTH_KEY? ELAN_MAC_ADDR: 80D4A5AB9700-10 HW_HWVER: 140DC601
> 
> Many thanks.

VSAUTHKEYGEN.EXE 80:D4:A5:AB:97:00 140DC601
VS_AUTH_KEY = 8D6668143DCE7FE0D7178C4D7F933B1C

[image]

closed 2021-10-09 by mauimauer