Anime4000/RTL960x #14: Compatibility of TWCGPON657 with Mikrotik

skon77, 2021-12-06:
Hi,

I have a TWCGPON657 stick with firmware V2801F_V1.9.0-201104.
In the optical media converter status is O5. Is ok.

But in my case Mikrotik RB5009 (SFP status - link on) does not start DHCP client on SFP port. Constantly status "searching...". There is no internet.

If you connect through mediaconverter to same RB5009 but in Eth1 port, then DHCP client gets ip right away. There is internet in bridge.

Have you ever encountered a similar situation?

Anime4000, 2021-12-06:
* SFP GPON ONU Stick is full **bridge**, from `nas0_0` to `eth0`:
```
# brctl show
bridge name     bridge id               STP enabled     interfaces
br0             8000.043389fe8f59       no              eth0
                                                        nas0_0
```

* SFP GPON ONU do not have DHCP server
  1. You need set `192.168.1.2/24` on SFP
  2. You can put `sfp1` as `wan` group in Mikrotik, so you can login 192.168.1.1 from 192.168.88.0/24 network (NAT)

* Router are responsible to have Internet connection
  1. Set a VLAN under `sfp1`
  2. Set a PPPoE under new `vlan`

I have RB3011UiAS-RM, now using EdgeRouter 12, both are work fine
[image]

I believe your media converter detect the stick dont have DHCP, so the device make one.

how your internet get connected originally?

for me, I need make a VLAN 621 on `sfp1` then PPPoE on `vlan621`. EdgeRouter 12 connect and get WAN Public IPv4 and IPv6

skon77, 2021-12-06:
Thank you so much for your feedback and for your work with GPON sticks.

In my case, my ISP operates on Gpon technology. PPPoE authorization is not used. As I understand it, my ISP uses IPoE, authorization via DHCP with option 82. A normal IP without PPP, VPN, etc., with obtaining the address via DHCP. 

There are also Triple Play services. Vlan 30 for Internet, Vlan 20 for TV and Vlan 10 for SIP. But I only use Internet, so I might not have those vlan. When connecting through media converter and Eth1 port, I enabled Torch in the Mikrotik and didn't see any Vlan. At the same time in the SFP stick I have enabled Vlan transparent mode.

I have a PLOAM password and that is enough to get the O5 status in the media converter. Or as originally, I had an optical router from my ISP - SerComm RV6699 model.

Also, if it is important, I now have a white static ip from the ISP.

In Mikrotik "quick set" settings if I select port "sfp1" and "address acquisition" to "automatic" mode, router does not put anything in the fields "ip, netmask, gateway". DHCP client is "searching..." Although in a similar situation with the connection through the media converter and port "eth1" - these fields are immediately filled in correctly automatically.

Anime4000, 2021-12-06:
If I reading correctly...

have you set **"Vlan 30 for Internet"** in mikrotik and set DHCP auto & `option 82` on `vlan30` ?

---

> In Mikrotik "quick set" settings if I select port "sfp1" and "address acquisition" to "automatic" mode, router does not put anything in the fields "ip, netmask, gateway". DHCP client is [...]

and

> I am currently accessing Gpon stick via Windows 10 and media converter in PC LAN port. By setting ipv4: 192.168.1.2 computer, 192.168.1.1 gateway. By the way, if I set it to auto, then I get my white ip directly to the computer (IP 46.138.258.160, Gateway 46.138.250.1).

I not sure how that Media Converter can get internet, is there any setting can enter or Media Converter being smart to auto detect?

Since your Internet come with VLAN 30, and DHCP Option 82, you need VLAN for sfp:
```
Name: vlan1.30
VLAN ID: 30
Interface: sfp1
```

Then set DHCP Client and DHCP Option 82 on `vlan1.30`

This should work...

---

> P.S. Unfortunately, this tip did not work:
> -You need set 192.168.1.2/24 on SFP
> -You can put sfp1 as wan group in Mikrotik, so you can login 192.168.1.1 from 192.168.88.0/24 network (NAT)
> **Going to 192.168.1.2 will open the Mikrotik web authorization. 192.168.1.1 won't open.**

Yes, 192.168.1.2 is IP for sfp, opening `http://192.168.1.2/` will open router WebGUI

To access SFP GPON ONU Stick WebGUI of `192.168.1.1` you need NAT it, if you using ~~Quick Setup~~, Mikrotik will make **Interface Group** named **LAN** and **WAN**

Interface Group WAN should be:
1. sfp1
2. sfp1.30

Interface Group LAN should be:
1. bridge (as all eth switch)

yes, sfp1 need get out from bridge/switch member

skon77, 2021-12-06:
Yes. You understand correctly.

Exact information about the authorization technology is not advertised.
I collected it from the forums among the same users. I'm not sure of everything myself.
Absolutely not PPPoE. Definitely DHCP and PLOAM.

It's not clear with the Vlans. Some of the users have them, some do not. This is probably due to the different settings of the head stations.

We can start from two facts:
1. SFP + Media Converter => Eth1 Mikrotik - is ok!
2. SFP + Media Converter => PC LAN (no additional settings,  auto only) - is ok!

Media Converter is very simple, dumb hardware - https://www.aliexpress.com/i/32951710541.html
No Vlans!

By the way, in RouterOs 7, if SFP1 is set as a master port via quick setup, it comes right out of the bridge. I maked Vlan 30 on SFP1. No result! I see TX packets and no RX. With or without Vlan 30. TX = XX Kb, RX = 0 all time. 

Can we use the computer port for detailed analysis, I have Wireshark?

Anime4000, 2021-12-07:
This seem odd problem, I believe RB5009 SFP is 10GbE, have you try set 1GbE if that work?

If not, I not sure how, my RB3011UiAS also reporting **bad** EEPROM, but have internet...

In my experience, Mikrotik SFP/SFP+ have limited supported model,

also my ER12 only support 1GbE SFP, if I insert SFP Stick that support 10/100/1000, ER12 refuse unless force link negotiate in 1000Mbps FULL

---

> And when Gpon stick is in media converter and connected to Eth1 in Mikrotik, can I access this stick via telnet?
I think you need both Static and DHCP on ETH1?

skon77, 2021-12-07:
Of course! Without a 1Gbps interface lock, SFP module is not detected.
After locking the module is present and link is on.

skon77, 2021-12-07:
> In my experience, Mikrotik SFP/SFP+ have limited supported model

I agree!
That's why I named the topic as a hardware compatibility problem.
I read on the forums that the problems started with RB4011 model. And on one router firmware SFP did not work, on another version everything worked correctly. It seems that new RB5009 with new RouterOS 7.1 is not so easy.

Finally, I would like to get your expert opinion:

1. Does it make sense to return original native TWCGPON657 firmware and try to work with it?

2. In your instructions for flashing TWCGPON657 there is a Modify section in Linux OS. I still don't understand what it's about. Why and what do you need to modify?

3. In my SFP + Media converter => Eth1 RB5009 configuration I applied these settings:
IP Addresses: Address - 192.168.1.2/24, Network - 192.168.1.0, Interface - Lan1
IP Routes: Dst Address - 192.168.1.0/24, Gateway - Lan1, Local Address - 192.168.1.2%LAN1 
And still no access to 192.168.1.1 What else do I need to write in the settings?

Anime4000, 2021-12-07:
> I agree!
> That's why I named the topic as a hardware compatibility problem.
> I read on the forums that the problems started with RB4011 model. And on one RB4011 firmware version did not work, on another version everything worked correctly. It seems that new RB5009 with new RouterOS 7.1 is not so easy.

I have plan to buy RB5009, but not arrive in my country.

there are few people have success with RB4011.

my RB3011UiAS read stick EEPROM as error, meanwhile EdgeRouter 12 read stick EEPROM correctly... What I understand, Ubiquiti support [Realtek EEPROM Emulator](https://lore.kernel.org/netdev/20210109231954.GC1551@shell.armlinux.org.uk/t/) right into kernel

I think Mikrotik need special treatment to read Realtek EEPROM Emulator to work? This I not sure...

---

Finally, I would like to get your expert opinion:

1. Does it make sense to return original native TWCGPON657 firmware and try to work with it?
> You can revert to original firmware by visiting `http://192.168.1.1/upgrade.asp`. I been testing that firmware are interchangeable 

2. In your instructions for flashing TWCGPON657 there is a Modify section in Linux OS. I still don't understand what it's about. Why and what do you need to modify?
> Before I discover **TWCGPON657** can use other firmware, Original **TWCGPON657**  make no Internet even `O5`
> In my country, fiber provide multiple ISP to choose, from Telekom Malaysia, Maxis, Celcom, DiGi, Viewqwest, etc...
> 1. Fiber Provider give various ONU brand, all have 4 LAN Port with no WiFi
> 2. Fiber Provider provision to ONU a Bridge Mode to specific LAN Port!!
> 3. Fiber Provider provision LAN 1 exclusive only to Telekom Malaysia, LAN 2 belong to any ISP, LAN 3 for 2nd ISP, so on...
>
> So, every stick on the market only accept LAN 1 provision, including **TWCGPON657** firmware, so I try to modify some OMCI file to force bridge on everything! Only **V2801F** support multiple LAN Port provision!!!
> You can boot into firmware using `qemu` to see the modification is success

3. In my SFP + Media converter => Eth1 RB5009 configuration I applied these settings:
IP Addresses: Address - 192.168.1.2/24, Network - 192.168.1.0, Interface - Lan1
IP Routes: Dst Address - 192.168.1.0/24, Gateway - Lan1, Local Address - 192.168.1.2%LAN1
And still no access to 192.168.1.1 What else do I need to write in the settings?

Your current setting, do you have Interface Group `LAN` and `WAN` ?

Try add 192.168.1.1 as `WAN Group`, so you can access 192.168.1.1 from LAN via NAT Firewall

---

P.S. Are you satisfied with Ubiquiti? How does it compare to Mikrotik? ))

> Since I have both RB3011UiAS and EdgeRouter 12, both are great!!!
> EdgeRouter 12 have dual SFP 1GbE
> EdgeRouter EdgeOS quite capable
> I like Mikrotik way of settings the connection
> Currently I use EdgeRouter 12 because have 10 LAN port, 2 SFP port in single unit, I can make every integration.

Halfeez92, 2021-12-29:
> Yes. You understand correctly.
> 
> Exact information about the authorization technology is not opened. I collected it from the forums among the same users. I'm not sure of everything myself. Absolutely not PPPoE. Definitely DHCP and PLOAM.
> 
> It's not clear with the Vlans. Some of the users have them, some do not. This is probably due to the different settings of the head stations.
> 
> We can start from two facts:
> 
> 1. SFP + Media Converter => Eth1 Mikrotik - is ok!
> 2. SFP + Media Converter => PC LAN (no additional settings,  auto only) - is ok!
> 
> Media Converter is very simple, dumb hardware - https://aliexpress.com/item/32951710541.html No Vlans!
> 
> As I said before, I only have Internet from my ISP. I do not use SIP. Users write that in such cases the Internet goes without vlan. The default IPTV channels also without vlan as broadcasting. They require IGMP, but that's not the point. There is IPTV are in the option from ISP, but I don't use it. It seems that in these configurations, the provider disables Triple Play.
> 
> By the way, in RouterOs 7, if SFP1 is set as a master port via quick setup, it comes right out of the bridge. I maked Vlan 30 on SFP1. No result! I see TX packets and no RX on SFP interface. With or without Vlan 30. TX = XX Kb, RX = 0 all time. _I think that maybe when I connect to a PC LAN or Mikrotik Eth1 it creates a condition when the provider up his dhcp server and provides IP. And there is no such condition when using SFP port on Mikrotik._
> 
> Also, Mikrotik does not display the status (temperature and more) of SFP module:
> 
> /interface ethernet monitor sfp1 name: sfp1 status: link-ok rate: 1Gbps full-duplex: yes tx-flow-control: no rx-flow-control: no advertising: link-partner-advertising: sfp-module-present: yes sfp-rx-loss: no eeprom-checksum: bad
> 
> As I understand it for the reason of eeprom-checksum: bad.
> 
> And when Gpon stick is in media converter and connected to Eth1 in Mikrotik, can I access this stick via telnet?

Have you tried with other MikroTik model? RB5009 is currently using v7 RouterOS, there is a lot of bug. Even DHCPv6-PD over PPPoE VLAN reported to have problem and having DHCP client in searching mode. Try with other model that support v6 RouterOS.

You might want to post your thread in MikroTik support forum regarding this issue so they can take a look at it.

skon77, 2022-01-14:
> Try with other model that support v6 RouterOS.
> You might want to post your thread in MikroTik support forum regarding this issue so they can take a look at it.

Unfortunately, I bet on the RB5009 model. As my first Mikrotik router. And ended up being hostage to the poor quality of the ROS 7. The Mikrotik team has [known](https://forum.mikrotik.com/viewtopic.php?t=165386) about this problem since 2020. Now, as far as I know, there is another kind of problem - lack of RX data coming from SFPs in their newer models, including CCRs. I have just such a thing. There are other users with the RX data problem.

Anime4000, 2022-01-15:
is best to wait until it fixed SFP RX issue.

I have use RB3011UiAS-RM and SFP working flawlessly, most any SFP device work just fine! 
The cost of RB3011 not that expansive, maybe you can start with RB3011UiAS-RM

skon77, 2022-03-27:
Hello!

I understand from the Lowyat forum that you bought an RB5009?

Can you share mikrotik config for TWCGPON657 to work correctly?

I also bought another [stick](https://detail.tmall.com/item.htm?spm=a230r.1.14.16.7f685270h31Lqm&id=654848544398&ns=1&abbucket=6&skuId=4727494476663).  And the same problem. Can't even access the interface 192.168.1.1

Anime4000, 2022-03-28:
RB5009 have issues with RTL9601.

For ODI xPON, you need fully power off, insert the stick, power back on

about TWCGPON657, when RB5009 fully boot, insert the stick.

or

you can try `flash set LAN_SDS_MODE 1` or `2` or `3`

`4` and `5`  is 2.5GbE mode which is not supported by RB5009, set these value render your Stick unusable

skon77, 2022-03-28:
That was great advice. Thank you very much.
My stick was set to "2". And nothing worked.
Everything worked miraculously when I applied the value of "1".
I can't access the web interface yet, but that's nothing.
Can I try changing these values on the ODI sticks?

skon77, 2022-03-28:
I checked this. My ODI stick was in mode number "3" and nothing worked on RB5009.
Just like that I put it in mode "1" and everything worked as it should.

Thus my conclusion :
-TWCGPON657 in mode "2" (default) does not work on RB5009;
-ODI in mode "3" (default) does not work on RB5009;
-Both sticks work fine in mode "1";

There is a question. In this mode of Fiber Transceiver can the web interface of the sticks be theoretically available?  I can't even ping 192.168.1.1 (timeout)

Anime4000, 2022-03-28:
do you have TTL-UART on these stick?
see br0 member eth0 and nas0?

skon77, 2022-03-28:
Regarding TTL-UART, I'm not sure.

# brctl show
bridge name    bridge id                      STP enabled       interfaces
br0                   8000.142e5ea21f78     no                       eth0, nas0_0

I can see the web interface through the media converter without any problems.
The sfp port only has internet, no access to 192.168.1.1 even with the built-in mikrotik terminal. At the same time 192.168.1.2 is pinging.

[image]

44.x.x is my external ip which received sfp via dhcp client

Anime4000, 2022-03-28:
oh, you can get internet without VLAN? just DHCP on `sfp1`

what I did with my internet:
* `sfp-sfpplus1` Address: `192.168.1.2/24`
[image]
* Add `sfp-sfpplus1` into `WAN` Interface List
[image]
* `pppoe-out1` via `vlan621` via `sfp-sfpplus1`
[image]

This way, I can access `192.168.1.1` via NAT like this:
[image]

I think... Mikrotik allow one interface to have multiple IP Address and Single DHCP, try this

skon77, 2022-03-29:
> oh, you can get internet without VLAN? 

I am currently using the ODI stick.
Thanks to your script I was able to get the VID information:

l2-table get entry address 4

LUT address: 0x0004 (1K LUT)
L2 Unicast table:
MACAddress        Spa Fid Age Vid  State  Ext  Hash
00:00:00:00:00:00 3   0   1   1    Static 0    IVL
CtagIf Auth DaBlock SaBlock Arp
Dis    Dis  Dis     Dis     Dis

RTK.0> command:l2-table get entry address 468

LUT address: 0x01d4 (1K LUT)
L2 Unicast table:
MACAddress        Spa Fid Age Vid  State  Ext  Hash
62:2E:BD:5A:A3:5E 2   1   7   30   Auto   0    SVL
CtagIf Auth DaBlock SaBlock Arp
En     Dis  Dis     Dis     Dis

RTK.0> command:l2-table get entry address 608

LUT address: 0x0260 (1K LUT)
L2 Unicast table:
MACAddress        Spa Fid Age Vid  State  Ext  Hash
4B:BD:4C:08:B3:B2 2   0   7   30   Auto   0    SVL
CtagIf Auth DaBlock SaBlock Arp
En     Dis  Dis     Dis     Dis

RTK.0> command:l2-table get entry address 612

LUT address: 0x0264 (1K LUT)
L2 Unicast table:
MACAddress        Spa Fid Age Vid  State  Ext  Hash
4B:BD:4C:08:B3:B2 2   1   7   30   Auto   0    SVL
CtagIf Auth DaBlock SaBlock Arp
En     Dis  Dis     Dis     Dis

RTK.0> command:l2-table get entry address 732

LUT address: 0x02dc (1K LUT)
L2 Unicast table:
MACAddress        Spa Fid Age Vid  State  Ext  Hash
6D:9B:7D:1A:88:39 2   1   7   9    Auto   0    SVL
CtagIf Auth DaBlock SaBlock Arp
En     Dis  Dis     Dis     Dis

RTK.0> command:l2-table get entry address 872

LUT address: 0x0368 (1K LUT)
L2 Unicast table:
MACAddress        Spa Fid Age Vid  State  Ext  Hash
3D:3B:D1:27:AB:74 3   1   1   9    Static 0    SVL
CtagIf Auth DaBlock SaBlock Arp
En     Dis  Dis     Dis     Dis

RTK.0> command:l2-table get entry address 984

LUT address: 0x03d8 (1K LUT)
L2 Unicast table:
MACAddress        Spa Fid Age Vid  State  Ext  Hash
00:0B:E0:F0:00:ED 0   0   7   0    Auto   0    SVL
CtagIf Auth DaBlock SaBlock Arp
Dis    Dis  Dis     Dis     Dis

RTK.0> command:l2-table get entry address 988

LUT address: 0x03dc (1K LUT)
L2 Unicast table:
MACAddress        Spa Fid Age Vid  State  Ext  Hash
00:0B:E0:F0:00:ED 0   1   7   9    Auto   0    SVL
CtagIf Auth DaBlock SaBlock Arp
Dis    Dis  Dis     Dis     En

ISP seems to be giving me VLAN 0,1,9,30. I know for a fact that my internet is VLAN 30.
But, when I look at sfp1 port in Mikrotik trough TORCH utility I don't see any vlan tags...

How do I configure my ODI stick to forward vlan tags to the Mikrotik? This will not burden the sfp stick additionally with tagging?

[image]

Anime4000, 2022-03-29:
you can try that PON WAN

LAN_1 port mapping seem no effect when I tried

skon77, 2022-03-29:
Tried different configurations on both of my sticks.
Transparent vlan tag forwarding did not work.
Either there were no tags in RB5009 or Internet didn't work.

But I noticed an interesting difference between two sticks.

 **TWCGPON657 (FW V2801F_V1.9.0-201104)**
[image]

**ODI XGPON DFP-34G-2C2 (FW original)**
[image]

I have Internet and TV in my tarif plan from my provider.
I don't have an IPTV set-top box connected, I don't watch TV.
Do I understand correctly that ODI stick sees and transmits to  RB5009 broadcast TV? And TWCGPON657 doesn't even see it?

Anime4000, 2022-03-30:
about heat, ODI is better, since they use newer SoC model than V2801F/TWCGPON657

ODI use RTL9601D System on Chip

also ODI handle IPTV better, what I found that they merge IGMP traffic into main

skon77, 2022-05-03:
Hi @Anime4000 

Is there a fix for bug with rebooting and TWCGPON657 link up?
When updating ROS or rebooting the router, I have to go to the router and physically rearrange the SFP module.
If I remember correctly, originally in the first firmware versions such a problem did not exist?

Anime4000, 2022-05-03:
> And ODI module seems to have the same problem. At least on my RB5009.
I use `LAN_SDS_MODE 0` on my RB5009, it takes 2 minute to connect, log show ODI Stick connect with RB5009 using `<4>change mode to 7(SGMII Force)`

V2801F and RB5009 not play very well, require to re-insert the stick to work,

V2801F and RB3011 just work fine, during boot, SFP cage is having power cycle, RB5009 don't have power cycle during boot