Anime4000/RTL960x #15: auto reboot even with correct VS_AUTH_KEY

amaumene, 2022-01-02:
Hi,

I have a V2801F stick. I have set the intel VS_AUTH_KEY found [here](https://github.com/Anime4000/RTL9601C1/blob/main/Docs/FLASH_GETSET_INFO.md) but I still need to do ```echo 3 > /proc/fiber_mode``` otherwise my stick will keep rebooting in loop.

I'm not too sure it's related but ```# cat /tmp/authinfo``` shows:
```auth success```

My firmware is version ```V1.9.0-201104 --  Wed Nov 4 09:41:21 CST 2020```

Any idea why? Thanks in advance.

Anime4000, 2022-01-02:
Nice Cat XD

when did you buy V2801F? I heard there so many revision.

Have you save previous value `flash get` ? try put original value see if reboot still happen

amaumene, 2022-01-07:
I bought it recently. I tried with the original values but I'm facing the same issue... :(

Anime4000, 2022-01-07:
if your stick is ODI with Realtek, it not work mostly, the laser driver quite different, currently I testing it

closed 2022-03-24 by Anime4000