Anime4000/RTL960x #16: IGMP

kofeisen, 2022-02-25:
I use Gpon+mikrotik, but can't watch iptv, do you have any information about IGMP command?

Anime4000, 2022-02-25:
GPON stick run in full bridge, you need configure IGMP at router side.
my country using ONU Box + Router. Router LAN Port 4 configured for IPTV

kofeisen, 2022-02-26:
> 

I configured igmp proxy, but can't see any packets. If it's not GPON reason, it's probably my configuration problem.Thank you.
[image]

Anime4000, 2022-02-26:
How your typical fiber setup? ONU Box and Router or something else?

kofeisen, 2022-03-01:
[image]

Anime4000, 2022-03-01:
Have you try ISP ONU as bridge mode and plug into Mikrotik if IPTV working?

vuducdong, 2022-03-01:
> [image]

There are 2 cases:

1. if sfp runs with auto vlan, then you are probably missing multicast vlan
(this vlan is different from vlan43, only appears when you watch live, you can torch port sfp to see if this vlan is available, when you try watching any 1 tv channel live)

2. if sfp supports manual vlan: transparent mode --> please switch to this mode and try again.

kofeisen, 2022-03-01:
You said GPON stick run in full bridge.so it's supposed transparent mode. but I ignored multicast vlan.I don't know how to configure ROS yet. I'll study it. Thank you

Anime4000, 2022-03-01:
it should be run in Full Bridge.

This Chinese guide tell how to setup IGMP in mikrotik:
https://danteng.org/ros-routing-igmp-proxy-to-watch-iptv/

however this guide using method IGMP on all LAN port instead of dedicated LAN port

vuducdong, 2022-03-01:
> You said GPON stick run in full bridge.so it's supposed transparent mode. but I ignored multicast vlan.I don't know how to configure ROS yet. I'll study it. Thank you

[image]

yes, pls try it

kofeisen, 2022-03-01:
@vuducdong hi, I can't find multicast VLAN. The ISP ONU igmp VLAN is "-1", yes, The device is filled in as-1.
When I use ros bridge touch, no VLAN id.
[image]

vuducdong, 2022-03-02:
hi @kofeisen ,

Where's your vlan 43 ??? and vlan internet ???

Please see SFP web management, if support vlan transparents mode
[image]

you need to make sure that there is vlan on sfp and then on sfp port mikrotik can see vlans from sfp passed.

kofeisen, 2022-03-02:
hi @vuducdong 
What's your stick firmware version? I use C00R657V00B15_20201222, have no this option like you.
[image]

vuducdong, 2022-03-02:
hi @kofeisen 

I'm using another sfp and the chipset is RTL9601B, each sfp manufacturer has their own firmware.

I'm wondering, does your internet have vlan?

kofeisen, 2022-03-02:
@vuducdong 
My ISP has no internet VLAN id, haha, Very unique. IPTV vlan 43.
[image]
[image]

skon77, 2022-04-18:
Hello, everyone!

Can anyone help me set up IPTV? Or advise me where to read?

Right now I have an ODI stick with transparent VLAN mode.

In my Mikrotik RB5009 I see in Torch on SFP1 interface:
-Multicast without any VLANs.
-Internet with VLAN 30 (getting IP through DHCP client).
I also know that ISP transmits IPTV via VLAN 20, but in reality I don't see it in Torch. Will it show up when IPTV box works or is the stick blocking it?

[image]
I learned from my ISP that streams from .5 subnet are encrypted with Verimatrix, in .4 subnet some of the streams are Securmedia encrypted. My SmartTV box will accept it, but I probably won't be able to check something on my computer via VLC.

Right now I have the following settings on my RB5009:
bridge -> all LAN ports
wan -> VLAN 30, VLAN 20
lan -> bridge
DHCP client -> VLAN 30
SFP1 is just the parent interface for VLAN 30/20 and is not included anywhere else.

When I turn on SmartTV box in LAN port it receives from the provider the settings of my account, correctly indicates a package of channels to which I am subscribed, shows the schedule of TV programs, but when you turn on any channel nothing happens. Blank screen.

I understand, that I have to set IGMP proxy, but I am confused with the settings. SmartTV box needs Internet via VLAN 30, it needs multicast without VLAN, and it may need VLAN 20 for IPTV... How do I get it all together?

<details> 

omcicli mib get 84

VlanTagFilterData

EntityID: 0x03
FilterTbl[0]: PRI 0,CFI 0, VID 10
FwdOp:  0x10
NumOfEntries: 1

EntityID: 0x04
FilterTbl[0]: PRI 0,CFI 0, VID 20
FwdOp:  0x10
NumOfEntries: 1

EntityID: 0x05
FilterTbl[0]: PRI 0,CFI 0, VID 30
FilterTbl[2]: PRI 0,CFI 0, VID 32
FwdOp:  0x10
NumOfEntries: 2

EntityID: 0x06
FilterTbl[0]: PRI 0,CFI 0, VID 40
FwdOp:  0x10
NumOfEntries: 1

EntityID: 0x07
FilterTbl[0]: PRI 0,CFI 0, VID 50
FwdOp:  0x10
NumOfEntries: 1

EntityID: 0x08
FilterTbl[0]: PRI 0,CFI 0, VID 60
FwdOp:  0x10
NumOfEntries: 1
</details>