Anime4000/RTL960x #17: HSGMII firmware

stich86, 2022-02-26:
Hi @Anime4000 

I saw you post on lowyat.net about TWCGPON657 that should support HSGMII on LAN side (2.5Gbit/a). Looks like that also HiSense LTE3415 reports this protocol on the data sheet. Chipset is still RTL9601C.

Do you have any info about that firmware? Here in Italy we are struggling with the various LantiQ sticks (at the moment the only one that support 2.5 also on line side) but for particulars FTTH provider we cannot achieve full speed compared to external ONT

Thanks in advance

Anime4000, 2022-02-27:
Firmware for TWCGPON657 is there in Firmware folder, I not sure Firmware is interchangeable, however I haven't tried 2.5GbE since I don't have 2.5GbE capable SFP+ Router

stich86, 2022-02-27:
> TWCGPON657

So we cannot use TWCGPON657 firmware on V2801F?

TWCGPON657 isn't available everywhere :(

Anime4000, 2022-02-27:
Yes, I have tried it, the PON laser didn't give any reading.

need someone reverse engineering rtl driver to support 2.5GbE

stich86, 2022-02-27:
> Yes, I have tried it, the PON laser didn't give any reading.
> 
> 
> 
> need someone reverse engineering rtl driver to support 2.5GbE

Do you have the TWGPON? Can be interesting to get an "no getenv" looks like that on the V2801F there are references to HoSGMII may be we just need to enable a flag on the uBoot or in XML

Anime4000, 2022-02-27:
> > Yes, I have tried it, the PON laser didn't give any reading.
> > need someone reverse engineering rtl driver to support 2.5GbE
> 
> Do you have the TWGPON? Can be interesting to get an "nv getenv". Looks like that on the V2801F's kernel there are references to HiSGMII may be we just need to enable a flag on the uBoot or XML

currently I use TWCGPON657 with V2801F firmware..

here my `nv getenv`
```
# nv getenv
Valid environment: 2
b0=setenv bootargs ${bootargs_base} ${mtdparts0} ${rst2dfl_flg}; bootm ${img0_kernel}
b1=setenv bootargs ${bootargs_base} ${mtdparts1} ${rst2dfl_flg}; bootm ${img1_kernel}
baudrate=115200
boot_by_commit=if itest.s ${sw_commit} == 0;then run set_act0;run b0;else run set_act1;run b1;fi
boot_by_tryactive=if itest.s ${sw_tryactive} == 0;then setenv sw_tryactive 2;setenv sw_active 0;saveenv;run en_wdt;run b0;else setenv sw_tryactive 2;setenv sw_active 1;saveenv;run en_wdt;run b1;fi
bootargs=console=ttyS0,115200 mtdparts=rtk_spi_nor_mtd:256K(boot),8K(env),8K(env2),240K(config),3M(k0)ro,4864K(r0)ro,3M(k1),4864K(r1),4K@0ro,4K@0ro,4K@0ro,4K@0ro,3M@512K(linux),4864K@3584K(rootfs) root=31:5
bootargs_base=console=ttyS0,115200
bootcmd=if itest.s ${sw_tryactive} == 2; then run boot_by_commit;else run boot_by_tryactive;fi
bootdelay=5
en_wdt=mw b8003268 e7c00000
erase_cfgfs=sf erase ${fl_cfgfs} +${fl_cfgfs_sz}
erase_env=sf erase ${fl_env} +${fl_env_sz};sf erase ${fl_env2} +${fl_env_sz}
ethact=LUNA GMAC
ethaddr=00:E0:4C:00:00:00
filesize=354800
fl_boot_sz=40000
fl_cfgfs=44000
fl_cfgfs_sz=3c000
fl_env=40000
fl_env2=42000
fl_env_sz=2000
fl_kernel1=80000
fl_kernel1_sz=300000
fl_kernel2=840000
fl_kernel2_sz=300000
fl_rootfs1=380000
fl_rootfs1_sz=4c0000
fl_rootfs2=b40000
fl_rootfs2_sz=4c0000
ftfp_base=80000000
fx1000_init=mw bb000084 00000048
img0_kernel=94080000
img1_kernel=94840000
ipaddr=192.168.1.3
mtdparts0=mtdparts=rtk_spi_nor_mtd:256K(boot),8K(env),8K(env2),240K(config),3M(k0)ro,4864K(r0)ro,3M(k1),4864K(r1),4K@0ro,4K@0ro,4K@0ro,4K@0ro,3M@512K(linux),4864K@3584K(rootfs) root=31:5
mtdparts1=mtdparts=rtk_spi_nor_mtd:256K(boot),8K(env),8K(env2),240K(config),3M(k0),4864K(r0),3M(k1)ro,4864K(r1)ro,4K@0ro,4K@0ro,4K@0ro,4K@0ro,3M@8448K(linux),4864K@11520K(rootfs) root=31:7
netmask=255.255.255.0
serverip=192.168.1.7
set_act0=if itest.s ${sw_active} != 0;then setenv sw_active 0;saveenv;fi
set_act1=if itest.s ${sw_active} != 1;then setenv sw_active 1;saveenv;fi
sgmii_init=mw bb000084 00000044
stderr=serial
stdin=serial
stdout=serial
sw_active=0
sw_commit=0
sw_crc0=4f036fd6
sw_crc1=4f036fd6
sw_tryactive=2
sw_valid0=1
sw_valid1=1
sw_version0=V1.9.0-201104
sw_version1=V1.9.0-201104
upb=tftp ${ftfp_base} plr.img && crc32 ${fileaddr} ${filesize} && sf erase 0 +${fl_boot_sz} && sf write ${fileaddr} 0 ${filesize}
upk=tftp ${ftfp_base} uImage && crc32 ${fileaddr} ${filesize} && sf erase ${fl_kernel1} +${fl_kernel1_sz} && sf write ${fileaddr} ${fl_kernel1} ${filesize}
upk1=tftp ${ftfp_base} uImage && crc32 ${fileaddr} ${filesize} && sf erase ${fl_kernel2} +${fl_kernel2_sz} && sf write ${fileaddr} ${fl_kernel2} ${filesize}
upr=tftp ${ftfp_base} rootfs && crc32 ${fileaddr} ${filesize} && sf erase ${fl_rootfs1} +${fl_rootfs1_sz} && sf write ${fileaddr} ${fl_rootfs1} ${filesize}
upr1=tftp ${ftfp_base} rootfs && crc32 ${fileaddr} ${filesize} && sf erase ${fl_rootfs2} +${fl_rootfs2_sz} && sf write ${fileaddr} ${fl_rootfs2} ${filesize}
upt=tftp 80000000 img.tar && upimgtar ${fileaddr} ${filesize}
upv=tftp 80000000 vm.img;upvmimg ${fileaddr}
yu=loady 80000000 && cp.b 80000000 81000000 ${filesize} && cmp.b 80000000 81000000 ${filesize} && sf erase 0 ${filesize} && sf write 80000000 0 ${filesize}

```

stich86, 2022-02-27:
Thanks!

So can I flash TWGCP FW on V2801F as second image? I want to check if it can links at 2.5GbE

Because we are using uBoot it's easy to rollback using serial. Do you know if v2801f has UART exposed on SFP side?

Thanks again

Anime4000, 2022-02-27:
You can flash V2801F on TWCGPON657, you need tweak it to make PON Laser working which is I dont know yet...

There 3 different version `C00R657V00B13` I not sure which one is 2.5GbE
```
C00R657V00B13_20191024.tar
C00R657V00B13_20191205.tar
B13_20200507.tar
```

You try all 3 and tell me which one support 2.5GbE, and check `nv getenv` especially `sgmii_init=` value

I have TWCGPON657 and V2801F stick, both stick are not expose UART on SFP even UART pin

stich86, 2022-02-27:
Which tweak is needed? Just use these tar file on the WebUI update?

Thx

Anime4000, 2022-02-27:
just flash it as usual, if not working, you need downgrade V2801F firmware first

stich86, 2022-02-27:
> just flash it as usual, if not working, you need downgrade V2801F firmware first

Currently I'm using the latest firmware found on this repos on my V2801F. I'll try it tonight an report back the findings ;)

Anime4000, 2022-02-27:
great...

if you plan to flash back V2801F firmware, set `flash set PON_MODE 3` to prevent V2801F auto restart, because TWCGPON657 did not have `VS_AUTH_KEY`

`flash set PON_MODE 3` in V2801F behavior is different, you need disconnect fiber then can get back telnet

stich86, 2022-02-27:
good and bad news...

good news: I was able to flash both 2019 and 2020 B13/B15 files on my V2801F. On first attempt i catch into /proc/kmsg this log:

`<4>change mode to 5(HiSGMII MAC)`

so looks like that the TWCGPON657 acts in a different way. also a "nv getenv" doesn't reports any difference between the two firmware. On the login page I was able to read the TX Power of laser, but not able to test with fiber attached.

Now the bad news

After some test I cannot logon anymore on the firmware :( i've tried all password but nothing to do, so i'm out and cannot access by telnet or web.

Do you know if there are a default credential that I can try?

thanks in advance

stich86, 2022-02-27:
Another little update:

the code into kernel 

`<4>change mode to 5(HiSGMII MAC)`

it's called when parameter `LAN_SDS_MODE` is set to `5`. Switch it to `1` reports `<4>change mode to 1(Fiber 1G)`

Now I cannot test it on my MacchiatoBIN because the eeprom is empty and Linux doesn't initialize the stick

@Anime4000 can you send me a `cat /proc/kmsg` of your TWCGPON657 with V2801F and `LAN_SDS_MODE` set to `5`?

Thanks in advance

Anime4000, 2022-02-27:
```
BusyBox v1.12.4 (2020-10-26 08:35:57 CST) built-in shell (ash)
Enter 'help' for a list of built-in commands.

# flash get LAN_SDS_MODE
LAN_SDS_MODE=2
```
That default value for my `LAN_SDS_MODE`

then I set `LAN_SDS_MODE` to `5` and do `cat /proc/kmsg` prints noting

then I `flash get LAN_SDS_MODE` also prints noting and stuck, I need `CTRL+C`

it seem V2801F firmware are not build to support 2.5GbE

> I've tried also to attach the fiber and the stick cannot go ahead of O3 status.

you got RX reading?

stich86, 2022-02-27:
Regarding laser reading, this is my current situation:

[image]

I've tried `LAN_SDS_MODE` to `2-3-4` and there are all reports

`<4>change mode to 2(SGMII PHY)`
`<4>change mode to 3(SGMII MAC)`
`<4>change mode to 4(HiSGMII PHY)`

Now my main problem is the eeprom reading to test if the 2.5GbE it's working..
Your TWCGPON with V2801F reports valid eeprom?

Thanks

Anime4000, 2022-02-27:
I use EdgeRouter 12 can read EEPROM like this:
[image]

Ubiquiti use this [kernel patch](https://lore.kernel.org/netdev/20210111113909.31702-2-pali@kernel.org/) to allow reading RTL9601C1 EEPROM emulator, yes, RTL9601C1 didn't have EEPROM, it just emulating it.

for Mikrotik, I not sure, I need but the device to fully test.

what OS you using inside [MacchiatoBIN](https://macchiatobin.net/)?

stich86, 2022-02-27:
> I use EdgeRouter 12 can read EEPROM like this:
>
> [image]
>
>
>
> Ubiquiti use this [kernel patch](https://lore.kernel.org/netdev/20210111113909.31702-2-pali@kernel.org/) to allow reading RTL9601C1 EEPROM [...]

On MCBIN I'm using OpenWRT snapshot. Before flashing TWCGPON on V2801F, the stick was correctly recognized and the patch you have linked was working. Now the kernel Linux (and also my Ubiquiti switch) didn't recognize any info from the SFP.

It's strage because I'm assuming that the EEPROM should be generated by the OS (may be configd daemon?)..

Anime4000, 2022-02-27:
[image]

there is 2 option I can think of,

1. modify TWCGPON657 firmware with `squashfs-tools` and `qemu-user-static` you can copy `configd` from V2801F to TWCGPON657

2. try DFP-34X-2C2 firmware 

I bought TWCGPON657 [here](https://item.taobao.com/item.htm?spm=a1z09.2.0.0.67002e8dCrqJJr&id=597031866488)
> need to login, use google chrome to translate page without proxy/iframe

I notice `LAN_SDS_MODE` have PHY (Layer 1) and MAC (Layer 2), have you try `2` or `4` (PHY) if made a different ?

stich86, 2022-02-27:
> [image]
>
>
>
> there is 2 option I can think of,
>
>
>
> 1. modify TWCGPON657 firmware with `squashfs-tools` and `qemu-user-static` you can copy `configd` from V2801F to TWCGPON657
>
>
>
> 2. try DFP-34X-2C2 [...]

As I've said until I have problems with the eeprom the OpenWrt on my MCBIN doesn't initialize the SFP port :(

Regarding DFP-34X-2C2 firmware, is it working on V2801F?

Anime4000, 2022-02-27:
> As I've said until I have problems with the eeprom the OpenWrt on my MCBIN doesn't initialize the SFP port :(
Sorry, this mean SFP has no connection? ~ 6 am here I haven't sleep 🤣 

> Regarding DFP-34X-2C2 firmware, is it working on V2801F?
This one I not tried yet, because my V2801F has been dead for 1 year because invalid `VS_AUTH_KEY`, my plan to read NAND flash and modify it

stich86, 2022-02-27:
The problem is that on Linux kernel SFP needs to pass various steps, in my case the SFP_MOD_PROBE that check the EEPROM doesn't pass validation and so the cage is not enabled. Unifi switch is less restrictive but it's only 1Gbit/s device so I cannot test HSGMII@2.5GbE

I don't understand one thing: if I revert back to the V2810F firmware what's are the correct step to follow and avoid boot loop?

Thanks

Anime4000, 2022-02-28:
I see

to revert back V2801F firmware, do this:
* Downgrade TWCGPON657 firmware to B13
* `flash set PON_MODE 3`
* Upgrade V2801F firmware latest
* After updated, `flash set VS_AUTH_KEY`

doing `PON_MODE 3` will put stick in Fiber Ethernet mode, disconnect fiber until updated `VS_AUTH_KEY`

stich86, 2022-02-28:
ok so PON_MODE 3 avoid auto-reboot right?

Today i'll try to force a link (with clean up sfp.c on kernel) to see if the stick can negotiate at 2500

stich86, 2022-02-28:
> ok so PON_MODE 3 avoid auto-reboot right?
> 
> Today i'll try to force a link (with clean up sfp.c on kernel) to see if the stick can negotiate at 2500

about this test, it went bad :(
i think that EEPROM should be populated in the right way, otherwise the PHY media cannot be activated.. so next step after possibility to flash the NAND i'll try to create an hybrid firmware as suggested: uImage from TWCGPON and rootfs from V2801F

stich86, 2022-02-28:
@Anime4000 

May be i've found where the parameter `LAN_SDS_MODE` is readed and configured

the file that change the behaviour is `runlansds.sh`

on the V2810F this is the code:

```
#!/bin/sh

echo 1 > proc/lan_sds/lan_sds_cfg
echo 1 > proc/lan_sds/sfp_app
sfpapp &
```

while TWCGPON657 is parsing the mode from XML file and set it based on the value on it:

```
#!/bin/sh

lan_sds_mode=`flash get LAN_SDS_MODE | sed 's/LAN_SDS_MODE=//g'`
echo $lan_sds_mode > proc/lan_sds/lan_sds_cfg
echo 1 > proc/lan_sds/sfp_app
sfpapp &
```

So I think that V2801F with modded files and new rootfs can change this behaviour.

Now i've to flash back my stick and try..

Anime4000, 2022-03-01:
Nice finding... 👍 

I also manage to update qemu script
https://github.com/Anime4000/RTL9601C1/tree/main/Tools/emulator

making thing easier!

stich86, 2022-03-01:
Little update:

this morning i've repacked teh V2801F firmware changing the file `runlansds.sh`
`LAN_SDS_MODE`  was set to `2`. Now the stick is up (ethernet link) but I cannot access it anymore..

So i've only two way to restore:

1) dump the flash and change the active image on the mtd1/mtd2 partition
2) try to find the UART TX/RX on the RTL9601CI to see what's going on

If I have any news i'll update this thread

Anime4000, 2022-03-01:
I have tried with my own stick `echo 5 > proc/lan_sds/lan_sds_cfg`

yes, it make stick stuck, it seem V2801F not build 2.5GbE in mind, I also notice TWCGPON657 firmware missing `/bin/sfpapp`

stich86, 2022-03-01:
> I have tried with my own stick `echo 5 > proc/lan_sds/lan_sds_cfg`
> 
> yes, it make stick stuck, it seem V2801F not build 2.5GbE in mind, I also notice TWCGPON657 firmware missing `/bin/sfpapp`

yes.. i've tried before flash the stick thinking that needs to be change on boot :(
have you tried to flash firmware of DFP-34X-2C2 on V2801F? looks like this stick read the `LAN_SDS_MODE `and also the file `runlansds.sh` is correct

Anime4000, 2022-03-01:
my V2801F is broken 😢 due to VS_AUTH_KEY and aother stuff, I not sure which BIOS/NAND programmer can read and write

I will update the doc about V2801F not suppor 2.5GbE

stich86, 2022-03-15:
> my V2801F is broken 😢 due to VS_AUTH_KEY and aother stuff, I not sure which BIOS/NAND programmer can read and write
> 
> I will update the doc about V2801F not suppor 2.5GbE

you can read/write the NAND using a CH341a USB programmer and some adaptor for WSON8 chip.

On my last attempt I was able to fix it, but then i've broken the NAND :(
Now i'm currently waiting to have some time to remove the NAND from the PCB and substitute with another one.

stich86, 2022-03-19:
I've received today my DFP-34X-2C2 from AliExpress. There is the latest firmware available (2021-05). Changing the LAN_SDS_MODE to 5 (on stick love via Telnet and not changing flash parameter), force the quirk on OpenWRT (running on MacchiatoBIN), I was able to negotiate the LAN size @ 2.5GbE :)

Now I've to test these things:

- if this stick can go on O5 and if I'm HISGMII mode works as expected
- wait my new FTTH with 2.5/1 profile to test if I can reach full bandwidth 

Keep you updated :)

stich86, 2022-03-20:
Two updates:

- I've restored (removing NAND and flash it back) my V2810F. Stock firmware doesn't work in HiSGMII mode (loss link a nothing else..)
- it likes the ODI (DFP-xxx) firmware, here is a screenshot where the laser information are recognized:

[image]

Now with ODI DFP original stick I'm stuck in o3.. so I've to understood why :(

Anime4000, 2022-03-24:
I received DFP-34X-C2C from Ali Express by ANDDEARSS Store
[image]

During my investigation...

This stick run `<4>change mode to 3(SGMII MAC)` cause my RB5009 cannot communicate even having OK LINK, need full power down, power on then manage to enter `192.168.1.1`

after enter webgui, set `GPON_PLOAM_PASSWD` alone my Fiber Vendor wont send GEM and T-CONT info
[image]

Then I set what my Fiber Vendor required
```
flash set GPON_PLOAM_PASSWD 1234567890
flash set OMCI_FAKE_OK 1
flash set OMCI_OLT_MODE 1
flash set GPON_ONU_MODEL HG8240H
flash set PON_VENDOR_ID HWTC
flash set HW_HWVER BF9.A
```
[image]

Problem is, this stick run as Router Mode, need add Bridge and VLAN
**Fiber Vendor VLAN:**
| VLAN | Desc |
|-------|----------------------|
| `209` | ISP 1 `tr069` |
| `400` | ISP 1 VoIP |
| `500` | ISP 1 Internet |
| `600` | ISP 1 IPTV |
| `621 | ISP 2 Internet |
| `821` | ISP 2 `tr069` |
| `822` | ISP 2 VoIP |
| `823` | ISP 2 IPTV |

I am using ISP 2, DFP-34X-C2C only manage to bridge `vlan209`, not `vlan621` which I use
[image]
[image]

---
I run `flash set LAN_SDS_MODE 5`... my RB5009 SFP+ failed to negotiate 2.5GbE

**hard** lesson learned @stich86 😆

stich86, 2022-03-24:
RB5009 on SFP+ doesn't sync at 2500baseSX, only 1GbE or 10GbE.
But I can assume that this stick can sink at 2.5. Now I've a problem reaching O5 :(

Anime4000, 2022-03-24:
@stich86 

Yea, hardware support it if use OpenWRT on RB5009, just ROS 7 not support it.

I buy TP-XDR5480 that have native 2.5GbE so I can set back to proper LAN_SDS_MODE

your ODI firmware seem old a bit: 2021-05

my stick using: 2021-07

I have upload the firmware, try that, it has half success to connect 😅

stich86, 2022-03-24:
> @stich86
>
> Yea, hardware support it if use OpenWRT on RB5009, just ROS 7 not support it.
>
> I buy TP-XDR5480 that have native 2.5GbE so I can set back to proper LAN_SDS_MODE
>
> your ODI firmware seem old [...]

wait: TP-Link can link at 2.5 into SFP? My ODI stick has also 2021-07, the one of 2021-05 is loaded on V2801F

Anime4000, 2022-03-24:
@stich86 

yea, https://m.tp-link.com.cn/product_1999.html

my friend help me buy at Taobao

I wondering if merge V2801F bridge function into ODI,

V2801F support 4-port Emulation

stich86, 2022-03-24:
> @stich86
> 
> yea, https://m.tp-link.com.cn/product_1999.html
> 
> my friend help me buy at Taobao
> 
> I wondering if merge V2801F bridge function into ODI,
> 
> V2801F support 4-port Emulation

ah ok, it's useless with just a 2.5 GbE port :(
i don't know if can be imported, may be we can swap only rootFS (keeping the kernel).. 

i'll try on my V2801F

stich86, 2022-03-24:
> that TP Link has one SFP 2.5GbE Support, 4 LAN Port and WiFi
> 
> I buy TWCGPON657 from Taobao, can import here.
> 
> the firmware package have `uImage` and `rootfs`. which kernel to keep?

i think to test HSGMII we should take kernel from DFP and rootFS from V2801F, like someone else has done for UF-Instant

stich86, 2022-03-24:
> that TP Link has one SFP 2.5GbE Support, 4 LAN Port and WiFi
> 
> I buy TWCGPON657 from Taobao, can import here.
> 
> the firmware package have `uImage` and `rootfs`. which kernel to keep?

do you have an internal photos of TWCGPON657?
just to understand if it's similar to DFP-34X-C2C.. i've found two pads near RTL9601C chip, may be are UART pinout

Anime4000, 2022-03-24:
I use TWCGPON657 Stick (with V2801F Firmware) currently, take out cause no Internet~

I have tried DFP-34X-C2C with V2801F Firmware, stick keep reboot

I tried V2801F with DFP-34X-C2C Firmware, stick died 😢 

Usually kernel reside at `uImage`, like Linux Desktop can choose kernel.

I try compare `uImage` see what different

stich86, 2022-03-24:
> I use TWCGPON657 Stick (with V2801F Firmware) currently, take out cause no Internet~
>
> I have tried DFP-34X-C2C with V2801F Firmware, stick keep reboot
>
> I tried V2801F with DFP-34X-C2C Firmware, [...]

my V2801F with DFP-34X-C2C boots without any issue and with laser recognition... very strange O_o
for the photo.. when you can it will be very appreciated ^_^

do you think that we can arrange an "remote" buy for TWCGPON657? I cannot receive it to Italy :(

Anime4000, 2022-03-24:
remote buy like, Taobao to Malaysia, then I ship back to you: Malaysia to Italy?
I need contact my Local Logistic how much cost to ship 😄 

what I found between DFP and V2801

```
anime4000@hitoha-vm:~/Desktop/xpon/M114_sfp_ODI_210702$ md5sum uImage 
493fa0e7bb6d67a91cdf5bde18029a3b  uImage
anime4000@hitoha-vm:~/Desktop/xpon/M114_sfp_ODI_210702$ ll uImage 
-rwxr-xr-x 1 root root 1281968 Jul  2  2021 uImage*
anime4000@hitoha-vm:~/Desktop/xpon/M114_sfp_ODI_210702$
```

```
anime4000@hitoha-vm:~/Desktop/xpon/V2801F_V1.9.0-201104$ md5sum uImage 
b98cbd9ab9e7d33a84b0e0d90d5a8d6f  uImage
anime4000@hitoha-vm:~/Desktop/xpon/V2801F_V1.9.0-201104$ ll uImage 
-rw-rw-r-- 1 1013 1013 1199526 Nov  4  2020 uImage
anime4000@hitoha-vm:~/Desktop/xpon/V2801F_V1.9.0-201104$
```

stich86, 2022-03-24:
> remote buy like, Taobao to Malaysia, then I ship back to you: Malaysia to Italy? I need contact my Local Logistic how much cost to ship 😄
>
> what I found between DFP and V2801
>
> ``` [...]

yes V2801F has a smaller kernel, looks like they have removed all "router mode" stuff
regarding "remote shipping", if we can check the internal board of TWCGPON657, we can see if it's similar to DFP-34X-C2C or not

Anime4000, 2022-03-24:
okay, I try take all picture of V2801F, TWCGPON657 and DFP-34X-C2C

I got DFP-34X-C2C with newer SFP casing
[image]

Compared with TWCGPON657, nearly all similar:
[image]

I afraid to try 2.5GbE on V2801F since I dont have NAND Programmer

What I want, make DFP-34X-C2C able to bridge like V2801F, not sure how to do it...

someone just replace `europa` related driver and work like this:
https://github.com/Anime4000/RTL9601C1/issues/2#issuecomment-1071290468

stich86, 2022-03-24:
DFP-34X-C2C doesn't use europa like UF-Instant :)
we can try to swap just rootFS on the DFP (so it will use omci_app and all other stuff from V2801F)

i've to fix the Image_A on my V2801 and then try to flash using WebUI.. if something get wrong i need to desolder e reflash the NAND ^_^"

Anime4000, 2022-03-24:
> i think 2.5GbE is another issue.. At the moment to get that sync (in HiSGMII mode) you need particular hardware, like MacchiatoBIN, Unifi Switch or BCM Card with patched driver

Yea,
MacchiatoBIN too expansive, so I buy that TP-Link just for 99 USD

> let's try to see if V2801F can boot with DFP-34X-C2C kernel and V2801F rootfs, that should solve both problem :)

this in my main priority!

Since we going to use original uImage, replace `runlansds.sh` with ODI, this way we can use 2.5GbE

---
UF-Instant V2801F modified

[image]

other files is same!

Anime4000, 2022-03-24:
I have modified firmware for TWCGPON657

* Using original `uImage` (B15)
* Using V2801F `rootfs` (V1.9.0-201104)
   * Using modified `Замена.rar`
   * Using original `runlansds.sh`

I also fix various stuff like, update webgui using latest HTML5 standard

[image]

stich86, 2022-03-24:
where did you get the latest WebUI?

Anime4000, 2022-03-24:
> where did you get the latest WebUI?

That I modified by self 😄 

# Here internal of TWCGPON657
[image]
[image]
[image]
[image]
[image]

I open it because...

* I was trying to flash my modified `rootfs`, I forgot to set `LAN_SDS_MODE 1` before flash
* V2801F always force `LAN_SDS_MODE 1`, so new script `runlansds.sh` cause it

# Internal of DFP-34X-C2C
[image]
[image]
[image]

Either wait TP-Link TL-XDR5480 to fix or read-write NAND Flash via programmer? how to do that?

stich86, 2022-03-24:
My DFP-34X-C2C looks like to your TWCGP.
I don't understand what you have done.

I've restore md my V2810F desoldering NAND (using heat gun) and a CH341a programmer. Essentials I'm done a dump and changed the "sw_commited" on uBoot to load the other image that doesn't use LAN_SDS_MODE.

DFP and TECGP use SOIP8 format that should be more easy to read with a clip without soldering. Because you may be will read the flash soldered, it necessary to power SFP from outside source and avoid powering the NAND. This is beacause if you try to power NAND from CH341a, the current flow pass to the PCB and tries to bring up all the stick

Anime4000, 2022-03-25:
Oh, I understand

I have TWCGPON657 `bin` file, it should look similar?

```
anime4000@hitoha-vm:~/Desktop/xpon/B15 Image Dump$ binwalk vm.img 

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
64            0x40            uImage header, header size: 64 bytes, header CRC: 0xB0850016, created: 2020-12-22 07:03:32, image size: 1198344 bytes, Data Address: 0x80000000, Entry Point: 0x80000000, data CRC: 0x6B27D9E8, OS: Linux, CPU: MIPS, image type: OS Kernel Image, compression type: lzma, image name: "Linux Kernel Image"
128           0x80            LZMA compressed data, properties: 0x5D, dictionary size: 16777216 bytes, uncompressed size: 3924124 bytes
1198536       0x1249C8        Squashfs filesystem, little endian, version 4.0, compression:lzma, size: 2232088 bytes, 899 inodes, blocksize: 131072 bytes, created: 2020-12-22 07:03:31
```

stich86, 2022-03-25:
this is a binwalk of NAND dump:

```

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
104896        0x199C0         uImage header, header size: 64 bytes, header CRC: 0x4739E651, created: 2018-11-28 17:36:07, image size: 69741 bytes, Data Address: 0x81C00000, Entry Point: 0x81C00000, data CRC: 0xE36EFC01, OS: Firmware, CPU: MIPS, image type: Firmware Image, compression type: lzma, image name: "U-Boot 2011.12.NA for rtl9601C"
104960        0x19A00         LZMA compressed data, properties: 0x5D, dictionary size: 16777216 bytes, uncompressed size: 181840 bytes
278528        0x44000         JFFS2 filesystem, big endian
438592        0x6B140         Zlib compressed data, compressed
439572        0x6B514         JFFS2 filesystem, big endian
499792        0x7A050         Zlib compressed data, compressed
500512        0x7A320         Zlib compressed data, compressed
501256        0x7A608         JFFS2 filesystem, big endian
502328        0x7AA38         Zlib compressed data, compressed
503464        0x7AEA8         Zlib compressed data, compressed
503808        0x7B000         JFFS2 filesystem, big endian
507984        0x7C050         Zlib compressed data, compressed
508336        0x7C1B0         Zlib compressed data, compressed
509472        0x7C620         Zlib compressed data, compressed
510224        0x7C910         JFFS2 filesystem, big endian
510352        0x7C990         Zlib compressed data, compressed
511304        0x7CD48         Zlib compressed data, compressed
512080        0x7D050         Zlib compressed data, compressed
512652        0x7D28C         JFFS2 filesystem, big endian
513240        0x7D4D8         Zlib compressed data, compressed
514320        0x7D910         Zlib compressed data, compressed
515140        0x7DC44         Zlib compressed data, compressed
516176        0x7E050         JFFS2 filesystem, big endian
516640        0x7E220         Zlib compressed data, compressed
517484        0x7E56C         JFFS2 filesystem, big endian
517604        0x7E5E4         Zlib compressed data, compressed
518684        0x7EA1C         Zlib compressed data, compressed
519896        0x7EED8         Zlib compressed data, compressed
520192        0x7F000         JFFS2 filesystem, big endian
524288        0x80000         uImage header, header size: 64 bytes, header CRC: 0xEE89A0DD, created: 2020-11-04 01:41:33, image size: 1199462 bytes, Data Address: 0x80000000, Entry Point: 0x80000000, data CRC: 0x4DEA4EF6, OS: Linux, CPU: MIPS, image type: OS Kernel Image, compression type: lzma, image name: "Linux Kernel Image"
524352        0x80040         LZMA compressed data, properties: 0x5D, dictionary size: 16777216 bytes, uncompressed size: 3923728 bytes
1966080       0x1E0000        Squashfs filesystem, little endian, version 4.0, compression:lzma, size: 1880366 bytes, 623 inodes, blocksize: 131072 bytes, created: 2020-11-04 01:41:32
4456448       0x440000        uImage header, header size: 64 bytes, header CRC: 0xEE89A0DD, created: 2020-11-04 01:41:33, image size: 1199462 bytes, Data Address: 0x80000000, Entry Point: 0x80000000, data CRC: 0x4DEA4EF6, OS: Linux, CPU: MIPS, image type: OS Kernel Image, compression type: lzma, image name: "Linux Kernel Image"
4456512       0x440040        LZMA compressed data, properties: 0x5D, dictionary size: 16777216 bytes, uncompressed size: 3923728 bytes
5898240       0x5A0000        Squashfs filesystem, little endian, version 4.0, compression:lzma, size: 1876094 bytes, 623 inodes, blocksize: 131072 bytes, created: 2022-03-01 08:46:13
```

I can share my NAND dump (of V2801F) so you can try to bring your back. My dump is in PON_MODE 3, so the stick should no enter bootloop for VS_AUTHKEY

Here is the dump that you can write with CH341a programmer

[fix_commit_1.zip](https://github.com/Anime4000/RTL9601C1/files/8348759/fix_commit_1.zip)

MD5 of extracted file is: 20876fe2af0242a1bbc7bed123d62a33

Anime4000, 2022-03-25:
Thanks! this help a lot!

I notice it has 3 uImage and 2 rootfs, based on k0, k1, r0, r1

Which Block/JFFS2 that store config file?

manias21, 2022-03-25:
this helps, bin
https://mega.nz/file/jC4mlKDR#1PPGZsDjuuLIKgIlPCXo7NGu44aCj2RXXPkBJOr1v0Y

stich86, 2022-03-25:
> this helps, bin https://mega.nz/file/jC4mlKDR#1PPGZsDjuuLIKgIlPCXo7NGu44aCj2RXXPkBJOr1v0Y

from which stick? It's a 16Mbyte dump, RTL9601C has only 8Mbyte of NAND O_o

manias21, 2022-03-25:
[image]
8mb and 16mb

manias21, 2022-03-25:
I put the V2801F_firmware-for-TWCGPON657.zip, but I have lost access, I have tried 192.168.88.50, 192.168.1.1, 192.168.1.10 and I have no response. any ideas ?

Anime4000, 2022-03-25:
> this helps, bin https://mega.nz/file/jC4mlKDR#1PPGZsDjuuLIKgIlPCXo7NGu44aCj2RXXPkBJOr1v0Y

wow... Thanks!
I extract last part `rootfs` and saw `C00R657V00B16`
This mean for TWCGPON657 right? have they fix 4-port Emulation in `C00R657V00B16` like V2801F ?

> > this helps, bin https://mega.nz/file/jC4mlKDR#1PPGZsDjuuLIKgIlPCXo7NGu44aCj2RXXPkBJOr1v0Y
> 
> from which stick? It's a 16Mbyte dump, RTL9601C has only 8Mbyte of NAND O_o

I just notice TWCGPON657 use 128Mb Flash

> I put the V2801F_firmware-for-TWCGPON657.zip, but I have lost access, I have tried 192.168.88.50, 192.168.1.1, 192.168.1.10 and I have no response. any ideas ?

Oh, this mean TWCGPON657 `uImage` (kernel, driver) and V2801F `rootfs` are not compatible... 😢

[307 more replies]