Anime4000/RTL960x #18: [question] vlan mapping all wrong

crrodriguez, 2022-03-17:
First of all the gpon stick is carlitosxxpro RTL9601C1 , OLT on the ISP side reports itself as ALCL (Alcatel lucent) and I got it working except I was never able to figure a few things, that im looking for feedback if anyone else has seen.

ISP is supposed to provide internet on VLAN 2,  IPTV on vlan N and VOIP on vlan Y (neither matter much because I am not subscribed to tv or phone)

Howver, in reality and after quite some pain I ended sniffing on the fiber to find out that the internet connection comes through vlan 2496, and no matter what, Im never getting the magic mapping right (the outer vlan mapped to the inner vlan as far as I understand) if there is any change in the mappings in the ISP side I'll be screwed again.

I am supposed to recieve the correct tagging information from the OLT and at least one person I have talked to that had a SFP stick that did this without problem, facing at least the same brand of OLT.

I have tried the vlan tagging in default, and manual transparent mode, nothing seems to help,  carlitosxxpro support did not help either.

crrodriguez, 2022-03-17:
Im a supposed to get OMCI ME 42 and 43 in other SFP sticks called VLAN_TREATMENT_TABLE , VLAN_RULE_TABLE but can't find it in the stick nor in the crap device my ISP gave me..

manias21, 2022-03-17:
With carlitosv2, the same thing happens to me, technical support did not get results.
But let's focus the case on the vlans, in my case it is vlan6+pppoe ( inet ), vlan3+dhcp (voip), I have only managed to get use of vlan3, in vlan6 it was impossible. I am stuck without a solution.

crrodriguez, 2022-03-17:
> With carlitosv2, the same thing happens to me, technical support did not get results. But let's focus the case on the vlans, in my case it is vlan6+pppoe ( inet ), vlan3+dhcp (voip), I have only managed to get use of vlan3, in vlan6 it was impossible. I am stuck without a solution.

in Your case what's the output of 
#diag
l2-table get entry address valid

vuducdong, 2022-03-18:
> First of all the gpon stick is carlitosxxpro RTL9601C1 , OLT on the ISP side reports itself as ALCL (Alcatel lucent) and I got it working except I was never able to figure a few things, that im [...]

That's vlan stacking (Q-inQ), if your sfp support auto vlan and remote access mode, you can check it yourself without having to manually search or ask your ISP.

[image]

As picture above: change vlan mode to remote access mode, and then, change back to vlan auto, keep pressing refresh button, you will see vlan Q-inQ. Finally, you return to vlan remote access mode, on your router will use the vlan found above.

crrodriguez, 2022-03-18:
The problem I have your suggestion @vuducdong I was never able to find that part of the UI, it doesn't appear to be included in any visible or hidden page. I can't count on the ISP, they flat out deny providing any information or help on anything related to this.

Anime4000, 2022-03-18:
@vuducdong 
that RTL9601B (Hisense LTE3415-SCA+) has stable firmware SDK, for RTL9601C1 firmware like V2801F, TWCGPON657 have broken `VLAN Settings`, it didn't show VLAN like that:
[image]

@crrodriguez 
I believe carlitosxxpro have similar firmware with V2801F.

My Fiber Vendor using n-Port Mapping, where my Internet come at port LAN 2 of Universal Huawei ONU

V2801F manage to map VLAN 1:1

see my `l2-table get entry address valid` [diag here](https://github.com/Anime4000/RTL9601C1/blob/main/Docs/DIAG.md#l2-table)

However... I didn't get Internet when using TWCGPON657 firmware, I believe problem is same like yours

---

Currently I waiting DFP-34X-2C2 to arrive, this new ODI Stick using RTL9601C1
[image]

crrodriguez, 2022-03-18:
@Anime4000 50 bucks is a very reasonable price compared to this stick that cost me arm and leg to import from spain..so please share your experiences with it.

crrodriguez, 2022-03-18:
from the omci logs I have stored I have a few operations returning error, almost certainly related to this problem

```
0x0c27 CircuitPack
Set(0x0101,AdminState=0x00,BridgedorIPInd=0x00,CardCfg=0x00,PowerShed=0x00000000)
0x0c27 CircuitPack SetRsp(0x0101,Result=0x5)
0x0c28 EthUni Set(k,0x0101,AutoDectectCfg=0x03,AdminState=0x00,MaxFrameSize=0x07d0,ARC=0x00)
0x0c28 EthUni SetRsp(0x0101,Result=0x5)
```

Above provisioning EthUni 0x0101 fails..  and most likely all the following operations on it do not complete sucessfully, but omci_fake_ok is set to 1; note that the stick advertises

```
0x0000 EthUni AVC(0x0401)
0x0000 EthUni AVC(0x0401)
0x0000 EthUni AVC(0x0401)
0x0000 EthUni AVC(0x0401)
```
so the OLT wants to configure a different explicit port? 

```
0x0c29 GalEthProf Create(0x0001,MaxGemPayloadSize=0x0030)
0x0c29 GalEthProf CreateRsp(0x0001,Result=0x0)
0x0c2a MacBriServProf
Create(0x0101,SpanTreeInd=0x00,LearningInd=0x00,AtmBriInd=0x00,Priority=0x8000,MaxAge=0x1400,HelloTime=0x0200,ForwardDelay=0x0f00,DiscardUnknow=0x00,MacLearningDepth=0x00,DynamicFilteringAgeingTime=0x00000000)
0x0c2a MacBriServProf CreateRsp(0x0101,Result=0x0)

0x0c2b MacBriPortCfgData
Create(0x0101,BridgeIdPtr=0x0101,PortNum=0x01,TPType=0x01,TPPointer=0x0101,PortPriority=0x0000,PortPathCost=0x0001,PortSpanTreeInd=0x00,EncapMethod=0x00,LanFCSInd=0x00,NumOfAllowedMac=0x00)
0x0c2b MacBriPortCfgData CreateRsp(0x0101,Result=0x0)
0x0c2c ExtVlanTagOperCfgData
Create(0x0101,AssociationType=0x02,AssociatedMePoint=0x0101)
0x0c2c ExtVlanTagOperCfgData CreateRsp(0x0101,Result=0x0)
0x0c2d ExtVlanTagOperCfgData
Set(0x0101,InputTPID=0x8100,OutputTPID=0x8100,DsMode=0x00,DscpToPbitMapping=0x0000002492494924926db6db924924b6db6ddb6db6ffffff)
0x0c2d ExtVlanTagOperCfgData SetRsp(0x0101,Result=0x0)
0x0c2e McastOperProf
Create(0x0101,IGMPVersion=0x02,IGMPFun=0x00,ImmediateLeave=0x01,UsIGMPTci=0x0000,UsIGMPTagControl=0x00,UsIGMPRate=0x00000010,Robustness=0x02,QuerierIpAddress=0x00000000,QueryInterval=0x00000040,QueryMaxResponseTime=0x00000064,DownstreamIgmpMulticastTci=0x000000)
0x0c2e McastOperProf CreateRsp(0x0101,Result=0x0)
0x0c2f McastOperProf
Set(0x0101,UsIGMPRate=0x00000010,QueryInterval=0x00000040,UnauthorizedJoinRequestBehaviour=0x01)
0x0c2f McastOperProf SetRsp(0x0101,Result=0x0)
0x0c30 McastSubConfInfo
Create(0x0101,MeType=0x00,McastOperProfPtr=0x0101,MaxSimGroups=0x0000,MaxMcastBandwidth=0x00000000,BandwidthEnforcement=0x01)
0x0c30 McastSubConfInfo CreateRsp(0x0101,Result=0x0)
0x0c31 ExtVlanTagOperCfgData
Set(0x0101,ReceivedFrameVlanTaggingOperTable=0xf8000000e8005000c00f0000000f8002)
0x0c31 ExtVlanTagOperCfgData SetRsp(0x0101,Result=0x0)
0x0c32 ExtVlanTagOperCfgData
Set(0x0101,ReceivedFrameVlanTaggingOperTable=0xe8005000e8000000c00f0000000f8003)
0x0c32 ExtVlanTagOperCfgData SetRsp(0x0101,Result=0x0)
0x0c33 ExtVlanTagOperCfgData
Set(0x0101,ReceivedFrameVlanTaggingOperTable=0xf8000000f0005000c00f0000000f0002)
0x0c33 ExtVlanTagOperCfgData SetRsp(0x0101,Result=0x0)
```

Anime4000, 2022-03-18:
@crrodriguez 
have you tried this? turn into Transparent Mode
```
flash set VLAN_CFG_TYPE 1
flash set VLAN_MANU_MODE 0
```

if your ISP using Q-in-Q, you may need untag VLAN twice...

crrodriguez, 2022-03-18:
I got this 

```
2-table get entry address valid

LUT address: 0x000c (1K LUT)
L2 Unicast table:
MACAddress        Spa Fid Age Vid  State  Ext  Hash
00:00:00:01:00:02 3   0   1   0    Static 0    SVL
CtagIf Auth DaBlock SaBlock Arp
Dis    Dis  Dis     Dis     Dis

LUT address: 0x012c (1K LUT)
L2 Unicast table:
MACAddress        Spa Fid Age Vid  State  Ext  Hash
00:0E:C6:5C:3C:E3 0   0   7   0    Auto   0    SVL
CtagIf Auth DaBlock SaBlock Arp
Dis    Dis  Dis     Dis     Dis

LUT address: 0x016c (1K LUT)
L2 Unicast table:
MACAddress        Spa Fid Age Vid  State  Ext  Hash
00:E0:4C:86:70:01 3   0   1   0    Static 0    SVL
CtagIf Auth DaBlock SaBlock Arp
Dis    Dis  Dis     Dis     Dis

LUT address: 0x01b0 (1K LUT)
L2 Unicast table:
MACAddress        Spa Fid Age Vid  State  Ext  Hash

--More--
C8:8D:83:92:C0:F8 2   0   7   3980 Auto   0    SVL
CtagIf Auth DaBlock SaBlock Arp
En     Dis  Dis     Dis     Dis

RTK.0> l2-table get vid-unmatch port all

Port 0 Action: Forward
Port 2 Action: Forward

 gpon show us-flow     
============================================================
    GPON ONU MAC U/S Flow Status
Flow ID | GEM Port | Type | TCont
      0 |      313 |  ETH |     0
     64 |       33 | OMCI |    16
============================================================

pon show ds-flow                                           
============================================================
     GPON ONU MAC D/S Flow Status
Flow ID | GEM Port | Type | Multicast | AES
      0 |     2047 |  ETH |         * |    
      1 |      313 |  ETH |           |    
     64 |       33 | OMCI |           |    
============================================================

port: 0  rate:1048568
         queue: 0  apr-index: 0
         queue: 1  apr-index: 0
         queue: 2  apr-index: 0
         queue: 3  apr-index: 0
         queue: 4  apr-index: 0
         queue: 5  apr-index: 0
         queue: 6  apr-index: 0
         queue: 7  apr-index: 0
port: 2  rate:4194296
         queue: 0  apr-index: 0
         queue: 1  apr-index: 0
         queue: 2  apr-index: 0
         queue: 3  apr-index: 0
         queue: 4  apr-index: 0
         queue: 5  apr-index: 0
         queue: 6  apr-index: 0
         queue: 7  apr-index: 0
```

It is not making much sense to me, whatever the mode,, internet still on vlan 2496. gpon stick has br0, eth0, eth0.2 eth0.3 and claims setting up port 1 failed to set PVID .

crrodriguez, 2022-03-18:
You maybe wondering how I got this vlan 2496 number, well.. 

1. Connect the ISP supplied ONU to the internet
2. have the gpon stick booted ready and connected to a media converter, wireshark already capturing on the ethernet side of the converter device.
3.  as fast as you can change the fiber from isp ONU to your SFP stick,
4. If you lucky stars are with you, you will get a pppoe termination packet from the ISP concentrator as the old connection from the isp onu times out, all carring the correct vlan tag.

stich86, 2022-03-20:
@crrodriguez how do you produce those debug logs?
i need to debug my O3 status on DFP-34X-2C2

My provider doesn't need LoID auth, but looks like the stick is trying to do it :(

crrodriguez, 2022-03-20:
> @crrodriguez how do you produce those debug logs? i need to debug my O3 status on DFP-34X-2C2
> 
> My provider doesn't need LoID auth, but looks like the stick is trying to do it :(

OMCI debug is enabled by 

```
# flash set OMCI_DBGLVL 2
# flash set OMCI_LOGFILE_MASK 2
# reboot
```
on the next boot, raw OMCI talk will be stored /tmp/omcilog , a human readable version at /tmp/omcilog.par

you may read /proc/ksmg for kernel logs and other info, you can also poke files in /proc/omci .. some will appear to generate no output.. those its contents to /proc/kmsg because manufacturer didn't bother writting reasonable standard drivers that just output to the stdout. :-(

stich86, 2022-03-20:
I've understood you have the CarlitoX version. Which firmware is running? Looking at the webui it seems based on ODI firmware.

In case can you share your MTD dump? Thx :)

manias21, 2022-03-20:
> in Your case what's the output of #diag l2-table get entry address valid

[https://ibb.co/hZSpX4Z](url)

manias21, 2022-03-21:
l2-table get entry address valid

LUT address: 0x000c (1K LUT)
L2 Unicast table:
MACAddress        Spa Fid Age Vid  State  Ext  Hash
00:00:00:01:00:02 3   0   1   0    Static 0    SVL
CtagIf Auth DaBlock SaBlock Arp
Dis    Dis  Dis     Dis     Dis

LUT address: 0x0058 (1K LUT)
L2 Unicast table:
MACAddress        Spa Fid Age Vid  State  Ext  Hash
32:B6:4F:3E:A2:41 2   0   7   182  Auto   0    SVL
CtagIf Auth DaBlock SaBlock Arp
En     Dis  Dis     Dis     Dis

LUT address: 0x0204 (1K LUT)
L2 Unicast table:
MACAddress        Spa Fid Age Vid  State  Ext  Hash
38:3A:21:27:A2:27 3   0   1   0    Static 0    SVL
CtagIf Auth DaBlock SaBlock Arp
Dis    Dis  Dis     Dis     Dis

LUT address: 0x0320 (1K LUT)
L2 Unicast table:
MACAddress        Spa Fid Age Vid  State  Ext  Hash
42:71:83:AA:40:92 2   0   7   180  Auto   0    SVL
CtagIf Auth DaBlock SaBlock Arp
En     Dis  Dis     Dis     Dis

LUT address: 0x039c (1K LUT)
L2 Unicast table:
MACAddress        Spa Fid Age Vid  State  Ext  Hash
E4:8D:8C:08:6D:67 0   0   7   0    Auto   0    SVL
CtagIf Auth DaBlock SaBlock Arp
Dis    Dis  Dis     Dis     Dis

-------------------------
l2-table get vid-unmatch port all

Port 0 Action: Forward
Port 2 Action: Forward

--------------------------
gpon show us-flow
============================================================
    GPON ONU MAC U/S Flow Status
Flow ID | GEM Port | Type | TCont
      0 |      364 |  ETH |     0
      1 |      365 |  ETH |     1
      2 |      366 |  ETH |     2
     64 |       40 | OMCI |    16
============================================================

----------------------------
gpon show ds-flow
============================================================
     GPON ONU MAC D/S Flow Status
Flow ID | GEM Port | Type | Multicast | AES
      0 |     2047 |  ETH |         * |
      1 |      364 |  ETH |           |   *
      2 |      365 |  ETH |           |   *
      3 |      366 |  ETH |           |   *
     64 |       40 | OMCI |           |   *
============================================================
-----------------------------

ami shows me this, I get a 05 because I assigned the Ploam data but I don't connect with vlan6+pppoe or vlan3+dhcp, any idea where or what is needed?

crrodriguez, 2022-03-21:
> 

There is vlan 180 and 182 in your output. see which one it is.

manias21, 2022-03-21:
I have mounted two, vlan180 + 182
 and two ppoe, without results..

https://ibb.co/2v2bY3j

Anime4000, 2022-03-24:
@crrodriguez 
I received [DFP-34X-C2C](https://www.aliexpress.com/item/4000086970977.html?spm=a2g0o.order_list.0.0.21ef1802udltDM)

after run same `flash set` like V2801F, this stick dont have 4-port LAN Emulation, I cannot bridge my Internet, only ISP 1 `tr069` VLAN able to...

my Fiber has multiple ISP, **I using ISP 2**:
**Fiber Vendor VLAN:**
| VLAN | Desc |
|-------|----------------------|
| `209` | ISP 1 `tr069` |
| `400` | ISP 1 VoIP |
| `500` | ISP 1 Internet |
| `600` | ISP 1 IPTV |
| `621` | ISP 2 Internet |
| `821` | ISP 2 `tr069` |
| `822` | ISP 2 VoIP |
| `823` | ISP 2 IPTV |

[image]
[image]

as you can see, only ISP 1 `tr069` vlan manage to have IP Address

Anime4000, 2022-03-28:
[DFP-34X-C2C](https://www.aliexpress.com/item/4000086970977.html?spm=a2g0o.order_list.0.0.21ef1802udltDM) using RTL9601D, TTL-UART pin is available

Anime4000, 2022-03-30:
channel mode is `BRIDGE` then tick `LAN_1`

skon77, 2022-03-30:
Hmm, when I select bridge I have no LAN_1 mapping

[image]

Anime4000, 2022-03-30:
no `LAN_1` like this?
[image]

skon77, 2022-03-30:
Sorry, didn't specify the stick.
I took your advice and chose ODI stick as my main
There is no Lan_1 in that link.

Anime4000, 2022-03-30:
that very odd, my ODI stick do have it

Anime4000, 2022-03-30:
currently I trying to port V2801F `rootfs` into ODI stick

skon77, 2022-03-30:
That would be great.
Right now my odi stick has Firmware Version V1.0-210702 and no Lan1 mapping...

skon77, 2022-04-11:
> no `LAN_1` like this?

Hi,
I updated my TWCGPON657 firmware to version C00R657V2801F_V1.9.0-220404. Thank you for nice design.

Hidden link "/multi_wan_generic.asp" still working?

In my case there are always two blocks of "PPP Settings" and "WAN IP Settings". 
[image]

To make them disappear I have to click the "Channel Mode" to "Bridged/IpoE" settings, but when I return to "Bridged" and set "Enable VLAN" I get an error after apply changes - Invalid IP/Subnet Mask combination!!!
When you roll back the firmware to version V2801F_V1.9.0-201104, the problem persists and does not go away. Do I have to do a factory reset after reflashing?

Anime4000, 2022-04-12:
> Thank you for nice WebGUI.
Thanks! Making SFP Stick WebGUI follow web standard, clean look across browsers...

it seem you need reset factory...

V2801F & ODI V1.0-`220304` is in full bridge mode...

# Factory Reset via Telnet
1. `flash set PON_MODE 3` *prevent invalid key*
2. `flash get ELAN_MAC_ADDR`, `VS_AUTH_KEY`, `HW_HWVER` *backup*
3. `flash reset cs`
4. `flash reset hs`

then `flash set ...` to original value...

# Wrong VLAN fix
[image]

## Do this
> - [x] Manual
> - [x] Transparent
*this will map VLAN as is `1:1 mapping`*

```
flash set LAN_SDS_MODE 1
```

If `LAN_SDS_MODE` not fix VLAN mapping, try `2` or `3`

crrodriguez, 2022-04-12:
LAN_SDS_MODE defaukts to 5 on carlitoxxpro v2.  do this chip use the same meaning as the table you have documented ? If I understand correctly, using thew SFP module lan facing port in PHY mode instead of MAC may fix vlan mapping ?

Anime4000, 2022-04-12:
> LAN_SDS_MODE defaukts to 5 on carlitoxxpro v2. do this chip use the same meaning as the table you have documented ? If I understand correctly, using thew SFP module lan facing port in PHY mode instead of MAC may fix vlan mapping ?

[Russian forum tell](http://forum.ru-board.com/topic.cgi?forum=8&topic=80480&start=1640#3), `LAN_SDS_MODE 1` should bring 1:1 VLAN Mapping

still, you need set VLAN
- [x] Manual
- [x] Transparent

using Auto will cause auto mapping when ONT didn't use Upstream OMCI

Transparent should copy Downstream OMCI VLAN filter,

to see OMCI VLAN Filter, type this on telnet: `omcicli mib get 84`, it will print like this, my OMCI:
```
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
VlanTagFilterData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityID: 0x1202
FilterTbl[0]: PRI 0,CFI 0, VID 621
FwdOp:  0x10
NumOfEntries: 1
=================================
=================================
EntityID: 0x1203
FilterTbl[0]: PRI 0,CFI 0, VID 821
FwdOp:  0x10
NumOfEntries: 1
=================================
=================================
EntityID: 0x1204
FilterTbl[0]: PRI 0,CFI 0, VID 822
FwdOp:  0x10
NumOfEntries: 1
=================================
=================================
EntityID: 0x120b
FilterTbl[0]: PRI 0,CFI 0, VID 621
FilterTbl[2]: PRI 0,CFI 0, VID 821
FilterTbl[4]: PRI 0,CFI 0, VID 822
FwdOp:  0x10
NumOfEntries: 3
=================================
=================================
EntityID: 0xe102
FilterTbl[0]: PRI 0,CFI 0, VID 209
FwdOp:  0x10
NumOfEntries: 1
=================================
=================================
EntityID: 0xe10b
FilterTbl[0]: PRI 0,CFI 0, VID 209
FwdOp:  0x10
NumOfEntries: 1
=================================
```

tdmadam, 2022-04-12:
> it seem you need reset factory...
>
> V2801F & ODI V1.0-`220304` is in full bridge mode...
> # Factory Reset via Telnet
>
>     1. `flash set PON_MODE 3` _prevent invalid key_
>
>     2. `flash get [...]

Thanks! This is exactly what I was looking for. I need to reset my stick!

btw. 
For firmware V2801F_V1.9.0-201104 the commands 3 and 4 look like this
```

# flash default cs
# flash default hs
```

crrodriguez, 2022-04-13:
no luck with carlitopro v2, changing LAN_SDS_MODE  makes no difference,  while it defaults to 5.. even if you change it to whatever you want, the device always boots in mode (1 FIBER 1G) omcicli app produces absolutely no output ever.

Anime4000, 2022-04-13:
this mean you need replace lansds.sh file

stich86, 2022-04-13:
> no luck with carlitopro v2, changing LAN_SDS_MODE makes no difference, while it defaults to 5.. even if you change it to whatever you want, the device always boots in mode (1 FIBER 1G) omcicli app produces absolutely no output ever.

as @Anime4000 has said, if the firmware is the one used on V2801F, the `runlansds.sh` has fixed value of `1` and doesn't read the `LAN_SDS_MODE` from xml file

Anime4000, 2022-04-17:
There is noting wrong with VLAN mapping...

ODI said just your OLT provide OMCI VLAN Info differently for `pon0` (inner) and `eth0` (Outer),

it seem your OLT trying to hide their PON VLAN

In Router Mode: You can map outer vlan via **PON WAN** page.
In Bridge Mode: Stick map inner vlan to outer as is (no change).
[image]

# Find VLAN
When use Bridge Mode firmware, enter this command on telnet:
```
omcicli mib get 84
```

# In conclusion...
Bridge Mode firmware, stick will using `pon` (inner) VLAN and map as is!

skon77, 2022-04-20:
@Anime4000 , @stich86 

Would you kindly explain to me the reason for the different data output?

ODI (FW 220407)
 
<details>
<summary>l2-table get entry address valid</summary>

LUT address: 0x000c (1K LUT)
L2 Unicast table:
MACAddress Spa Fid Age Vid State Ext Hash
XX:XX:XX:XX:XX:XX 3 0 1 0 Static 0 SVL
CtagIf Auth DaBlock SaBlock Arp
Dis Dis Dis Dis Dis

LUT address: 0x0140 (1K LUT)
L2 Unicast table:
MACAddress Spa Fid Age Vid State Ext Hash
XX:XX:XX:XX:XX:XX 0 0 7 0 Auto 0 SVL
CtagIf Auth DaBlock SaBlock Arp
Dis Dis Dis Dis Dis

LUT address: 0x016c (1K LUT)
L2 Unicast table:
MACAddress Spa Fid Age Vid State Ext Hash
XX:XX:XX:XX:XX:XX 3 0 1 0 Static 0 SVL
CtagIf Auth DaBlock SaBlock Arp
Dis Dis Dis Dis Dis

LUT address: 0x01d0 (1K LUT)
L2 Unicast table:
MACAddress Spa Fid Age Vid State Ext Hash
XX:XX:XX:XX:XX:XX 2 0 7 30 Auto 0 SVL
CtagIf Auth DaBlock SaBlock Arp
En Dis Dis Dis Dis

LUT address: 0x0260 (1K LUT)
L2 Unicast table:
MACAddress Spa Fid Age Vid State Ext Hash
XX:XX:XX:XX:XX:XX 2 0 6 30 Auto 0 SVL
CtagIf Auth DaBlock SaBlock Arp
En Dis Dis Dis Dis

LUT address: 0x02d8 (1K LUT)
L2 Unicast table:
MACAddress Spa Fid Age Vid State Ext Hash
XX:XX:XX:XX:XX:XX 2 0 7 9 Auto 0 SVL
CtagIf Auth DaBlock SaBlock Arp
En Dis Dis Dis Dis
</details>

<details>
<summary>omcicli mib get 84</summary>

VlanTagFilterData

EntityID: 0x03
FilterTbl[0]: PRI 0,CFI 0, VID 10
FwdOp:  0x10
NumOfEntries: 1

EntityID: 0x04
FilterTbl[0]: PRI 0,CFI 0, VID 20
FwdOp:  0x10
NumOfEntries: 1

EntityID: 0x05
FilterTbl[0]: PRI 0,CFI 0, VID 30
FilterTbl[2]: PRI 0,CFI 0, VID 32
FwdOp:  0x10
NumOfEntries: 2

EntityID: 0x06
FilterTbl[0]: PRI 0,CFI 0, VID 40
FwdOp:  0x10
NumOfEntries: 1

EntityID: 0x07
FilterTbl[0]: PRI 0,CFI 0, VID 50
FwdOp:  0x10
NumOfEntries: 1

EntityID: 0x08
FilterTbl[0]: PRI 0,CFI 0, VID 60
FwdOp:  0x10
NumOfEntries: 1
</details>

<details>
<summary>omcicli mib get 171</summary>

ExtVlanTagOperCfgData

EntityId: 0x01
AssociationType: 2
ReceivedFrameVlanTagOperTableMaxSize: 0
InputTPID: 0x8100
OutputTPID: 0x8100
DsMode: 0
ReceivedFrameVlanTaggingOperTable
INDEX 0
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 15,VID 4096, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 4096, TPID 0, RemoveTags 0
Treatment Inner   : PRI 0,VID 30, TPID 4
INDEX 1
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 8,VID 60, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 1
Treatment Inner   : PRI 8,VID 60, TPID 0
INDEX 2
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 8,VID 32, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 1
Treatment Inner   : PRI 8,VID 32, TPID 0
INDEX 3
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 8,VID 30, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 1
Treatment Inner   : PRI 8,VID 30, TPID 0
INDEX 4
Filter Outer   : PRI 14,VID 4096, TPID 0
Filter Inner   : PRI 14,VID 4096, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3
Treatment Inner   : PRI 15,VID 0, TPID 0
INDEX 5
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 14,VID 4096, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3
Treatment Inner   : PRI 15,VID 0, TPID 0
AssociatedMePoint: 0x101
DscpToPbitMapping:
        0x000000
        0x000000
        0x000000
        0x000000
        0x000000
        0x000000
        0x000000
        0x000000

EntityId: 0x02
AssociationType: 10
ReceivedFrameVlanTagOperTableMaxSize: 0
InputTPID: 0x8100
OutputTPID: 0x8100
DsMode: 0
ReceivedFrameVlanTaggingOperTable
INDEX 0
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 8,VID 10, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 1
Treatment Inner   : PRI 8,VID 10, TPID 0
INDEX 1
Filter Outer   : PRI 14,VID 4096, TPID 0
Filter Inner   : PRI 14,VID 4096, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3
Treatment Inner   : PRI 15,VID 0, TPID 0
INDEX 2
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 14,VID 4096, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3
Treatment Inner   : PRI 15,VID 0, TPID 0
AssociatedMePoint: 0x601
DscpToPbitMapping:
        0x000000
        0x000000
        0x000000
        0x000000
        0x000000
        0x000000
        0x000000
        0x000000
</details>

So what kind of VLANs does the SFP receive at the input?

It is not clear what is wrong with Multicast. As far as I know, my ISP sends Multicast through VLAN 9. And we see this in the l2-table report. However, on SFP interface Multicast goes without any VLAN. 

[image]

On previous stick firmware, I tried to specify VLAN enabled tag number 9 via the VLAN enabled option. The stick wouldn't let me do that, wrote that number 9 and a few others were specially reserved (where and how...). 

Could this then indicate that the latest firmware supports not quite transparent mode and still does untag some tags (reserved...?) ? 

If it's helpful, here are screenshots from the WebUI of my ISP's default router. This is when the fiber optic is connected to it, the Internet is working and TV box shows the channels.

[image]

[image]

stich86, 2022-04-21:
looking at your MIBs, there is no VLAN 9 on OMCI stuff, only 10,20,30,32,40,50,60.

If you don't that, there is an Inner that should be tagged to 30

it's very funny where it get VID 9, because on both MEID 84/171 there is no trace :\

skon77, 2022-04-21:
@stich86 

Thanks!

Which is primary and which is secondary?
l2-table or MEID...
I admit that VLAN 9 is on the input...

But there is also this situation:
[image]

[image]

Could it be that VLAN 9 is just untaget and not in MEID?

My ISP uses 2 brands of OLT. Huawei and ZTE. I am lucky, I have Huawei and these modems have VLANs 10,20,30 (+ maybe 9 Multicast).

But ZTE  use VLANs with numbers up to 10, for example the analogue of my VLAN 30 has number 3 there. I wonder how stick will work with VLANs of this numbering...

tdmadam, 2022-05-13:
> from the omci logs I have stored I have a few operations returning error, almost certainly related to this problem
>
> ```
> 0x0c27 CircuitPack [...]

Have you fixed this bug yet? Some clever guy who likes to hang out on the Turris forum has created a neat script that fixes these errors on the fly.

crrodriguez, 2022-05-13:
> > so the OLT wants to configure a different explicit port?
> 
> Have you fixed this bug yet? Some clever guy who likes to hang out on the Turris forum has created a neat script that fixes these errors on the fly.

No, I have not.. owvere I would b einterested to see what kind of hack is been used to workaround this.

tdmadam, 2022-05-13:
```
#!/bin/sh

############################################################################
# Author _DS_
# The script corrects the MIB entity IDs in the data sent by the OLT.
#
# Make sure the following is configured:
#  flash set OMCI_FAKE_OK 1
#  flash set OMCI_LOGFILE 1
#
# To run the script from omci_app save it as /etc/config/fix.sh and run:
#  chmod a+x /etc/config/fix.sh
#  nv setenv sw_active
#  nv setenv sw_commit ';/etc/config/fix.sh&'
#
############################################################################

WAIT_TIME=5

OMCILOG=/tmp/omcilog
FAKE_PACKETS=/tmp/fake_packets
LOCK=/tmp/fix.lck

ZERO_PAD=$(awk 'BEGIN {while (i++ < 35) printf "00 "}')

alias wc="awk 'END{print NR}'"

alias requests="awk 'NR % 2 == 1 { print; }'"

#fetchEntity (entityClass, entityId)
fetchEntity () {
    cat "$OMCILOG" | grep -i "^\([0-9a-f]\{2\} \)\{4\}$1 $2" | requests
}

#updateEntity (entityClass, entityId, offset, val)
updateEntity () {
    valLen=${#4}
    fetchEntity "$1" "$2" | awk "{ print substr(\$0, 1, $3 - 1) \"$4\" substr(\$0, $3 + $valLen); }"
}

#recreateEntity (entityClass, entityId, offset, val)
recreateEntity () {
    valLen=${#4}
    fetchEntity "$1" "$2" | awk "NR == 1 { print substr(\$0, 1, 6) \"46 \" substr(\$0, 10, 15) \"$ZERO_PAD\" substr(\$0, 130, 15);
            print substr(\$0, 1, $3 - 1) \"$4\" substr(\$0, $3 + $valLen); }
            NR != 1 { print; }"
}

# The script is called on startup and each MIB reset, we want only one instance.
mkdir "$LOCK" || exit

while [ $(fetchEntity "00 ab" "01 01" | wc) -eq 0 ]; do
    sleep 1
done
sleep 1
cntOld=0
cnt=$(fetchEntity "00 ab" "01 01" | wc)
while [ $cnt -ne $cntOld ]; do
    sleep "$WAIT_TIME"
    cntOld=$cnt
    cnt=$(fetchEntity "00 ab" "01 01" | wc)
done

#Circuit Pack
updateEntity "00 06" "01 01" 19 "01 04" > "$FAKE_PACKETS"
#PPTP Ethernet Uni
updateEntity "00 0b" "01 01" 19 "04 01" >> "$FAKE_PACKETS"
#MAC bridge port configuration data
recreateEntity "00 2f" "01 01" 37 "04 01" >> "$FAKE_PACKETS"
#Extended VLAN tagging operation configuration data
recreateEntity "00 ab" "01 01" 28 "04 01" >> "$FAKE_PACKETS"

omcicli debug loadfile "$FAKE_PACKETS"

: > "$OMCILOG"

rm -rf "$LOCK"
```

stich86, 2022-05-13:
> ```
>
> #!/bin/sh
>
>
>
> ############################################################################
>
> # Author _DS_
>
> # The script corrects the MIB entity IDs in the data sent by the OLT.
>
> #
>
> # Make sure [...]

So what this script fix? I see it modify MIB 171..

tdmadam, 2022-05-13:
The issue seems to be specific to sticks based on V2801F. For some MEs: PPTP Ethernet Uni, Circuit Pack, MAC bridge, Extended VLAN they advertise their Managed Entity Instance as 0x0401, 0x0104. Some OLTs insist on 0x0101 and will not agree to any other value which results in an error.

stich86, 2022-05-16:
> The issue seems to be specific to sticks based on V2801F. For some MEs: PPTP Ethernet Uni, Circuit Pack, MAC bridge, Extended VLAN they advertise their Managed Entity Instance as 0x0401, 0x0104. Some OLTs insist on 0x0101 and will not agree to any other value which results in an error.

ok thx for share, doesn't looks like my problem

rajkosto, 2022-12-13:
This script might become useful for ODI, because it's VEIP EntityID is 0x0601 while most OLT expect 0x0E01 there. Maybe this is why ODI firmware ignore mib 171 ? It's set for 0x0E01 entity ids, not 0x0601

und3rd06012, 2023-02-01:
I have the Carlitoxx Pro CPGOS03-0490v2 and I set the GPON SN of my ONT and the password and the module gets to O5 state. The problem is that whenever I try make a pppoe connection from my udm-pro I do not receive an IP from my ISP (my credentials are correct). The pppoe authentication uses VLAN ID 835 but as far as I read here I must set somehow the VLAN for the bridge connection between the udm-pro and gpon onu module. I read somewhere that my isp uses VLAN ID 100 for the ftth connection but I am not sure. I ran omcicli mib get 84 but it outputs nothing. I ran also diag and then l2-table get entry address valid and all the entries had VID 0. What does that mean exactly? Any help is much appreciated.

Anime4000, 2023-02-02:
sell that Carlitoxx Pro CPGOS03-0490v2 and buy ODI stick

because it can't show omcicli mib which is very hard to troubleshooting

und3rd06012, 2023-02-02:
Hi! Thank you for your reply! Is this the one you are referring to: https://www.alibaba.com/product-detail/ODI-ONU-PON-Stick-SC-UPC_1600270356700.html

crrodriguez, 2023-02-02:
I have flashed the VSOl firmware in CarlitoXXpro that should also work for him right ?

und3rd06012, 2023-02-02:
Hi crrodriguez! How did you do that? Do you have a link with the files and the procedure?

crrodriguez, 2023-02-02:
I was successful doing so 
flash set LAN_SDS_MODE 1 
flashed https://github.com/Anime4000/RTL960x/raw/main/Firmware/V2801F/V2801F_V1.9.0-220425.tar
Via web interface..
 Was welcomed by @Anime4000 modded interface I have, however not tested with the fiber plugged.

Note that this procedure for some reason do not work for some people.

xDSx, 2023-11-17:
> sell that Carlitoxx Pro CPGOS03-0490v2 and buy ODI stick
> 
> because it can't show omcicli mib which is very hard to troubleshooting

Small correction: by default it doesn't, but if you kill the omci_app processes and run them yourself, from the telnet session, the results of the omcicli mib commands will become visible.

manias21, 2023-11-17:
No small correction at all, say goodbye to the carlitov2, and buy an ODI, a person who has helped you here tells you to get the sfpgon system working once and for all, up up Anime4000 and friends.

crrodriguez, 2023-11-17:
> Just a small correction, nothing at all, say goodbye to the carlitov2, and buy an ODI, a person tells you this who has helped here to get the sfpgon system working once and for all, up up Anime4000 and friends.

Changing LAN_SDS_MODE to 3 soft bricked mine and since I don't have a debug board it is now a paperweight. ODI working just fine atm.

qatar2030, 2024-02-15:
> I was successful doing so flash set LAN_SDS_MODE 1 flashed https://github.com/Anime4000/RTL960x/raw/main/Firmware/V2801F/V2801F_V1.9.0-220425.tar Via web interface.. Was welcomed by @Anime4000 [...]

I flash my DFP and now I can't connect to it

qatar2030, 2024-02-15:
what is defaultdefault ip for VSOL V2801F

crrodriguez, 2024-02-15:
> I flash my DFP and now I can't connect to it

This firmware is for VSOL stick..and OEM ones like CarlitoPRO v2 stick.,. it is not for the DFP stick.

qatar2030, 2024-02-16:
Is there a solution to this problem that is troubling  me in particular  please?

d5aqoep, 2024-06-08:
Hello @Anime4000 any solution in sight? My ISP does bridge mode on LAN Port 4 and my Vsol V2802RH has just LAN1 and LAN2