Anime4000/RTL960x #20: Custom WEB UI - boa configuration change needed?

stich86, 2022-04-13:
Hi @Anime4000,

i'm open this issue to tracking a problem that i'm currently have on swap the WEB UI of V2801F with yours.
Do you have added this line to `/home/httpd/boa.conf` file?

`AddType text/css css`

Thx in advance

Anime4000, 2022-04-13:
yes, I do edit `boa.conf` for file type

```
AddType application/json map
AddType text/css css
AddType text/javascript js
```

this stop browser screaming `text/plain` on `css` and `js`

no need `map`, this use for troubleshoot

stich86, 2022-04-13:
> yes, I do edit `boa.conf` for file type
> 
> ```
> AddType application/json map
> AddType text/css css
> AddType text/javascript js
> ```
> 
> this stop browser screaming `text/plain` on `css` and `js`
> 
> no need `map`, this use for troubleshoot
> 
> and you need edit `index.html` at [line 22](https://github.com/Anime4000/RTL9601C1/blob/44649012c91d365ceb1d66ce396059e44fdc23f3/WebGui/V2801F/index.html#L22)
> 
> change TWCGPON657 back to V2801F or anything you like

thx, i'll try it tonight :)
i think it will be useful to add this changes on the Web MD :)

Anime4000, 2022-04-13:
> thx, i'll try it tonight :)
> i think it will be useful to add this changes on the Web MD :)

yea, I forgot to tell~ updating now

Anime4000, 2022-04-13:
@stich86 when you done, I want to know your feedback 😄

Anime4000, 2022-04-13:
> > @stich86 when you done, I want to know your feedback 😄
> 
> positive feedback mate! Now the UI is working perfect :)

That's good to hear that 😊

EnfermeraSexy, 2022-04-14:
> @EnfermeraSexy no, it seem original code don't have it.
> 
> Firefox cannot execute when wrong Mime Type
> 
> Chrome can execute even wrong Mime Type
> 
> this is reason original WebGUI look horrible in Firefox browser

You mean that is not used or that it wasn't writed in the file? Because i found the line in the original code

Anime4000, 2022-04-14:
Yes, they do have, I just arrange neatly

johndoe1983, 2024-01-25:
> yes, I do edit `boa.conf` for file type
>
> ```
> AddType application/json map
> AddType text/css css
> AddType text/javascript js
> ```
>
> this stop browser screaming `text/plain` on `css` and `js`
>
> no need [...]

@Anime4000 currently, where I could find the files (index.html, .asp...) for Modern WebUI on ODI DFP-34X-2C2? 

On https://github.com/Anime4000/RTL960x/tree/main/WebGui only appears pngs, psd, ai files, but here there aren't the files for that modern WebUI

Anime4000, 2024-01-25:
the WebGUI is not complete yet, since we can't reverse engineer the binary to show Link Speed, etc like OpenWRT, I planing to approach differently at CPU efficient

still, for modern ODI WebGUI is based off V2801F with correct ASP code, if you have time to improve, can start off your own version based on V2801F, or I can upload incomplete and let anyone improve