Anime4000/RTL960x #22: Stock firmware for XPON 1,25G/2,5G (RTL960x)

mcheev, 2022-04-18:
Hi!
Thank you for your hard work and dedication to the community.
I apologize in advance, but I did not find another way to contact. The other day I received a stick and while I put experiments.
With your firmware, the work of the stick itself became stable, on the stock there were problems with accessing the settings through Mikrotik, and the stick did not always start.

Now it works just fine, judging by the temperature measurements, it has become less heated.

But I ask for help, for the complete purity of the experiment, I really need a stock firmware. I wrote to all the mail indicated by the manufacturer (http://www.odi-tec.com/), but did not receive any response, I also wrote to the Fibershow Store seller (https://aliexpress.ru/) - he read my message, but also didn't answer.

If there is a stock firmware, I will be very grateful if you can share it.

The stick was bought on your recommendation: https://aliexpress.ru/item/1005003515662920.html?gatewayAdapt=glo2rus&sku_id=12000026135734082

Thank you!

Anime4000, 2022-04-18:
hi @mcheev 😄 

What I understand, you want original firmware that come from factory right?

I believe factory firmware is `V1.0-210702`

The Stick has 2 `rootfs` partition:
```
# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00040000 00001000 "boot"
mtd1: 00002000 00001000 "env"
mtd2: 00002000 00001000 "env2"
mtd3: 0003c000 00001000 "config"
mtd4: 0014c000 00001000 "k0"
mtd5: 00274000 00001000 "r0"
mtd6: 0014c000 00001000 "k1"
mtd7: 00274000 00001000 "r1"
mtd8: 00001000 00001000 "Partition_008"
mtd9: 00001000 00001000 "Partition_009"
mtd10: 00001000 00001000 "Partition_010"
mtd11: 00001000 00001000 "Partition_011"
mtd12: 0014c000 00001000 "linux"
mtd13: 00274000 00001000 "rootfs"
```

Partition are:
1st `rootfs` Partition: `mtd5: 00274000 00001000 "r0"`
2nd `rootfs` Partition: `mtd7: 00274000 00001000 "r1"`

Try navigate `http://192.168.1.1/omci_info.asp` and see `Software Version X`, which one you updated:
[image]

if old version there, you can switch via Telnet:

Software Version 1:
```
nv setenv sw_commit 0
nv setenv sw_commit 0
```

Software Version 2:
```
nv setenv sw_commit 1
nv setenv sw_commit 1
```

mcheev, 2022-04-18:
@Anime4000 
That's right, I wanted to return the firmware to stock, unfortunately in a hurry I did not save the default configuration and firmware, and now I have the same as your screenshot :)

I have the same :(
First: 1.0-220304
Second: 1.1-220407 (Beta - https://github.com/Anime4000/RTL960x/issues/17#issuecomment-1100737094)

And most likely due to my stupidity, I was unable to establish a connection with gpon, the status, if I'm not mistaken, is O0.
On the stock was O5, but the stick behaved extremely strange.
I also changed the parameters via Telnet and now it's not clear why there is no connection.

To be honest, it seems to me that restoring the stick to the factory version will no longer work and it's easier to throw it away and buy a new one :)

By the way, which stick is preferable to choose now?
1. https://aliexpress.ru/item/1005003515662920.html?spm=a2g2w.productlist.0.0.534a2f9cTKIVhF&sku_id=12000026135734083
2. https://aliexpress.ru/item/4000086970977.html?spm=a2g2w.productlist.0.0.534a2f9cTKIVhF&sku_id=10000000231912386

Anime4000, 2022-04-18:
If you got `O0` means `MAC_KEY` is wrong for `ELAN_MAC_ADDR`

backup your configuration and erase `config` partition:
```
flash_eraseall /dev/mtd3
```

then restore config without changing `ELAN_MAC_ADDR`

this will use default MAC Address hard coded inside.

reason ODI Firmware `22x` like this is unknown, this behaviour is similar to VSOL V2801F `VS_AUTH_KEY`. Probably Realtek SDK Bridge Mode require a "KEY"

mcheev, 2022-04-19:
That's it, I finally finished off the stick, now I can't get through to it :)
But there is good news, now I will order the original stick directly from the manufacturer, in a new design.

The only thing I did not understand was whether a colleague from forum.ru-board.com (https://github.com/Anime4000/RTL960x/issues/14#issue-1071915963) managed to run it in bridge mode (so that Mikrotik would receive an external IP address) .

Thank you very much for your help.

skon77, 2022-04-19:
> The only thing I did not understand was whether a colleague from forum.ru-board.com ([#14 (comment)](https://github.com/Anime4000/RTL960x/issues/14#issue-1071915963)) managed to run it in bridge mode (so that Mikrotik would receive an external IP address) .

Yes. Everything is working fine on my RB5009.

I have two sticks:
[TWCGPON657](https://item.taobao.com/item.htm?id=597031866488)
LAN_SDS_MODE=1 (bridge: only Internet)

[DFP-34X-2C2](https://detail.tmall.com/item.htm?id=654848544398)
LAN_SDS_MODE=7 (bridge: VLANs + IPTV multicast)

mcheev, 2022-04-19:
@skon77 ,hello!
> [**DFP-34X-2C2**](https://detail.tmall.com/item.htm?id=654848544398) (FW: M114_sfp_ODI_Vlan_220407.zip BETA) LAN_SDS_MODE=7 (bridge: VLANs + IPTV multicast)

Thanks, now I understand that it works through VLAN, I tried to get an external address directly on Mikrotik

> Is this a HSGQ seller? I could find him on TMALL. only.
Yes, HSGQ representatives contacted me.

skon77, 2022-04-19:
> Exactly LAN_SDS_MODE=7? Not 1?

Yes, exactly 7. Anime4000 recommends 0 for autoconfiguration, but it didn't work for me. I set it to 7 manually.

And a little more additional information. My ISP turned on ANYSERVICE option for me to get service on any ONT port. It allows IPTV and Internet to work on any port in the same VLAN. And it's also the reason why multicast is streaming without VLAN, IPTV can work through the VLAN dedicated to the Internet. All of this has made it difficult to manage IPTV via Mikrotik. Maybe I should ask my ISP to disable this option ANYSERVICE and bring back a separate VLAN for IPTV. Or forget about the ISP's IPTV, I don't watch TV anyway.

For that reason, I really like the simple and reliable TWCGPON657. Just the internet and no unnecessary incoming streams.

Anime4000, 2022-04-25:
@mcheev if you want older firmware `210702` I have keeping it here:
https://github.com/Anime4000/RTL960x/raw/main/Firmware/DFP-34X-2C2/M114_sfp_ODI_210702.tar

This firmware in Router Mode,
This firmware did not work with my GPON 😢

mcheev, 2022-04-29:
@Anime4000 Thank you very much! Is this a stock version or a modified one? When installing, will MAC_KEY "fly off"?
As it turned out, it works perfectly for me. The truth is unstable when restarting the device :(

Anime4000, 2022-05-08:
@mcheev you could try this firmware [M114_sfp_ODI_Vlan_220414.zip](https://github.com/Anime4000/RTL960x/files/8546714/M114_sfp_ODI_Vlan_220414.zip) which is it fix some VLAN issue, also support PVID (VLAN untag) like Huawei MA5671a