Anime4000/RTL960x #24: ODI NEW FIRMWARE

macafeeee, 2022-04-23:
I get the new firmware V220414. I dont know what changed and I dont have the mac_key calculator. Please share the mac_key calculator if you have it. Thank you.
[M114_sfp_ODI_Vlan_220414.zip](https://github.com/Anime4000/RTL960x/files/8546714/M114_sfp_ODI_Vlan_220414.zip)

Anime4000, 2022-04-23:
Great!

This firmware allow you to untag VLAN for router that not support VLAN, such as TP-Link TL-XDR5480

macafeeee, 2022-04-23:
> 

Do you have the mac_key calculator?

Anime4000, 2022-04-23:
> > 
> 
> Do you have the mac_key calculator?

no, I don't have...
you can request at ODI

macafeeee, 2022-04-23:
> > > 
> > 
> > 
> > Do you have the mac_key calculator?
> 
> no, I don't have... you can request at ODI

ODI is not friendly and the response speed is slow. Looking forward to hack it.

skon77, 2022-04-23:
> ODI is not friendly and the response speed is slow.

And it's not certain that they will generate codes for everyone. It is unlikely that they are interested. A systemic solution is needed here.

Anime4000, 2022-04-23:
> > > > 
> > > 
> > > 
> > > Do you have the mac_key calculator?
> > 
> > 
> > no, I don't have... you can request at ODI
> 
> ODI is not friendly and the response speed is slow. Looking forward to hack it.

What I understand, GPON Standard not using MAC Address to Authenticate:

This what OLT see:
[image]
[image]

I believe your Router WAN MAC Address need to change

skon77, 2022-04-23:
> What I understand, GPON Standard not using MAC Address to Authenticate

In my country, one of GPON providers (not mine) uses binding to  MAC address. I don't know all the nuances, but it seems to use the Port Security option.
Port Security is a feature of the switch that allows you to specify MAC addresses of hosts that are allowed to transmit data through the port.

Anime4000, 2022-04-23:
> In my country, one of GPON providers (not mine) uses binding to MAC address. I don't know all the nuances, but it seems to use the Port Security option. Port Security is a feature of the switch that [...]

it seem your GPON Provider using proprietary OMCI...

`pon0` in ODI Stick using Null MAC, you can change by using `ipconfig`.

GPON should be Layer 3, only router MAC Address need to change for DHCP Server authenticate the router.

in my country (Telekom Malaysia) are using Universal Deployment... Mix and Match various vendor and various model, Telekom Malaysia only need valid PLOAM Password & Hardware Version.

Without valid Hardware Version, OMCI upstream VLAN not open even having `O5`

stich86, 2022-04-24:
still same problem of other firmware, the MEID 171 is not honored.

Only way to bring up PPPoE is to using the VID in MEID 84 :(

stich86, 2022-04-24:
> GPON should be Layer 3, only router MAC Address need to change for DHCP Server authenticate the router.

usually MAC address is used on the LAN side.
May be they check that one? on the GPON stuff MAC is not used, everything is encapsulated into a GEM frame which doesn't contain any MAC information of PON port

Anime4000, 2022-04-24:
> everything is encapsulated into a GEM frame which doesn't contain any MAC information of PON port
Indeed... 😄 

If you using Stick as Router Mode, then you need change Stick MAC

Newer ODI firmware is Bridge Mode, MAC Address not used

macafeeee, 2022-04-24:
> > > > > 
> > > > 
> > > > 
> > > > Do you have the mac_key calculator?
> > > 
> > > 
> > > no, I don't have... you can request at ODI
> > 
> > 
> > ODI is not friendly and the response speed is slow. Looking forward to hack it.
> 
> What I understand, GPON Standard not using MAC Address to Authenticate:
> 
> This what OLT see: [image] [image]
> 
> GPON using ATM like ADSL2+ and connection at Layer 3, no MAC Address is involved
> 
> I believe your Router WAN MAC Address need to change
> 
> if you using EPON, then different story...
> 
> Sorry

Unfortunately，I'm using the EPON Line.

Anime4000, 2022-04-25:
> Unfortunately，I'm using the EPON Line.

You can buy TWCGPON657 then use V2801F firmware in it, from that you can change MAC Address by using Keygen provided on the guide

stich86, 2022-04-27:
just for information, that firmware doesn't seems stable when the stick run in HiSGMII mode.
I had to revert back on 220404 that doesn't have link loss behaviour

Anime4000, 2022-04-27:
since I know `nv setenv sw_commit` `sw_active`
I simply switch 😄

stich86, 2022-04-27:
> since I know `nv setenv sw_commit` `sw_active` I simply switch 😄

yep I know ;)
hope they don't screw up the HiSGMII part!!

mcheev, 2022-04-29:
Can you explain what is the difference between VS_AUTH_KEY and MAC_KEY
I'm confused about these concepts.

Anime4000, 2022-04-29:
the key like a License or Lock MAC Address or other info from changing...

# VS_AUTH_KEY
this for V2801F, it prevent changing MAC Address and Hardware Version

# MAC_KEY
this for ODI DFP stick, it prevent changing MAC Address only...

TWCGPON657 stick/firmware don't have these problem

macafeeee, 2022-05-28:
New Firmware V220527! That's support pppoe or ipoe. It still need mac key.
[M114_sfp_ODI_hybrid_220527.zip](https://github.com/Anime4000/RTL960x/files/8791246/M114_sfp_ODI_hybrid_220527.zip)

Anime4000, 2022-05-28:
wow, this hybrid can switch mode between Router and Bridge Mode ?

macafeeee, 2022-05-28:
> wow, this hybrid can switch mode between Router and Bridge Mode ?

Yes! You are right!

stich86, 2022-05-28:
> wow, this hybrid can switch mode between Router and Bridge Mode ?

looks like this firmware still doesn't honors `DEVICE_TYPE` 
setting it to `0` or `1` it's the same, "spppd" is still running.

But looks like, that MEID `171` for my FTTH connection now has only on entry

I'll try in the next days to connect over swapped VLAN to see if it fix my initial issue :)

stich86, 2022-05-28:
they have also changed the WebUI (looks very weird..)

[image]

stich86, 2022-05-28:
I confirm this firmware has same behaviour of 2107XX, so I can bring up PPPoE using my "swapped" VLAN 835 instead of native 100.

Now @Anime4000 you should check if the 4-port emulation is working, because this firmware seems a pure "router" mode

stich86, 2022-05-28:
> New Firmware V220527! That's support pppoe or ipoe. It still need mac key. [M114_sfp_ODI_hybrid_220527.zip](https://github.com/Anime4000/RTL960x/files/8791246/M114_sfp_ODI_hybrid_220527.zip)

just a question (if you know): MAC_KEY is needed also to bring up PPPoE\IPoE? I got a second ODI (the one with embedded radiator), but also on this one the MAC_KEY is empty :(

Anime4000, 2022-05-28:
@stich86 
Firmware `220527` doesn't work with 4-port emulation, `omcicli mib get 84` show same table as before, `pon_wan.asp` try to map vlan `621` doesn't work...

I had to revert back to `220414` firmware.

this mean...
ISP use ONU as SFU (Switch Fabric Unit) like my ISP, need to use `220414` bridge firmware
ISP use ONU as IGD (Internet Gateway Device), need to use `220527` router firmware

`220527` is not hybrid, it act like `210702.tar` older firmware, worthless for ONU in SFU

Interface now going too like V2801F...

vuducdong, 2022-05-30:
[image]

Anime4000, 2022-05-30:
my ISP use SFU that has POTS build-in, only V2801F and DFP-34X-2C2 `220304` work just fine

I tried G-010S-P, it cause stick disconnect from RB5009 when applying OMCI VLAN filter:

https://user-images.githubusercontent.com/1908715/170985519-2ecabc06-bc59-4616-a32f-81900e4a1d19.mp4

This mean, not all xPON ONU SFP support SFU mode. MA5671a, G-010S-P, TWCGPON657 `stock`, DFP-34X-2C2 `220527`... failed to bridge and failed to emulating 4 LAN Ports...

akw28888, 2022-07-26:
Some interesting things:

Usually, SFU mode uses PPTP (Physical Path Termination Point), and HGU/IGD uses VEIP (Virtual Ethernet Interface Point).
But DFP-34X-2C2 is more preferred using VEIP when OLT supports it, even on 220304 and 220414 fw.
The ONU from my ISP work as HGU, but 220527 firmware is not working, only 220414 works.

And the MA5671A, G010SA, and G010SP (lantiq sticks) only support PPTP mode, and seems the stick don't support ME 171.
On the firmware for lantiq sticks from right.com.cn, actually they make a script to make ME 171 rule manually.

Anime4000, 2022-07-27:
@akw28888 

Interesting... this is new information...

no wonder I can not use firmware ODI `210702` and `220527`, only works `220304` and `220414`

it show my OLT use VEIP, making my Lantiq stick cannot be use

akw28888, 2022-07-28:
@Anime4000

Do you receive ME 329 (VEIP) on the ODI stick? Or you only receive ME 11 (PPTP).

Lantiq sticks have some bugs on bash scripts, like equipment id and software version will be converted to wrong strings, but it can be solved.

And, how many parameters will verify on TM's OLT?

Anime4000, 2022-07-28:
ME 329, VEIP
[image]

ME 11, EthUni, PPTP ?
[image]

ME 53 prints noting...

what I understand, my ISP can provision specific ONU LAN port and disable other ONU LAN port, this called `EthUni` in ME 11

when I use single LAN ONU, the LAN port disabled after O5, it try to enable LAN 2 on single LAN ONU which is not exist...

akw28888, 2022-08-03:
@Anime4000 try to check me 264

Anime4000, 2022-08-04:
@akw28888 
```
# omcicli mib get 264
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Unig
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityID: 0x0101
CfgOptionStatus: 0x0000
AdminState: 0
ManageCapability: 2
NonOmciPointer: 0x0000
RelayAgentOptions: 0x0000
=================================
=================================
EntityID: 0x0601
CfgOptionStatus: 0x0000
AdminState: 0
ManageCapability: 0
NonOmciPointer: 0x0000
RelayAgentOptions: 0x0000
=================================
```

What this mean?

akw28888, 2022-08-04:
ME 264 is UNI-G (PPTP Ethernet UNI), the ME 264 works with ME 11.

I am trying to find out which ME defines 4 LAN port.

Anime4000, 2022-08-04:
I see, this is odd, entity `0x0601` exist on ME 264 and ME 329

ME 329
```
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
VEIP
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityId: 0x0601
AdminState: 0
OperState: 0
InterDomainName: 0x00000000000000000000000000000000000000000000000000
TcpUdpPtr: 0x0000
IansAssignPort: 0
CustomizedSlotId: 0xff
=================================
```

ME 11
```
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
EthUni
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityID: 0x0101
ExpectedType: 47
SensedType: 47
AutoDectectCfg: 0
EthLoopbackCfg: 0
AdminState: 1
OpState: 0
DuplexInd: 3
MaxFrameSize: 2000
DTEorDCEInd: 0
PauseTime: 65535
BridgedorIPInd: 2
ARC: 0
ARCInterval: 0
PppoeFilter: 0
PowerControl: 0
PhysicalPortId: 0
=================================
```

ME 264
```
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Unig
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityID: 0x0101
CfgOptionStatus: 0x0000
AdminState: 0
ManageCapability: 2
NonOmciPointer: 0x0000
RelayAgentOptions: 0x0000
=================================
=================================
EntityID: 0x0601
CfgOptionStatus: 0x0000
AdminState: 0
ManageCapability: 0
NonOmciPointer: 0x0000
RelayAgentOptions: 0x0000
=================================
```

ME 329 + ME 11 = ME 264

tdmadam, 2022-08-05:
> I confused now, it should be PPTP or VEIP but not both right?

It is normal for dual-managed ONUs that ME 264 includes VEIP. The days when ONUs only implemented PPTP or VEIP entity are slowly becoming a thing of the past. And DFP-34X-2C2 is not an ordinary SFP it has many software hacks to emulate some functions.

Most Routing Gateways (RGs) are now certified as dual-managed utilizing either VEIP or PPTP to manage non-OMCI domain. Check the "Notes" in the latest list of devices

<span>www</span>.broadband-forum.org/wp-content/uploads/2022/08/BBF.247-GPON-ONU-Products-2022-08-01.pdf

Anime4000, 2022-08-05:
@tdmadam understood

based ME report I put, my fiber is confirmed using VEIP right?

only VEIP can provision enable/disable certain LAN Port and map inner outer VLAN

tdmadam, 2022-08-05:
The reports from the DFP-34X-2C2 can only be trusted to a certain extent, as we know this SFP is known for "fake" responses to OLT to make it happy. The SFP even between firmware versions behaves differently. Ideally, it is best to capture the OMCI log on the original device provided by the ISP to see how services are configured.

When the ONT is in HGU mode it allows many classes of traffic to flow through the VEIP, but the actual provisioning of VLANs, LAN ports, WiFi, VoIP on non-OMCI domain is done on the HGU either remotely by the maintenance staff or via an auto-provisioning method like ftp, dhcp, TR-069 or pushing xml file. If the user has access to the original ONT then can preview how non-OMCI settings are configured and replicate them on the web interface of the SFP.

This page contains plenty of examples of how to set up different types of service for different OLTs like NOKIA, Huawei, Fiberhome, Calix which gives insight into how it is done.

<span>halny.com</span>/knowledge-base/

lanseyujie, 2022-11-01:
New Firmware

Anime4000, 2022-11-01:
Nice!

luizfeliperj, 2023-05-23:
Regarding the MAC_KEY, have you guys used this script to generate a new key?

https://gist.github.com/rajkosto/29c513b96ea6262d2fb1f965a52ce16f

jackblackjjx, 2024-02-03:
@Anime4000  Hey,friend. Now I use the newer firmware, the same  as you before,I can't use pppoe or bridge  to connect my internet.And How to downgrade firmware?
(I directly upgraded with the old firmware, but after the upgrade, it was garbled and still not working, so I used the latest again)
 So how can I fix this? Buy a new managed switch? Or a dual 2.5g router（i wanna buy zte ax3000pro+,looks like not bad）？

onion83, 2024-02-03:
> I get the new firmware V220414. I dont know what changed and I dont have the mac_key calculator. Please share the mac_key calculator if you have it. Thank you. [M114_sfp_ODI_Vlan_220414.zip](https://github.com/Anime4000/RTL960x/files/8546714/M114_sfp_ODI_Vlan_220414.zip)

> echo -n "hsgq1.9a[MACADDR]" | md5sum

[MACADDR] format Like this: AABBCCDD1234 (Uppercase)

kelt84, 2024-10-10:
Hello everyone, tell me, maybe someone knows how to solve this issue, I bought a DFP-34X-2C2 and reflashed it with all available firmware, the situation is as follows, authentication with the provider is carried out by the serial number of the device, when entering the serial number and then saving, it asks to enter the PLOAM password, but since authentication is carried out by the serial number, the PLOAM field must be left empty, how can this be done?

vuducdong, 2024-10-10:
> Hello everyone, tell me, maybe someone knows how to solve this issue, I bought a DFP-34X-2C2 and reflashed it with all available firmware, the situation is as follows, authentication with the [...]

Please don't spam. You post 4 comment the same in 4 post.

Please show up your photos about your problem.

@Anime4000

kelt84, 2024-10-10:
Sorry for the posts, as I didn't know what topic to write in, now there is no way to attach a photo, while at work. I will try to describe my problem in detail. My provider authentication is only by serial number, PLOAM password is not needed, when you enter the serial number in the stick settings, you must enter the PLOAM password to apply the settings, but in my situation the ploam password field must be empty. How to save the serial number without ploam password?

vuducdong, 2024-10-10:
> Sorry for the posts, as I didn't know what topic to write in, now there is no way to attach a photo, while at work. I will try to describe my problem in detail. My provider authentication is only by [...]

You can use commands from telnet

Ex: flash set GPON_SN

kelt84, 2024-10-10:
ok, everything is clear, what command then serves for a full reset of sfp to factory settings, if the reset is done through the web interface, after the reset the Ploam password field still contains data.

vuducdong, 2024-10-10:
> ok, everything is clear, what command then serves for a full reset of sfp to factory settings, if the reset is done through the web interface, after the reset the Ploam password field still contains data.

flash default hs

remember MAC and MACKEY to fill again after reset done.

kelt84, 2024-10-10:
I did everything via telnet with the command flash default hs, the result is the same, the ploam field looks like 12345! Where to dig?
[image]

kelt84, 2024-10-10:
via the command flash default cs the same thing only with different numbers
[image]

vuducdong, 2024-10-10:
> via the command flash default cs the same thing only with different numbers
> 
> [image]
> 
> 

Please read this instruction:

[https://github.com/Anime4000/RTL960x/blob/main/Docs/factory_reset.md](https://github.com/Anime4000/RTL960x/blob/main/Docs/factory_reset.md)

vuducdong, 2024-10-10:
> I did everything via telnet with the command flash default hs, the result is the same, the ploam field looks like 12345! Where to dig?
> 
> [image]
> 
> 

After that command you need reboot:

flash default hs
reboot

kelt84, 2024-10-12:
who can tell me what the problem might be, the connection does not rise, the status jumps from 02-05 cyclically, and maybe someone knows how to enter the full serial number in the format 48575443D1444C91 16 characters? I blame the OLT that does not accept HWTCD1444C91 in this form
[image]

kelt84, 2024-10-12:
more errors added
[image]

vuducdong, 2024-10-12:
> I blame the OLT that does not accept HWTCD1444C91 in this form

Please show a full image of your settings screen.

kelt84, 2024-10-12:
[image]
[image]
[image]
[image]

vuducdong, 2024-10-12:
@kelt84 try upgrade this firmware

[https://github.com/Anime4000/RTL960x/blob/main/Firmware/DFP-34X-2C2/M110_sfp_ODI_220923.tar](url)

and read instruction for setup here: [https://github.com/Anime4000/RTL960x/blob/main/Docs/Setup_Stick.md](url)

kelt84, 2024-10-15:
Hello everyone, I tried all possible settings, the authentication status is constantly in the 02-05 cycle, I blame the serial number in the HWTC 12-character format, I am currently using the EG8145x6 terminal, authentication is carried out by the serial number in the 16-character format, I cannot find the GPON SERIAL NO line in the configuration file. Can anyone tell me how to enter a 16-character number?

[14 more replies]