Anime4000/RTL960x #26: ODI Firmware 2203XX/2204XX don't honorate OCMI MEID 171 - ExtVlanTagOperCfgData

stich86, 2022-04-24:
Hi,

currently on most of FTTH lines in Italy, the PPPoE session is established using VLAN tag 835, then the OLT provisioning the MEID 171 `ExtVlanTagOperCfgData` that instruct the ONT to swap this VLAN with the one reported into MEID 84 `VlanTagFilterData`

The current full output is this one:

MEID 84

```
# omcicli mib get 84
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
VlanTagFilterData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityID: 0x03
FilterTbl[0]: PRI 0,CFI 0, VID 100
FwdOp:  0x10
NumOfEntries: 1
=================================
```

MEID 171

```
# omcicli mib get 171
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
ExtVlanTagOperCfgData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityId: 0x01
AssociationType: 2
ReceivedFrameVlanTagOperTableMaxSize: 0
InputTPID: 0x8100
OutputTPID: 0x8100
DsMode: 0
ReceivedFrameVlanTaggingOperTable
INDEX 0
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 8,VID 835, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 1
Treatment Inner   : PRI 0,VID 100, TPID 0
INDEX 1
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 15,VID 4096, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 4096, TPID 0, RemoveTags 0
Treatment Inner   : PRI 0,VID 1, TPID 4
AssociatedMePoint: 0x101
DscpToPbitMapping:
	0x000000
	0x000000
	0x000000
	0x000000
	0x000000
	0x000000
	0x000000
	0x000000
=================================
=================================
EntityId: 0x02
AssociationType: 10
ReceivedFrameVlanTagOperTableMaxSize: 0
InputTPID: 0x8100
OutputTPID: 0x8100
DsMode: 0
ReceivedFrameVlanTaggingOperTable
INDEX 0
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 15,VID 4096, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3
Treatment Inner   : PRI 15,VID 0, TPID 0
AssociatedMePoint: 0x601
DscpToPbitMapping:
	0x000000
	0x000000
	0x000000
	0x000000
	0x000000
	0x000000
	0x000000
	0x000000
=================================
```

The part that should be honored is this one:

```
INDEX 0
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 8,VID 835, TPID 0, EthType 0x00 <-- inbound VLAN tag 835
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 1 <-- remove VLAN tag 835
Treatment Inner   : PRI 0,VID 100, TPID 0 <-- Add VLAN tag 100 (the one pushed on MEID 84)
```

So at the moment the only way to make PPPoE working is to tag on client/router side the connection with `VID 100`.

I've also tried to use Manual VLAN with these settings without success:

- Transparent Mode
- PVID 835
- PVID 100

Hope someone of ODI can read this issue and found a solution.

On latest 2107XX firmware I was able to bring up PPPoE with VLAN 835 successfully

Thx

manias21, 2022-04-24:
It is true, because in movistar it is Vlan6 and it works for me with another Vlan. But it works!!!

crrodriguez, 2022-06-22:
same here.
```
omcicli mib get 171
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
ExtVlanTagOperCfgData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityId: 0x101
AssociationType: 2
ReceivedFrameVlanTagOperTableMaxSize: 0
InputTPID: 0x8100
OutputTPID: 0x8100
DsMode: 0
ReceivedFrameVlanTaggingOperTable
INDEX 0
Filter Outer   : PRI 8,VID 2, TPID 5
Filter Inner   : PRI 8,VID 4096, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 1
Treatment Inner   : PRI 9,VID 2496, TPID 3
INDEX 1
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 8,VID 2, TPID 5, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 1
Treatment Inner   : PRI 8,VID 2496, TPID 2
INDEX 2
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 15,VID 0, TPID 5, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3
Treatment Inner   : PRI 15,VID 0, TPID 2
INDEX 3
Filter Outer   : PRI 14,VID 4096, TPID 5
Filter Inner   : PRI 14,VID 4096, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3
Treatment Inner   : PRI 15,VID 4096, TPID 3
INDEX 4
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 14,VID 4096, TPID 5, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3
Treatment Inner   : PRI 15,VID 4096, TPID 2
```

SUpposedly I should not have to use vlan 2492 but vlan 2 as internet. (movistar CL)