Anime4000/RTL960x #31: Opton GP801R OEM stick in Turris Omnia

AreYouLoco, 2022-05-11:
Hi I recently bought Opton GP801R and it seems like its normal V2801F underneat. I can login there with admin/stdONU101 and webui title says V2801F.

My router Turris Omnia has SFP cage and its 2,5Gbps capable. I was wondering is it safe to flash latest firmware of yours and will it work @2,5Gbps.

What steps are needed to flash this firmware?

Is doing https://github.com/Anime4000/RTL960x/tree/main/Firmware/V2801F only needed or do I have to worry about VS_AUTH_KEY even if I have no need to change MAC address.

I've read the whole https://github.com/Anime4000/RTL960x/issues/17 Great work!

Anime4000, 2022-05-11:
No need to worry if keeping MAC Address and/or HW version original value...

Do note that V2801F support 2.5GbE in limited of device, so far I tried with TL-XDR5480 just fine

also V2801F don't have automatic `LAN_SDS_MODE` negotiation like ODI Stick, incase of lock out or cannot login, you still can access via UART

AreYouLoco, 2022-05-11:
I have UART adapter and SPI programmer in case but I would like to avoid taking apart the SFP. So when I run flash set LAN_SDS_MODE 1 and then flash your firmware that should be enough?

Also you should add Opton GP801R as another clone to the documentation and when I succeed with flashing and testing I will report back if it works with my router. Now it detects 1Gbps link because EEPROM patch is included in this openwrt version. Will see if it does 2,5Gbps.

Also how do I know if the firmware flashed successfully?

Anime4000, 2022-05-11:
Yes, you need run `flash set LAN_SDS_MODE 1`, then you can update

it will reboot automatically when firmware update is complete, but still no indicator if stick is finish booting or hang... only way to know is read UART output...

I add **Opton GP801R** as V2801F clone

AreYouLoco, 2022-05-11:
So I flashed your firmware and after reboot the webui changed to yours but the negotiated speed is still 1Gbps. My router has 2,5Gb sfp port. Even to get that webui it was worth it. But what I could try to set speed to 2,5GbE?

AreYouLoco, 2022-05-11:
root@router:~# dmesg | grep -E "(sfp|eth2)"                                               
[    1.854999] mvneta f1034000.ethernet eth2: Using hardware mac address d8:58:d7:00:3d:9f
[    7.281393] sfp sfp: Host maximum power 3.0W
[    7.619766] sfp sfp: Detected broken RTL8672/RTL9601C emulated EEPROM
[    7.626231] sfp sfp: Switching to reading EEPROM to one byte at a time
[    7.672993] sfp sfp: module OEM              V2801F           rev 1.0  sn 202110220325
    dc 211027
[    7.682342] mvneta f1034000.ethernet eth2: interfaces=[mac=4,8-11,16,18-19 sfp=18] sele
cted 18 (1000base-x)
[    7.692108] mvneta f1034000.ethernet eth2: switched to inband/1000base-x link mode
[    7.702155] sfp sfp: skipping hwmon device registration due to broken EEPROM
[    7.709220] sfp sfp: diagnostic EEPROM area cannot be read atomically to guarantee data
 coherency
[   11.684938] mvneta f1034000.ethernet eth2: configuring for inband/1000base-x link mode
[   11.860345] mvneta f1034000.ethernet eth2: Link is Up - 1Gbps/Full - flow control off
[   11.877244] IPv6: ADDRCONF(NETDEV_CHANGE): eth2: link becomes ready

AreYouLoco, 2022-05-11:
Another problem I encountered is that I can login to WebUI with admin/stdONU101 but cannot login to telnet

AreYouLoco, 2022-05-11:
Ok I flashed firmware again. It was upgrading firmware -> rebooting and instead of rebooting the router I did ifup on the interface because of the errors (below) I saw in log and I could login via telnet from my mobile. So false alarm.

[2780.147161] mvneta f1034000.ethernet eth2: Link is Down
[ 2780.152425] sfp sfp: module transmit fault indicated
[ 2785.592264] sfp sfp: module persistently indicates fault, disabling
[ 2825.419723] mvneta f1034000.ethernet eth2: configuring for inband/1000base-x link mode
[ 2825.487596] mvneta f1034000.ethernet eth2: Link is Up - 1Gbps/Full - flow control rx/tx
[ 2825.495648] IPv6: ADDRCONF(NETDEV_CHANGE): eth2: link becomes ready

tdmadam, 2022-05-11:
@AreYouLoco 
I prefer to use netcat to backup the whole mtd3 partition(config)

```
# Listen on port 9999
Ubuntu# nc -l -p 9999 > mtd3

# Send mtd3 to Ubuntu
V2801F# cat /dev/mtd3 | nc 192.168.1.150 9999
```

AreYouLoco, 2022-05-11:
Yeah thats a solution but for now I have nothing to backup. Default config.
Was trying to make a point that WebUI might be broken. But thanks!

Anime4000, 2022-05-11:
I plan to create new repo for WebGUI but... I found out ASP variable is different each firmware and stick, cannot made into universal, unless we have RTL960x source code, we can build and fix our own firmware...

I make new WebGUI just to follow HTML5 standard and big screen, I also notice change value in GUI seem not work especially in `omci_info.asp` page. but good interface to view...

You can try `LAN_SDS_MODE 6` (2500BaseX) if your router support Base-X
Most router support `LAN_SDS_MODE 1` (1000BaseX) because of supported IEEE standard...

Do note 2500BaseX is not an IEEE standard, you might need set `0x8000` in ethtool

| Mode        |  Advert |
|-----------|-------|
| 2500BaseT | 0x8000 |
| 2500BaseX | 0x800000000000 |

tdmadam, 2022-05-12:
> I plan to create new repo for WebGUI but... I found out ASP variable is different each firmware and stick, cannot made into universal, unless we have RTL960x source code, we can build and fix our [...]

I was thinking more about using V1.0-220414 as a base, fix the WebUI, maybe add some new pages that show what VLANs are available, read EEPROM, custom scripts and variables that are written to /var/config and available after reboot.

Anime4000, 2022-05-12:
seem possible if you can reverse engineering `boa` to add new ASP variable.

view available VLAN in WebGUI is very neat feature!

tdmadam, 2022-05-13:
To some extent boa could be tweaked or at least replicated some of its functions, but I get your point. It's not sustainable in the long run. This is why I say that CLI should be the primary method of configuration. With 40 stars now, you have a lot of audience. If everyone can share their insights, then a common set of commands can be developed for different service configuration scenarios(RGU,Bridge,Mixed) and different VLAN implementations(Transparent, Double tagged, Translated).

Anime4000, 2022-05-13:
Indeed...
What I know, ODI stick have issue with IPTV VLAN #32

it seem Hisense RTL9601B and Firmware has very stable firmware and have working list of VLAN id on vlan.asp like this
https://github.com/Anime4000/RTL960x/issues/18#issuecomment-1072015426

tdmadam, 2022-05-13:
> it seem Hisense RTL9601B and Firmware has very stable firmware and have working list of VLAN id on vlan.asp like this
> [#18 (comment)](https://github.com/Anime4000/RTL960x/issues/18#issuecomment-1072015426)

I would really like to get this working properly on our sticks. I think I've seen something similar implemented on my 1-port box, and even captured the command used to "discover" VLANs.

A little offtopic:
Has anyone physically opened the LTE3415 to see what SoC is inside? On the internet there are two versions of the LTE3415, probably with two different chips
```
LTE3415-SCA+     RTL9601B
LTE3415-SH+      RTL9601CI
```

Can anyone confirm.

vuducdong, 2022-05-13:
> I would really like to get this working properly on our sticks. I think I've seen something similar implemented on my 1-port box, and even captured the command used to "discover" VLANs.
>
>
>
> A little [...]

LTE3415-SCA+ support 1G

LTE3415-SH+ support 2.5G

I will send you tomorrow

tdmadam, 2022-05-13:
Thanks! Great information!

crrodriguez, 2022-05-13:
if I try LAN_SDS_MODE 6 on the ODI stick with the latest firmware and the router does notr support HSGMII will I permantently D.o.S myself out of the stick ? or it will fallback to mode 1 ?

AreYouLoco, 2022-05-16:
My router is 2500BaseT not 2500BaseX so I guess no 2,5Gbps for me:(

stich86, 2022-05-28:
> > > it seem Hisense RTL9601B and Firmware has very stable firmware and have working list of VLAN id on vlan.asp like this
> > 
> > 
> > > [#18 (comment)](https://github.com/Anime4000/RTL960x/issues/18#issuecomment-1072015426)
> > 
> > 
> > I would really like to get this working properly on our sticks. I think I've seen something similar implemented on my 1-port box, and even captured the command used to "discover" VLANs.
> > A little offtopic:
> > Has anyone physically opened the LTE3415 to see what SoC is inside? On the internet there are two versions of the LTE3415, probably with two different chips
> > ```
> > 
> > LTE3415-SCA+     RTL9601B
> > 
> > LTE3415-SH+      RTL9601CI
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Can anyone confirm.
> 
> LTE3415-SCA+ support 1G (SGMII)
> 
> LTE3415-SH+ support 2.5G (HSGMII)
> 
> [image]
> 
> [image]
> 
> [image]
> 
> [image]
> 
> [image]

Can you share your MTD dump? I want to test it on UF-Instant :)

vuducdong, 2022-05-29:
@stich86 @Anime4000   here's full dump mtd for LTE3415-SCA+ (RTL9601B)

[image]

[Hisense_LTE3415-SCA+ v1.7.3-200909.zip](https://github.com/Anime4000/RTL960x/files/8792386/Hisense_LTE3415-SCA%2B.v1.7.3-200909.zip)

vuducdong, 2022-05-30:
@stich86 hi, I'm from Vietnam.

marcinkuczera, 2022-06-09:
LTE3415-SH+ is one of best in class hardware design of SFP GPON stick, it's version is industrial (up to +85 celsius) and it contains additional MCU that handles most of I2C operations (instead of passing through via 9601C)

We do our own sticks based on that hardware from Hisense, but our own firmware. Much newer SDK (Luna 3.3) and kernel 3.18.x + ofcourse our well modified OMCI stack for wide interoperability.

stich86, 2022-06-09:
> LTE3415-SH+ is one of best in class hardware design of SFP GPON stick, it's version is industrial (up to +85 celsius) and it contains additional MCU that handles most of I2C operations (instead of passing through via 9601C)
> 
> 
> 
> We do our own sticks based on that hardware from Hisense, but our own firmware. Much newer SDK (Luna 3.3) and kernel 3.18.x + ofcourse our well modified OMCI stack for wide interoperability.
> 
> The model is LXT-010S-H, and will be available soon.

Wow you are the same person on LaFibra for the Ethernet 2.5 ONT? It will support HiSGMII on SFP?

Let us know when available :)

Anime4000, 2022-06-09:
> 

Nice, I would to buy and reasonable price,
I want to try with my Fiber Vendor (using 4-port ONU SFU) and my internet explicitly only on LAN 2 and OLT disable LAN 1, 3, 4...

Model number seem like Alcatel/Nokia

marcinkuczera, 2022-06-09:
Seems that answering on mail does not appears in web interface conversation.

Well, in case of price - if you have working product that is even 2x more expansive than crappy annoying "sell and forget" units - and you buy 2-3-4 or more of these for tests... what is cheaper ? :)

With 4 port emulation, not that easy. I mean - I would have to do a fake EthUni entities (0x101 to 0x104) - all bound with same physical port. And - if your ISP does configuration of all ports as untagged, such ONT would need to enforce tag to be left as is on PON port - otherwise you get a conflict. Quite complicate, I mean - lot of time to do it.
If it is like you say - maybe it is simpler, I mean - instead of 0x101 to present with EthUni 0x102..
But maybe - your ISP uses 4 port ONTs that work in VEIP mode (bridges are create by TR069 provisioning or other XML method)... lot of options. I would need to see OMCI provisioning log to recognize how they do it.

Anime4000, 2022-06-09:
I was finding good firmware that able to overcome 4-port Emulation, it seem ODI release 2 different firmware which is my fiber only work with SFU mode

I become obsess with GPON, so I learn hard way 😄 

I don't mind if cost, as long have good supported stable firmware,
I even make HTML5 Bootstrap WebGUI for V2801F just for High DPI Screen and look nice...

In my country,
Fiber Vendor provide multiple ISP and provision only OMCI, since OLT need to support multiple brand of ONU, VLAN mapping is 1:1, PON bridge to ETH with VLAN aware

In case `ONT would need to enforce tag to be left as is on PON port`, probably do like OpenWRT in MA5671a:
1. Display available VLAN at PON side
2. Create VLAN Mapping PON:ETH (2:100, 4:500, 7:155)
3. Support PVID for router don't support VLAN such as TL-XDR5480

I would be nice allow user to upload custom OMCI Profile, user can switch any OMCI Profile to spoof OLT

I'll try capture my PON OMCI Wireshark to you discover. This my VLAN Filter Data:
```
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
VlanTagFilterData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityID: 0x1202
FilterTbl[0]: PRI 0,CFI 0, VID 621
FwdOp:  0x10
NumOfEntries: 1
=================================
=================================
EntityID: 0x1203
FilterTbl[0]: PRI 0,CFI 0, VID 821
FwdOp:  0x10
NumOfEntries: 1
=================================
=================================
EntityID: 0x1204
FilterTbl[0]: PRI 0,CFI 0, VID 822
FwdOp:  0x10
NumOfEntries: 1
=================================
=================================
EntityID: 0x120b
FilterTbl[0]: PRI 0,CFI 0, VID 621
FilterTbl[2]: PRI 0,CFI 0, VID 821
FilterTbl[4]: PRI 0,CFI 0, VID 822
FwdOp:  0x10
NumOfEntries: 3
=================================
=================================
EntityID: 0xe102
FilterTbl[0]: PRI 0,CFI 0, VID 209
FwdOp:  0x10
NumOfEntries: 1
=================================
=================================
EntityID: 0xe10b
FilterTbl[0]: PRI 0,CFI 0, VID 209
FwdOp:  0x10
NumOfEntries: 1
=================================
```

crrodriguez, 2022-06-09:
> LTE3415-SH+ is one of best in class hardware design of SFP GPON stick, it's version is industrial (up to +85 celsius) and it contains additional MCU that handles most of I2C operations (instead of [...]

All righty.. looking forward to a reasonably priced international shipping suplier to consider trying it.

marcinkuczera, 2022-06-09:
I rather need extended vlan tag operations MIB, not a vlan filter.
If your firmware emulates 4x1G ONT - please send me complete:
omcicli mib get all

you say:
>In my country, Fiber Vendor provide multiple ISP and provision only OMCI, since OLT need to support multiple brand of ONU, VLAN mapping is 1:1, PON bridge to ETH with VLAN aware

Provisioning via OMCI - ok, for such case, but this is mostly ONT vendors that take care of interoperability, very rarely OLT vendors... (would have to be like Orange France do discuss such case)..

>I would be nice allow user to upload custom OMCI Profile, user can switch any OMCI Profile to spoof OLT
This is impossible.
The design is that OLT uploads at least some vlan-tagging-ops-config or extended-vlan-tagging-ops-confg to OLT + some supplementary objects. And these, strict config related objects are allowed only OLT->ONT.
It is only OLT that can do MIB_ADD, MIB_SET, MIB_DEL operations. ONT can only send back so called AVC (Attribute Value Change) information, and only if this is defined in MIB (parameter has AVC attribute).
AVC is used for example to send information about change of LAN port status (up/down) or if IP address of iphost has changed.. but there is no way to upload vlan config to OLT as this is impossible by design.

marcinkuczera, 2022-06-09:
> All righty.. looking forward to a reasonably priced international shipping suplier to consider trying it.

Where are you from ?

crrodriguez, 2022-06-09:
> > All righty.. looking forward to a reasonably priced international shipping suplier to consider trying it.
> 
> Where are you from ?

Chile.. if sold in aliexpress or amazon or whatever international site will be fine.

marcinkuczera, 2022-06-09:
> > > All righty.. looking forward to a reasonably priced international shipping suplier to consider trying it.
> > 
> > 
> > Where are you from ?
> 
> Chile.. if sold in aliexpress or amazon or whatever international site will be fine.

Nice place, I'd been once in Santiago.

We have never sold anything on ebay or amazon... how do you handle taxes on small low value packages ? Is it as simple as registred letter ?

crrodriguez, 2022-06-09:
> Nice place, I'd been once in Santiago.
>
> We have never sold anything on ebay or amazon... how do you handle taxes on small low value packages ? Is it as simple as registred letter ?

Well.. Im pretty sure that varies according to destination/origin  amazon ships from US to Chile. DDP only. Ebay normally does DAP, some products go through ebay international program or something and those go DDP.. small stuff from ebay is usually sent via registered mail and takes quite a while to arrive, but it is cheap and customs seldom have the resources to check all mail so it is rarely taxed, but it is.. is a pain in the ass and will cost you quite some money and bureocracy.

I recommend for more than one stick or for business shipping to never send DAP or said differently avoid anything that does not have clear, whole process upfront fees. DHL, FEDEX , UPS charge some ridiculous exorbitant fees for whatever extra service they think they provide otherwise.

stich86, 2022-06-09:
> Nice place, I'd been once in Santiago.
>
>
>
> We have never sold anything on ebay or amazon... how do you handle taxes on small low value packages ? Is it as simple as registred letter ?

I'm also interested.. I'm from Italy

Anime4000, 2022-06-10:
@marcinkuczera 
> I rather need extended vlan tag operations MIB, not a vlan filter.
If your firmware emulates 4x1G ONT - please send me complete:
omcicli mib get all

Here `putty.log` (131KB)
[putty.log](https://github.com/Anime4000/RTL960x/files/8878577/putty.log)

> Provisioning via OMCI - ok, for such case, but this is mostly ONT vendors that take care of interoperability, very rarely OLT vendors... (would have to be like Orange France do discuss such case)..

OLT: ALCL (Nokia brand last I saw on Fiber Technician iPad)

I found my OLT behavior:
1. Using **HG8240H5** Info `HW_HWVER 168D.A`: ALCL OLT just `O5`, provisioning, have internet.
2. Using **G-240G-E** Info `HW_HWVER 3FE48153CBAA`: ALCL OLT wont provision even `O5`, OMCI MIB detect OLT as `0x20202020` (Unknown OLT), Then flash set complete info, ALCL OLT authenticated,

based from that, I assume OLT are responsible for ONU interoperability, it seem Fiber Vendor configured all OLT to support many ONU brands

marcinkuczera, 2022-06-10:
ALCL = Alcatel Lucent ISAM 7360 OLT. Today - it is Nokia, but in system still keeps ALCL ID.
if G-240G-E does not go to 05, that may be more  complicate. Maybe PLOAM password is incorrect ?
There are different authentication methods, but most popular is GPON_SN.

Btw, one of my "legs" is regular telco operator ;)

Anime4000, 2022-06-10:
> ALCL = Alcatel Lucent ISAM 7360 OLT. Today - it is Nokia, but in system still keeps ALCL ID. if G-240G-E does not go to 05, that may be more complicate. Maybe PLOAM password is incorrect ? There are [...]

PLOAM password is same,
I just change Huawei to Nokia ONU Hardware version, ALCL OLT act differently...

I feel Nokia OLT very troublesome, giving `O5` even password is wrong... very stupid Nokia
saw my friend under Huawei OLT no issue even same Fiber Vendor....

dominiaz, 2022-06-10:
How to flash V2801F_V1.9.0-220425.tar to Opton GP801R?

I have installed V1.9.0-191015 and there isn`t upgrade from webui avaiable.

I can only backup/restore settings.

AreYouLoco, 2022-06-10:
@dominiaz login to the stick via telnet run ```flash set LAN_SDS_MODE 1``` IMPORTANT!
And then go to https://192.168.1.1/upgrade.asp and place firmware there. Wait for it to flash and then wait around 5-10 minutes and reboot device. You should be greeted with brand new HTML5 login page made by @Anime4000

dominiaz, 2022-06-10:
Thank you!

Unfortunately I have changed IP address to 192.168.100.200 and ping is ok but telnet and www is now dead. How to fix it?

AreYouLoco, 2022-06-11:
This I wont tell you but if it doesnt work try some development or beta version as it needs some newer kernel.

marcinkuczera, 2022-06-11:
> PLOAM password is same, I just change Huawei to Nokia ONU Hardware version, ALCL OLT act differently...
>
> I feel Nokia OLT very troublesome, giving `O5` even password is wrong... very stupid Nokia [...]

You don't know what authentication method is used on OLT side. To find a reason of not getting to O5 requires lot of debugging on frames leve.
Do you know if SN phase is passed ? (O3-O4) ? Maybe there is a problem with ranging ?

Each OLT vendor has different, own OMCI implementation, and trust me, there are lot of issues on all platforms and all vendors say they do it the best way.. and my last debugging was related to.. Huawei OLT (successful).

dominiaz, 2022-06-12:
I have tested GP801R (ver. V2801F_V1.9.0-220425) with MT 5009 and its working very well.

I am going to test 2.5Gb with CRS305-1G-4S+IN. Which LAN_SDS_MODE should I use to operate with GP801R with CRS305?

I don`t have UART so i am little scary before doing it...

stich86, 2022-06-12:
> I have tested GP801R (ver. V2801F_V1.9.0-220425) with MT 5009 and its working very well.
> 
> I am going to test 2.5Gb with CRS305-1G-4S+IN. Which LAN_SDS_MODE should I use to operate with GP801R with CRS305?
> 
> I don`t have UART so i am little scary before doing it...

on my ODI LAN_SDS_MODE is set to 4, but we are talking about a different chip (RTL9601D), so take care.. you can screw up everything if the stick doesn't sync and the only was to restore is with UART

marcinkuczera, 2022-06-12:
> we are talking about a different chip (RTL9601D),

9601D and 9601C are almost the same.
Differences:
- 9601D has additional 1GBaseT PHY build in
- 9601D runs with faster clock (300MHz) while 9601C runs only 112MHz

SGMII/HiSGMII port - is the same, as well as PON port, switch chip, registers etc..

dominiaz, 2022-06-16:
I have CRS305-1G-4S+IN and Opton GP801R (ver. V2801F_V1.9.0-220425) died after:

`fix the SFP port to 2.5 without auto-neg:

set [ find default-name=sfp-sfpplus3 ] advertise=2500M-full auto-negotiation=no speed=2.5Gbps

On ODI stick, because there is no auto-neg, it's necessary to fix LAN_SDS_MODE to 4:

flash set LAN_SDS_MODE 4 ; reboot`

Mikrotik shows that 2.5G link ok, but there is no ping to the device after all.

Can you fix firmware with script "if LAN_SDS_MODE 4 fails after 15mins then it automatically switch to LAN_SDS_MODE 1?"

Anime4000, 2022-06-16:
@dominiaz 
it's possible if you set sfp+ IP address 192.168.1.2

when ping failed 10 times, it revert back to 0 and reboot ..

dominiaz, 2022-06-16:
I`ve set 192.168.1.2/24 on sfp+ but the module doesn`t reboot.

It even link 1G,2.5G,10G with no autonegotiation but it doesn`t ping anymore.

closed 2022-09-18 by AreYouLoco

marcinkuczera, 2022-10-11:
> LTE3415-SH+ is one of best in class hardware design of SFP GPON
>     stick, it's version is industrial (up to +85 celsius) and it
>     contains additional MCU that handles most of I2C operations [...]
Yes :)
>
> It will support HiSGMII on SFP?
It does, but SGMII/HISGMII does not support auto-detect, so once you set 
it to 2.5G and your SFP link is not up, you need to restore it via 
console (there are pads on PCB, but not most comfortable in use, 
ubiquiti's pads are better (with holes)).
>
> Let us know when available :)
Delays delays delays... there is one production ongoing now, but there 
is also one large customer who will take most.. maybe I will keep some 
single box (5 units) for people from lafibre ;) But we alredy plan next 
manufacturing and - we DO HAVE chipsets on stock, so it will be quick.

Regards,
Marcin

qatar2030, 2023-08-19:
i have LEOX GPON STICK LXT-010S-H  and it was working great with mikrotik rb5009 and when i change mode from flash set LAN_SDS_MODE 1 to flash set LAN_SDS_MODE 8 the sfp stop working any help please

vuducdong, 2023-12-03:
> Is anybody can help me to find where to find firmware update option in WebUI (or trough telnet).
> 
> <img alt="Знімок екрана 2023-12-03 о 13 08 37" width="1121" src="https://private-user-images.githubusercontent.com/7815789/287514403-a9574128-1f7a-458b-b30b-1b03ebe50441.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDE2MDYzODksIm5iZiI6MTcwMTYwNjA4OSwicGF0aCI6Ii83ODE1Nzg5LzI4NzUxNDQwMy1hOTU3NDEyOC0xZjdhLTQ1OGItYjMwYi0xYjAzZWJlNTA0NDEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTIwMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMDNUMTIyMTI5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NTRhNTVmYzNiZWM2NTJlODlkNmMyMmU2NTcxYTZiNWNkMTc3YzgzOWRkYjdmYmE0MDkzMGYzZDAyMzJhNzE3OSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.Da6mSy7g5tfazy5Kke6Do6eFB1Rt0zRTdNQumQ4aVVg"> Just can't find it anywhere...

On webui, enter the following address

192.168.1.1/upgrade.asp

stich86, 2023-12-04:
V2801F's kernel doesn't support HiSGMII..

Only ODI, so now only way to recover the stick is using TTL

AreYouLoco, 2023-12-21:
How to access UART on V2801F?