Anime4000/RTL960x #32: ODI Firmware 220304 - hangs when turning on stb iptv

vuducdong, 2022-05-12:
Hi, 

Currently I'm using sfp Hisense LTE3415-SCA+ with RTL9601B, working fine with Huawei OLT (internet ok for vlan11, iptv ok with vlan12 and vlan9 for multicast)

`# omcicli mib get 84
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
VlanTagFilterData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
VlanTagFilterData
EntityID: 0x01
FilterTbl[0]: PRI 0,CFI 0, VID 11
FwdOp: 4
NumOfEntries: 1
=================================
=================================
VlanTagFilterData
EntityID: 0x02
FilterTbl[0]: PRI 0,CFI 0, VID 12
FwdOp: 4
NumOfEntries: 1
=================================
=================================
VlanTagFilterData
EntityID: 0x03
FilterTbl[0]: PRI 0,CFI 0, VID 13
FwdOp: 4
NumOfEntries: 1
=================================
=================================
VlanTagFilterData
EntityID: 0x04
FilterTbl[0]: PRI 0,CFI 0, VID 4000
FwdOp: 4
NumOfEntries: 1
=================================
=================================
VlanTagFilterData
EntityID: 0xffff
FilterTbl[0]: PRI 0,CFI 0, VID 9
FwdOp: 4
NumOfEntries: 1
=================================
#
`

Today, I have sfp ODF v220304, i tried, internet with vlan11 ok, but can't use vlan12, just I use iptv(set-top-box power on) will make sfp hang. Each time, I had to remove the sfp and fast connector, and then reattach it.

[image]

[image]

Do you have any suggestions for this case?

Anime4000, 2022-05-12:
that's odd, have you tried older version? that requires manual WAN VLAN ?

vuducdong, 2022-05-13:
> that's odd, have you tried older version? 

you mean, this version 210702 [ODI_210702.tar](https://github.com/Anime4000/RTL960x/blob/main/Firmware/DFP-34X-2C2/M114_sfp_ODI_210702.tar) and upgrade directly with WebUI, right? I will try tomorrow.

that requires manual WAN VLAN ?
I tried with vlan mode auto or manual transparent. The same issue.

Anime4000, 2022-05-13:
yes, 210702 firmware
this firmware not work in my fiber, but works in some users...

it as PON WAN page, different page from vlan.asp

closed 2022-09-05 by vuducdong