Anime4000/RTL960x #35: TWCGPON657 bandwidth limit to 30Mbps

ccy, 2022-05-13:
My ISP is Maxis 100Mbps package.

I am using TWCGPON657 SFP stick.  I flash the firmware to `C00R657V2801F_V1.9.0-220404.tar`.

I also follow the instruction perform necessary setup for the SFP stick.  I am able to connect to the ISP via PPPoE.

However, when perform [speedtest](https://fast.com).  The download speed cap at `30Mbps` and upload is below `1Mbps`.

Plug the fiber cable back to original Maxis/TM GPON, the speed is usual 100M/50M.

I also make sure the SFP port of network switch is set to `1000M`.

This is the Chipset of the GPON stick:
```
# cat /proc/cpuinfo
system type             : RTL8672
processor               : 0
cpu model               : 56322
BogoMIPS                : 299.00
tlb_entries             : 64
mips16 implemented      : yes
```

Any steps I didn't do right?

Anime4000, 2022-05-13:
oh that's very odd...

have you copy old ONU into to new stick? clone OMCI info to trick OLT,

also, can you print and paste your bandwidth status like this
https://github.com/Anime4000/RTL960x/blob/main/Docs/DIAG.md#bandwidth

ccy, 2022-05-14:
```
# cat /etc/version
V1.9.0-220404 -- Sun Apr 03 17:05:04 UTC 2022

# diag
RTK.0> bandwidth get egress port all
port: 0  rate:1048568
         queue: 0  apr-index: 0
         queue: 1  apr-index: 0
         queue: 2  apr-index: 0
         queue: 3  apr-index: 0
         queue: 4  apr-index: 0
         queue: 5  apr-index: 0
         queue: 6  apr-index: 0
         queue: 7  apr-index: 0
port: 2  rate:4194296
         queue: 0  apr-index: 0
         queue: 1  apr-index: 0
         queue: 2  apr-index: 0
         queue: 3  apr-index: 0
         queue: 4  apr-index: 0
         queue: 5  apr-index: 0
         queue: 6  apr-index: 0
         queue: 7  apr-index: 0
RTK.0> bandwidth get ingress port all
port: 0 rate:1048568
port: 2 rate:4194296
RTK.0> port get status port all
Port Status Speed    Duplex TX_FC RX_FC
---- ------ -----    ------ ----- -----
0    Up     1000M    Full   Dis   Dis
2    Up     1000M    Full   Dis   Dis
```

Anime4000, 2022-05-14:
bandwidth is fine...

With TM fiber, you need correct Hardware Version, then OLT give correct OMCI Parameters

Unifi HG8240H5 ONU Hardware Version:
```
flash set HW_HWVER 168D.A
```

since you using V2801F firmware on TWCGPON657, you need generate VS_AUTH_KEY

give me your ONU Model, Stick MAC Address, I make one

ccy, 2022-05-14:
This is my current setting:

```
# flash get HW_HWVER
HW_HWVER=168D.A
# flash get GPON_ONU_MODEL
GPON_ONU_MODEL=HG8240H5
# flash get ELAN_MAC_ADDR
ELAN_MAC_ADDR=6cefc6bed812
# flash get VS_AUTH_KEY
VS_AUTH_KEY=7e,c8,c9,dc,c0,23,54,85,1d,c2,32,80,36,8f,2f,af,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00
```

After reboot, my GPON stick will enter O5 state, so I assume it works.

However, my download/upload bandwidth still cap at 30M/1M

Anime4000, 2022-05-14:
try this:

```
flash set GPON_ONU_MODEL HG8240H5
flash set GPON_SN HWTC12345678
flash set PON_VENDOR_ID HWTC
flash set OMCI_SW_VER1 V5R019C00S125
flash set OMCI_SW_VER2 V5R019C00S125
flash get OMCI_OLT_MODE 1
flash get OMCI_FAKE_OK 1
flash set HW_HWVER 168D.A
flash set ELAN_MAC_ADDR 6cefc6bed812
flash set VS_AUTH_KEY 7EC8C9DCC02354851DC23280368F2FAF
```

This should work with TM, this making your stick like Huawei ONU,
Allo and TIME Fiber just works

ccy, 2022-05-14:
No luck after execute your script.  I assume these two lines has typo mistake:
```
flash get OMCI_OLT_MODE 1
flash get OMCI_FAKE_OK 1
```
I have changed to `set` when executing.

Anime4000, 2022-05-14:
my bad...

after enter these `flash set` have you tried to reboot?

ccy, 2022-05-14:
Yes, I reboot before I try after execute those `flash set` command.

ccy, 2022-05-14:
I also try to plug the GPON stick to another old model L2 switch, still the same.

Anime4000, 2022-05-14:
that's very odd, what kind of switch are you using?

ccy, 2022-05-14:
[S5860-20SQ](https://www.fs.com/sg/products/108710.html)
[DGS-1210-20](https://www.dlink.com/en/products/dgs-1210-20-20-port-gigabit-smart-managed-switch)

Anime4000, 2022-05-14:
V2801F using 1000baseX
it's works well with RB3011 and Fortigate but not RB5009

have other SFP host or PCIe SFP to test?

ccy, 2022-05-14:
I don't have RB3011 and Fortigate.  Is MikroTik ok?

Anime4000, 2022-05-15:
Yes, Mikrotik RB5009 is currently I using with ODI DFP-34X-2C2 (TM-Maxis 500/100 Plan)
[image]

RB3011 + TWCGPON657 now at my friend home, TIME 500mbps
[image]

ccy, 2022-05-15:
I have [CRS326-24G-2S+](https://mikrotik.com/product/CRS326-24G-2SplusRM).  Will try tomorrow.

Anime4000, 2022-05-15:
Yes, PON Stick have limited SFP Host support, RTL960x firmware kind picky which router and switch

ccy, 2022-05-16:
I have test all other switches I have.

Fail to detect the GPON stick:

- MikroTik [CRS326-24G-2S+](https://mikrotik.com/product/CRS326-24G-2SplusRM) cannot detect the GPON stick at all.

These switches can ping to the GPON stick but fail to ping once plug in TM fiber cable:

- [TP Link T1600G-28TS](https://www.tp-link.com/us/business-networking/smart-switch/t1600g-28ts/)

- [Dell n1148t-on](https://i.dell.com/sites/doccontent/shared-content/data-sheets/en/Documents/Dell-EMC-Networking-N1100-Series-Spec-Sheet.pdf)

- [FS S3900-48T4S](https://www.fs.com/sg/products/72946.html)

These switches can ping to GPON Stick, can establish PPPoE connection, can serve Internet but limited to 30Mbps down/1Mbps up:

- [FS S5860-20SQ](https://www.fs.com/sg/products/108710.html)

- [D-Link DGS-1210-20](https://www.dlink.com/en/products/dgs-1210-20-20-port-gigabit-smart-managed-switch)

- [TP-Link TL-SG1428PE](https://www.tp-link.com/uk/business-networking/poe-switch/tl-sg1428pe/)

Do you face the same problem on TWCGPON657?

Anime4000, 2022-05-16:
wow, you have a lot of switches...

What I know, RTL9601B, RTL9601CI, RTL9601D has 1000base-T `LAN_SDS_MODE 0` and 1000base-X `LAN_SDS_MODE 1`

Meanwhile other `LAN_SDS_MODE` value, is `SGMII` and `HiSGMII`, which is require SFP to have intermediate `MII` translation chipset to Ethernet. I afraid that not all Switch and Router support `MII`

but you can try:
```
echo 0 > /proc/lan_sds/lan_sds_cfg
echo 2 > /proc/lan_sds/lan_sds_cfg
echo 3 > /proc/lan_sds/lan_sds_cfg
```

Here my Mikrotik RB5009 + ODI DFP-34X-2C2
```
# cat /proc/lan_sds/lan_sds_cfg
lan_sds_mode = 3(SGMII MAC)
# flash get LAN_SDS_MODE
LAN_SDS_MODE=0
```
Value are not match, ODI DFP-34X-2C2 can auto nego with host with appropriate mode

However, you can buy this as `MII` translation
[image]

Or buy ODI [DFP-34X-2C2](https://www.aliexpress.com/item/1005003515662920.html) with Mikrotik [CRS305-1G-4S+IN](https://www.lazada.com.my/products/crs305-1g-4sin-mikrotik-smart-switch-10g-media-converter-4-x-sfp-10gbps-ports-dual-boot-routeros-swos-malaysia-i521078124.html) or [CCR2004-1G-12S+2XS](https://www.lazada.com.my/products/mikrotik-ccr2004-1g-12s2xs-carrier-grade-router-vpn-10g-sfp-25g-sfp28-gigabit-fiber-router-cloud-core-router-i1283220788.html), 100% compatible including 2.5Gb support!!!

ccy, 2022-05-16:
I prefer direct plug the GPON stick to my L2/L3 switch so I reduce a 12VDC power adapter to power the external device like TM's Huawei GPON.

Anime4000, 2022-05-16:
I understand, I just to make sure that TWCGPON657 have an issue or something else,
I afraid TWCGPON657 that you have is different hardware revision...

Mikrotik Router [CCR2004-1G-12S+2XS](https://www.lazada.com.my/products/mikrotik-ccr2004-1g-12s2xs-carrier-grade-router-vpn-10g-sfp-25g-sfp28-gigabit-fiber-router-cloud-core-router-i1283220788.html) and [CRS305-1G-4S+IN](https://www.lazada.com.my/products/crs305-1g-4sin-mikrotik-smart-switch-10g-media-converter-4-x-sfp-10gbps-ports-dual-boot-routeros-swos-malaysia-i521078124.html) Switch currently full supported, it has Port Extender which can support various `MII` (SGMII & HiSGMII) and base-X

ccy, 2022-05-16:
How to identify the hardware of my GPON stick?

Anime4000, 2022-05-16:
> How to identify the hardware of my GPON stick?

this you need disassemble and compare

ccy, 2022-05-17:
I try these command:
```
echo 0 > /proc/lan_sds/lan_sds_cfg
echo 2 > /proc/lan_sds/lan_sds_cfg
echo 3 > /proc/lan_sds/lan_sds_cfg
```

Once execute, the GPON stick immediately freeze and no response.  Need to unplug and plug to reboot.

This is the default value:
```
# cat /proc/lan_sds/lan_sds_cfg
lan_sds_mode = 1(Fiber 1G)
```

ccy, 2022-05-24:
I bought a new *ODI DFP-34X-2C2* GPON stick

```
$ cat /proc/cpuinfo
system type             : RTL8672
processor               : 0
cpu model               : 56322
BogoMIPS                : 299.00
tlb_entries             : 64
mips16 implemented      : yes

# cat /etc/version
V1.0-220304 --  Fri Mar 4 11:22:18 CST 2022
```
I didn't patch the firmware and direct enter my PLOAM password.  I get connect to internet but the speed still remain at 30M download / 0.5M upload.

Should I patch the firmware?  Which firmware should I patch?

Anime4000, 2022-05-24:
since you have ODI Stick, try changing LAN_SDS_MODE

0 = 1GbaseT
1 = 1GbaseX
2 = SGMII PHY
3 = SGMII MAC

ccy, 2022-05-25:
Try these modes and same issue too:

```
flash set LAN_SDS_MODE 0
reboot

flash set LAN_SDS_MODE 1
reboot

flash set LAN_SDS_MODE 2
reboot

flash set LAN_SDS_MODE 3
reboot
```

Anime4000, 2022-05-25:
if 0 until 3 produce same effect, highly chance that SFP Host having issue,

I have no idea how fix it 😅

yea, ODI have hard coded sw version,
for Telekom Malaysia OLT, doesn't care about it, only care HW Ver,

it will get O5 no matter what, since stock ISP ONU have own provisioning,

ccy, 2022-05-25:
Even if I use different `HW_HWVER`, I still can get O5 and connect to internet.

ccy, 2022-05-25:
I further debug the issue.  I found out the GPON ignore my value set to LAN_SDS_MODE.  It will also reset to 1. 

```
# flash get LAN_SDS_MODE
LAN_SDS_MODE=2

# cat /proc/lan_sds/lan_sds_cfg
lan_sds_mode = 1(Fiber 1G)
```
And this is the **/proc/kmsg** output:
```
# cat /proc/kmsg
al: 8250/16550 driver, 1 ports, IRQ sharing disabled
<6>serial8250: ttyS0 at MMIO 0x0 (irq = 49) is a 16550A
<6>8686 Ethernet driver v0.0.1 (Feb 17, 2012)[Dual Band Disable]
<4>TX[1024,64,64,64,64] , RX[256,32,32,32,64,64]
<6>eth0 (): not using net_device_ops yet
<6>eth0: RTL-8686 at 0xb8012000, 00:00:00:01:00:02, IRQ 26
<6>eth0.2 (): not using net_device_ops yet
<6>eth0.2: RTL-8686 at 0xb8012000, 00:00:00:01:00:02, IRQ 26
<6>eth0.3 (): not using net_device_ops yet
<6>eth0.3: RTL-8686 at 0xb8012000, 00:00:00:01:00:02, IRQ 26
<6>nas0 (): not using net_device_ops yet
<6>nas0: RTL-8686 at 0xb8012000, 00:00:00:01:00:02, IRQ 26
<4>>>>> Set nas0 carrier off !!!
<6>pon0 (): not using net_device_ops yet
<6>pon0: RTL-8686 at 0xb8012000, 00:00:00:01:00:02, IRQ 26
<4>RTL8686 GMAC Probing..
<4>first entry: 0, portmask 127, rx: 80232f6c
<4>nas0 -> 0x4
<4>Init RTK Driver Module....*** [RT_DBG] drivers/net/rtl86900/sdk/src//dal/rtl9602c/dal_rtl9602c_classify.c:3567: In function 'dal_rtl9602c_classify_unmatchAction_set' action 2
<4>
<4>OK
<4>Init RTK Core Dev Module....OK
<4>
<4> rtdrv_init !!! for netfilter
<4>pkt_redirect_init complete
<5>interrupt broadcaster init ok!
<4>u32 classifier
<4>nf_conntrack version 0.5.0 (479 buckets, 1916 max)
<6>ip_tables: (C) 2000-2006 Netfilter Core Team
<6>TCP cubic registered
<6>NET: Registered protocol family 17
<5>Bridge firewalling registered
<6>Ebtables v2.0 registered
<6>802.1Q VLAN Support v1.8 Ben Greear <greearb@candelatech.com>
<6>All bugs added by David S. Miller <davem@redhat.com>
<4>Realtek GPIO Driver for Flash Reload Default
<5>physmap platform flash device: 0b000000 at 14000000
<4>Luna SPI NOR FLASH G3 driver-C22017/MMIO16-1 <5>master->name physmap-flash
<5>cmd: cmdlinepart
<5>14 cmdlinepart partitions found on MTD device physmap-flash
<5>Creating 14 MTD partitions on "physmap-flash":
<5>0x000000000000-0x000000040000 : "boot"
<5>0x000000040000-0x000000042000 : "env"
<5>0x000000042000-0x000000044000 : "env2"
<5>0x000000044000-0x000000080000 : "config"
<5>0x000000080000-0x0000001cc000 : "k0"
<5>0x0000001cc000-0x000000440000 : "r0"
<5>0x000000440000-0x00000058c000 : "k1"
<5>0x00000058c000-0x000000800000 : "r1"
<5>0x000000000000-0x000000001000 : "Partition_008"
<5>0x000000000000-0x000000001000 : "Partition_009"
<5>0x000000000000-0x000000001000 : "Partition_010"
<5>0x000000000000-0x000000001000 : "Partition_011"
<5>0x000000080000-0x0000001cc000 : "linux"
<5>0x0000001cc000-0x000000440000 : "rootfs"
<4>VFS: Mounted root (squashfs filesystem) readonly on device 31:5.
<6>Freeing unused kernel memory: 100k freed
<4>re8670_open 3341
<4>omci platform attached!
<4>*** [RT_ERR] drivers/net/rtl86900/sdk/src//dal/rtl9602c/dal_rtl9602c_ponmac.c:1039: In function '_rtl9602c_ponMacGponModeV2_set'
<4>             Error Code: 0x8848
<4>
<4>*** [RT_DBG] drivers/net/rtl86900/sdk/src//dal/rtl9602c/dal_rtl9602c_classify.c:3567: In function 'dal_rtl9602c_classify_unmatchAction_set' action 1
<4>
<4>*** [RT_DBG] drivers/net/rtl86900/sdk/src//dal/rtl9602c/dal_rtl9602c_classify.c:3567: In function 'dal_rtl9602c_classify_unmatchAction_set' action 1
<4>
<3>register_gpon_evt_state: register gpon event state Successfully !
<4>priority: 0, portmask 127, rx: 80232f6c
<4>insert here, priority: 7, portmask 255, rx: c026b34c
<4>change mode to 2(SGMII PHY)
<4>port 2 assign to pon0
<4>pon0 -> 0x4
<4>priority: 7, portmask 255, rx: c026b34c
<4>priority: 0, portmask 127, rx: 80232f6c
<4>insert here, priority: 4, portmask 4, rx: 80173ca4
<4>write watchdog_flag to 0x00000001
<4>[Kthread : watchdog ] init complete!
<4>************ Watchdog Setting ****************
<4>WDT_E=1, (1-enable, 0-disable)
<4>LX(MHz)=200
<4>WDT_CLK_SC=3
<4>PH1_TO=31
<4>PH2_TO=0
<4>WDT_RESET_MODE=0
<4>**********************************************
<6>device eth0 entered promiscuous mode
<6>br0: port 1(eth0) entering learning state
<4>drivers/net/rtl86900/sdk/src/rtk/pon_led.c:1152 led sw blink init complete!
<6>br0: port 1(eth0) entering forwarding state
<4>omci_set_flow_map_ds_queue: Failed to call rtk_qos_schedulingQueue_get
<4>change mode to 1(Fiber 1G)
```
There are some error messages, but I am not sure it matter.

Anime4000, 2022-05-25:
Yes, ODI Stick has Auto Negotiation, it will auto switch to find right LAN_SDS_MODE

in my RB5009:
```
# cat /proc/lan_sds/lan_sds_cfg
lan_sds_mode = 3(SGMII MAC)
# flash get LAN_SDS_MODE
LAN_SDS_MODE=0
```

It seem your device only accept 1000baseX

ODI has another mode, `flash set LAN_SDS_MODE 7`, it will force SGMII mode, try that

tdmadam, 2022-05-25:
Have you checked the VLANs and UNI port mapping? 30Mbps/1Mbps makes sense for VoIP or management VLANs, but not for Internet services. In some countries, users also saw low speeds until they marked the appropriate priority bits(PCP).

Turn off OMCI_FAKE_OK, turn on OMCI readable log(par) and see if you see many errors. 
Is the IP address in the same range like original Maxis/TM ONT?
Make sure that the hardware, software version and other parameters are exactly the same as the original.
What is your OLT? `omcicli mib get 131` Does it match OLT mode?

There is a nice picture of the Maxis network showing VLANs and port mapping:

https://github.com/Anime4000/RTL960x/issues/17#issuecomment-1092687398

Anime4000, 2022-05-25:
he seem using same fiber as I do

here `omcicli` for reference
```
# omcicli mib get 131
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
OltG
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityId: 0x00
OltVendorId: 0x414c434c
EquipId:
Version:
ToDInfo:
        Sequence number of GEM superframe: 0x0
        Timestamp: secs 0, nanosecs 0
=================================
```

```
# omcicli mib get 84
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
VlanTagFilterData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityID: 0x1202
FilterTbl[0]: PRI 0,CFI 0, VID 621
FwdOp:  0x10
NumOfEntries: 1
=================================
=================================
EntityID: 0x1203
FilterTbl[0]: PRI 0,CFI 0, VID 821
FwdOp:  0x10
NumOfEntries: 1
=================================
=================================
EntityID: 0x1204
FilterTbl[0]: PRI 0,CFI 0, VID 822
FwdOp:  0x10
NumOfEntries: 1
=================================
=================================
EntityID: 0x120b
FilterTbl[0]: PRI 0,CFI 0, VID 621
FilterTbl[2]: PRI 0,CFI 0, VID 821
FilterTbl[4]: PRI 0,CFI 0, VID 822
FwdOp:  0x10
NumOfEntries: 3
=================================
=================================
EntityID: 0xe102
FilterTbl[0]: PRI 0,CFI 0, VID 209
FwdOp:  0x10
NumOfEntries: 1
=================================
=================================
EntityID: 0xe10b
FilterTbl[0]: PRI 0,CFI 0, VID 209
FwdOp:  0x10
NumOfEntries: 1
=================================
```

ccy, 2022-05-26:
How to mark the appropriate priority bits(PCP)?

This is my omcicli info:
```
# omcicli mib get 131
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
OltG
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityId: 0x00
OltVendorId: 0x414c434c
EquipId:
Version: 
ToDInfo:
        Sequence number of GEM superframe: 0x0
        Timestamp: secs 0, nanosecs 0
=================================
```

I switch to TM Unifi now, so this is the vlan info:
```
# omcicli mib get 84
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
VlanTagFilterData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityID: 0x1102
FilterTbl[0]: PRI 0,CFI 0, VID 500
FwdOp:  0x10
NumOfEntries: 1
=================================
=================================
EntityID: 0x1103
FilterTbl[0]: PRI 0,CFI 0, VID 209
FwdOp:  0x10
NumOfEntries: 1
=================================
=================================
EntityID: 0x110b
FilterTbl[0]: PRI 0,CFI 0, VID 209
FilterTbl[2]: PRI 0,CFI 0, VID 500
FwdOp:  0x10
NumOfEntries: 2
=================================
=================================
EntityID: 0xe102
FilterTbl[0]: PRI 0,CFI 0, VID 400
FwdOp:  0x10
NumOfEntries: 1
=================================
=================================
EntityID: 0xe103
FilterTbl[0]: PRI 0,CFI 0, VID 209
FwdOp:  0x10
NumOfEntries: 1
=================================
=================================
EntityID: 0xe10b
FilterTbl[0]: PRI 0,CFI 0, VID 209
FilterTbl[2]: PRI 0,CFI 0, VID 400
FwdOp:  0x10
NumOfEntries: 2
=================================
```

Anime4000, 2022-05-26:
@ccy wow, this is new...

during my testing `OMCI_TM_OPT` on my fiber doesn't do anything, I have been wondering what is this...
witn TM-Maxis currently I use, `0`, `1`, `2` give same results, so that time I not sure this can cause speed slow down:
[image]

I'll update this new info

Anime4000, 2022-05-26:
@ccy
I have tried swapping Maxis and Unifi PPPoE Account.

Home with Unifi 100/50 subscription:
I swap with 800/200 pppoe account and do get 800+ in speedtest

Home with Maxis 500/100 subscription:
I tried with 800/200 pppoe account and speed still same...

This mean:
For Unifi, TM OLT didn't set bandwidth limit, only limited via PPPoE

For Maxis under TM fiber, TM OLT do set bandwidth limit, using higher speed PPPoE account will get speed limit at OLT (L2)

but this varied place to place, how TM configure.

like yours, OMCI_TM_OPT has effect your speed meanwhile mine remain same effect with any value.

but you can try access `diag` utility in telnet and try play with

ccy, 2022-05-27:
I bought the GPON stick from [here](https://item.taobao.com/item.htm?spm=a1z0d.6639537.1997196601.4.e6427484KCyH56&id=673661619814)

Anime4000, 2022-05-28:
@ccy 

tonight I tried again `OMCI_TM_OPT` `0`, `1`, `2`

Results:
`OMCI_TM_OPT 0`
[image]

`OMCI_TM_OPT 1`
[image]

`OMCI_TM_OPT 2`
[image]

with `0` (priority controlled) mode, my speed never reach 500Mbps, always under `400` or worst

with `1` (rate controlled) mode, my speed can over 500Mbps

with `2` (both) mode, my speed still can over 500Mbps, but I stick with `1`

MSufianS, 2022-11-23:
> CCR2004-1G-12S+2XS

May I know, Allo ISP working with this hardware ODI DFP-34X-2C2 with Mikrotik CRS305-1G-4S+IN as you mention?

and what type SFP need to buy GPON, XPON or EPON?

ccy, 2022-12-30:
@Anime4000 , it is me again facing speed issue.

Previously I subscribed to 100M/50M maxis package, and the download speed from the GPON is 30M.  Then I set `OMCI_TM_OPT=0` and I can get `100M/50M` bandwidth.

Now I upgrade my plan to 500M/100M, and my speed stay in `100M/50M`, using the GPON from the ISP, I can get `500M/100M`.  Any idea?

Anime4000, 2022-12-30:
have you try other `OMCI_TM_OPT` value? `1` or `2` ?
I am using `2` as Maxis 500/100 under Unifi line

ccy, 2022-12-30:
Yes, I have tried 0, 1, 2.  All speed remain `100M/50M`.

Anime4000, 2022-12-30:
try change the ONU information to something else, I afraid OLT locks on something, also try update to newer firmware

ccy, 2022-12-31:
Where to get latest firmware?

This is my current firmware version:

```
# flash get OMCI_SW_VER1
OMCI_SW_VER1=V1.0-220414

# flash get OMCI_SW_VER2
OMCI_SW_VER2=V1.0-210702
```

Anime4000, 2022-12-31:
here
https://github.com/Anime4000/RTL960x/tree/main/Firmware/DFP-34X-2C2

maybe you can try change OMCC Version

ccy, 2022-12-31:
I update **GPON_SN=HWTC**, and I get full bandwidth speed now:
```
flash set GPON_SN HWTC
flash set OMCI_TM_OPT 0
```

Anime4000, 2022-12-31:
your previous ONU is ALCL/Nokia?
It's best to use Huawei OMCI Information as most widely use for Universal ONU (brand interchangeable between OLT and ONU)

ccy, 2022-12-31:
My previous ONU is huawei HG8240H5