Anime4000/RTL960x #36: C00R657V00B16 --  Wed Mar 3 02:59:49 UTC 2021 in no-name stick from AliExpress

Kitsok, 2022-05-18:
Hello!

I've just bought SFP module named CM5671A.
I've log in using admin/system and found that /etc/version contains C00R657V00B16

Do you need this firmware for your collection?

tdmadam, 2022-05-18:
@Kitsok 
Welcome!

Can you identify the processor?
`# cat /proc/cpuinfo`

If it's Realtek do you know which SoC RTL9601CI or RTL9601D

Anime4000, 2022-05-19:
The version look like TWCGPON657 stick
I would like to add `C00R657V00B16` into the collection, but need to verify the stick is RTL9601CI or RTL9601D

Kitsok, 2022-05-19:
I've identified SoC by looking at it - it's RTL9601C1, but  /proc/cpuinfo says different:
```
# cat /proc/cpuinfo 
system type             : RTL8672
processor               : 0
cpu model               : 56322
BogoMIPS                : 112.23
tlb_entries             : 64
mips16 implemented      : yes
# 
```
As in /proc/mtd I see 3 sets (k0/r0, k1/r1, linux/rootfs), I've added all three in the archive.

Meanwhile can you please explain what is linux/rootfs? Is it mirror partition or a real 3rd set? 
[C00R657V00B16.zip](https://github.com/Anime4000/RTL960x/files/8725057/C00R657V00B16.zip)

tdmadam, 2022-05-19:
The GPON stick have two sets firmwares k0,r0 or k1,r1. During the boot it passes to kernel via command line partition map and indicates which one of two firmware sets is the active one (linux/rootfs)

```
mtdparts0=mtdparts=rtk_spi_nor_mtd:256K(boot),8K(env),8K(env2),240K(config),1408K(k0)ro,2432K(r0)ro,1408K(k1),2432K(r1),4K@0ro,4K@0ro,4K@0ro,4K@0ro,1408K@512K(linux),2432K@1920K(rootfs) root=31:5
mtdparts1=mtdparts=rtk_spi_nor_mtd:256K(boot),8K(env),8K(env2),240K(config),1408K(k0),2432K(r0),1408K(k1)ro,2432K(r1)ro,4K@0ro,4K@0ro,4K@0ro,4K@0ro,1408K@4352K(linux),2432K@5760K(rootfs) root=31:7
```

The Realtek GPON SoC RTL960x uses the CPU block from the RTL8672. It's weird, but I guess it must be more cost effective for them to do so.

tdmadam, 2022-05-19:
> The version look like TWCGPON657 stick

You're right, looking at the firmware size, it appears that this GPON stick has 16 MiB of flash memory.

Anime4000, 2022-05-19:
by the name CM5671A, it's look like a replacement for Huawei MA5671a