Anime4000/RTL960x #37: Forwarding EAPOL frames through the kernel network bridge

devicelocksmith, 2022-05-28:
It seems like RTL960x 2.6.30 kernel is filtering EAPOL frames due to this kernel patch: `[Bridge] [RFC] bridging: don't forward EAPOL frames` 
My ISP requires 802.1x authentication. ISP-provided ONT is using a newer kernel with `echo 8 > /sys/class/net/brXXX/bridge/group_fwd_mask` workaround, however it is not available on older kernel.
Has anyone found a way to pass such frames on this chipset?

Anime4000, 2022-05-28:
which Stick are you using? if ODI brand, you can try ask them.
we don't have RTL source code here, kind hard to patch it

devicelocksmith, 2022-05-29:
I am using ODI stick. It doesn't look like there is a way to get this to work with 2.6 kernel without undoing the patch. 
I have ordered Lantiq stick from FS to see if that works better, if not, I'll be looking into patching kernel in memory to modify the EAPOL frame signature to something else, since I don't have kernel SDK to re-compress kernel.

Anime4000, 2022-05-29:
i see, have you try newer firmware just in case?

devicelocksmith, 2022-06-05:
Yes, I have tried that, it did not work. I've also compiled wpa_supplicant for the stick and tried sending authentication on pon0, but it did not work. The stick remains in O5 with just an omci channel open, and no VLANs set up.

marcinkuczera, 2022-06-09:
Hello, can you please dump some frames of EAPOL bi-di communication including full ethernet header ?
I'am a small (for now) vendor of Realtek based GPON ONTs, including SFPs. I mean, we make firmware. And the firmware for 9601C/9601D is based on most recent SDK for this chipset family that uses kernel 3.18.x.

However, EAPOL blocking is done on switch-chip level, not on kernel level.
It seems to be case of initialization of default behavior that drops frames without information in L2 table (from PON port side).
Last time I guess in similar way I have unlocked LLDP in case of EPON firmware, but for GPON initialization process of switch-chip is very similar.

So I guess, I can enable EAPOL in my devices.

devicelocksmith, 2022-06-16:
Here is a link to [eap-tls exchange](https://mega.nz/file/a2xmWLwC#moK09Pk1FfLHG9heKWHD6238gRxMbLY0JniCnGHoTWE)

Anime4000, 2022-06-16:
> As mentioned in other thread - emulation of 4 ports might not be easy due to conflict of untagged traffic on different physical ports. Full bridge and IGMP - works fine. Lot of ISPs in Poland use our devices.

but allow to user to enable 4-port Emulation in WebGUI or Command-Line, disable by default.

it would be nice allow user to custom VLAN Mapping from PON Interface

devicelocksmith, 2022-06-18:
@marcinkuczera
Not sure if that is what you are looking for, but it looks like RMA is already set to forward the correct multicast LSB
```
RTK.0> rma dump
RMA 01-80-C2-00-00-xx
trail action
00    Forward       01    Drop          02    Drop
03    Forward       04    Forward       05    Forward
06    Forward       07    Forward       08    Forward
09    Forward       0a    Forward       0b    Forward
0c    Forward       0d    Forward       0e    Forward
0f    Forward       10    Forward       11    Forward
12    Forward       13    Forward       14    Forward
15    Forward       16    Forward       17    Forward
18    Forward       19    Forward       1a    Forward
1b    Forward       1c    Forward       1d    Forward
1e    Forward       1f    Forward       20    Forward
21    Forward       22    Forward       23    Forward
24    Forward       25    Forward       26    Forward
27    Forward       28    Forward       29    Forward
2a    Forward       2b    Forward       2c    Forward
2d    Forward       2e    Forward       2f    Forward
```
The correct LSB is 03 and it is already set to `forward` by default.
Could it be filtering by ethertype? I am not finding how to check or set that. The correct ethertype for EAPoL is `0x888e`

marcinkuczera, 2022-06-20:
yes, ethertype is most probably used for filtering..

git-nerd, 2022-08-23:
@devicelocksmith can you share your wpa_supplicant binary here?