Anime4000/RTL960x #41: What is OMCI_CUSTOM_RDP?

ccy, 2022-06-16:
Previously, I have a **DFP-34X-2C2** using software version **V1.0-220304**.  I configure the GPON to access to Maxis or Unifi with these minimum configuration:

```
flash set GPON_PLOAM_FORMAT 1
flash set GPON_PLOAM_PASSWD <hex code>
flash set OMCI_TM_OPT 0
```
I purchase another **DFP-34X-2C2** recently using software version **V1.0-220414** but perform the above setting doesn't allow me to get thru PPPoE authentication.  The GPON do show **O5** state.

After compare and try all OMCI value of both GPON device one by one, the new GPON works with the additional **OMCI_CUSTOM_RDP** setting:

```
flash set GPON_PLOAM_FORMAT 1
flash set GPON_PLOAM_PASSWD <hex code>
flash set OMCI_TM_OPT 0
flash set OMCI_CUSTOM_RDP 4
```

I couldn't find any information about **OMCI_CUSTOM_RDP**.  Do you know what is this for?

crrodriguez, 2022-06-16:
The firmware startup scripts refer to it as "custom route" ( I do not know in this context what it means)
and the default on firmware 220414 is 4...

ccy, 2022-06-17:
Yes. Both stick show same **OltVendorId**.  Any clue?

Anime4000, 2022-06-17:
I also use `220414`
[image]
```
# flash get OMCI_CUSTOM_RDP
OMCI_CUSTOM_RDP=4
```
and OLT is `0x414c434c` = Nokia

if I factory reset, `OMCI_CUSTOM_RDP` still same value `4`

every OLT has different effect, mine need `HW_HWVER` then can be use, if I use Nokia `HW_HWER`, I need put other OMCI info to work...

So, I need to know what your OLT id is, Huawei or ZTE or Ericsson or something else...

ccy, 2022-06-17:
`OltVendorId: 0x414c434c`

Anime4000, 2022-06-17:
oh damn, same Nokia OLT

have you tried change RDP value to 3 or something else? see what kind of effect,

it seem your OLT configured differently than mine...

ccy, 2022-06-18:
I thought OLT is device of ISP?  I have no control of it.

crrodriguez, 2022-06-18:
> I thought OLT is device of ISP? I have no control of it.

well ..yes the olt is at the fiber operator side.. @Anime4000 has the same ISPs as you, but because reasons ISPs are not homogenous, the have OLTs configured diffferently or running different brands, This is in most cases due to inertia (if it aint broke dont fix it) or they have special requirements etc.. this usually translates into a shit show and larger long term mainteniance costs  ..the world aint perfect out there..

ccy, 2022-06-18:
I perform reset to factory on my new GPON and notice the value `OMCI_CUSTOM_RDP` is 4.

I think the vendor ship a GPON with dirty configuration.

Anime4000, 2022-06-18:
@ccy
I see, it seem good to backup the key and factory reset after receiving the Stick.

yet, we still wondering with RDP parameters...

I feel OMCI_CUSTOM_RDP is like how many virtual LAN port? maybe not...

no information on this...

rajkosto, 2022-11-18:
OMCI_CUSTOM_BDP controls which bdp_xxx.so/.ko plugins get loaded
OMCI_CUSTOM_RDP controls which rdp_xxx.so/.ko plugins get loaded
OMCI_CUSTOM_MCAST controls which mc_xxx.so/.ko plugins get loaded
OMCI_CUSTOM_ME controls which me_xxx.so/.ko plugins get loaded
each of these "plugins" makes a small tweak to some internal function/"feature", and getting the values of these flags right might be important
additionally, the DEVICE_TYPE value controls a lot too, 0 = bridge, 1 = router, 2 = hybrid, and unless it is set to bridge, a lot of things are skipped (like making pon0 iface and adding it to bridge), its not just for reporting to OLT

rajkosto, 2022-12-14:
I'm maintaining a list of all the available features (described as function names inside the modules) here:
https://gist.github.com/rajkosto/79034a1f7b3de3f40edf50ffbd8396b0