Anime4000/RTL960x #43: ODI running 220414 unresponsive after successful connection

devicelocksmith, 2022-06-21:
Once ODI stick on 220414 gets connected and I can pull DHCP, within seconds the CPU part of the stick stops responding - ssh session freezes, no response to web UI, the CPU part acts like it's dead - not even responding to ARP frames. Is anyone else seeing this?

The switch part of the stick continues to work fine and my Internet connection remains up, but I can no longer monitor fiber connection stats, stick temperature, etc, which is annoying. Even more annoying is the fact that the port shutdown on my SFP host does not cut the power to the stick, so I can't even reboot it if I need to.

akw28888, 2022-07-04:
I also got this issue when using untag mode on the stick.
There is no issue with transparent mode.

git-nerd, 2022-08-06:
I have the same issue but only after changing default management ip. I opened up another issue for this just now.