Anime4000/RTL960x #47: DFP-34X-2C2 v210702

Karetov, 2022-07-05:
Hello. 
With a successful connection, VLAN 10 is created, which I cannot edit.
I reset the ODI settings, disconnected from the network, then created my VLAN 10 connection, but after connecting, another VLAN 10 connection appeared next to it. 
On the tab with editing VLAN parameters, VLAN 10 created (apparently) by the provider is not displayed. 
Can I edit this connection somehow
[image]

Anime4000, 2022-07-05:
if got LOID Status Wrong or `O0`
high chance that your MAC_KEY is wrong

Karetov, 2022-07-05:
The problem with Loidstatus on all firmware except 210702. The MAC address on the ODI is registered on the provider's terminal. The question is more about VLAN

Anime4000, 2022-07-05:
This one I not really sure, 

the PON WAN, only one VLAN can be Port Mapping

when you add another PON WAN entry and tick Port Mapping, it will cause previous mapping unticked

I not sure that PON WAN is meant for PVID or something else...

Have you try not to tick Port Mapping and set VLAN at router side?

Karetov, 2022-07-05:
Perhaps I don't quite understand you, I'm sorry. Initially, I had a question about the nas0_2 connection, which is added by itself, after a successful connection and the status 05. And I can't edit it in any way and even see it in the settings, only on the screen shown in the screenshot

Karetov, 2022-07-10:
Everything worked out, it was in the vendor's parameters, as you said, thank you!

Is it possible to assign different vlan IDs on the SFP module. for VoIP and Internet?

Karetov, 2022-07-14:
Hello
I tried to update the MAC_KEY after changing the MAC address, but I get this response. Please tell me what the problem may be

# flash set MAC_KEY 96a3c*****645
[ERR] Warning, unknown dir name: MAC_KEY
[ERR] Warning, unknown dir name: MAC_KEY
GET fail.

Anime4000, 2022-07-14:
oh, which version are you using?

Karetov, 2022-07-14:
v210702

Karetov, 2022-07-15:
Figured it out
On this version, apparently, MAC_KEY is not used

Karetov, 2022-07-23:
@Anime4000 Hello

I managed to start the internet on the hybrid firmware M114_sfp_ODI_hybrid_220527.tar, but apparently I need M114_sfp_ODI_220304.tar or M114_sfp_ODI_Vlan_220414.tar to start VoIP. The problem is that on these firmware I can't start the internet and the VLAN tab is displayed like this
[image]

Anime4000, 2022-07-23:
@Karetov 
try `omcicli mib get 84` in telnet, past here the results

that command will show available VLAN

Karetov, 2022-07-23:
[image]

Karetov, 2022-07-23:
[image]
[image]

Anime4000, 2022-07-24:
try each VLAN 10,20,30,32,50,60

Karetov, 2022-07-24:
I know for sure that the Internet is 30, telephony is 10, IPTV is 20.

I wonder why the interface is not fully displayed on the firmware for the VLAN

Karetov, 2022-09-13:
The problem is solved, VoIP works on the firmware M114_sfp_ODI_220817.tar

closed 2022-09-13 by Karetov