Anime4000/RTL960x #49: How to change 'OltVendorId'

SMSegura, 2022-07-07:
### Discussed in https://github.com/Anime4000/RTL960x/discussions/48

<div type='discussions-op-text'>

<sup>Originally posted by **SMSegura** July  7, 2022</sup>

Hello.
My Internet provider is Movistar, and it uses the VLans 6:1 for Data and 3:4 for VoIP. With these VLans, I perfectly raise Data and Voice with a ZIZA OP151S module.
I acquired the DFP-34X-2C2 module and when I execute the command:

'# omcicli mib get 84
XXXXXXXXXXXXXXXXXXXXXXX
VlanTagFilterData
XXXXXXXXXXXXXXXXXXXXXXX
' ==================
EntityID: 0x1102
FilterTbl[0]: PRI 0,CFI 0, VID 832
FwdOp:  0x10
NumOfEntries: 1
' ==================
' ==================
EntiryID: 0x1103
FilterTbl[0]: PRI 0,CFI 0, VID 3
FwdOp:  0x10
NumOfEntries: 1
' ==================
' ==================
EntiryID: 0x110b
FilterTbl[0]: PRI 0,CFI 0, VID 3
FwdOp:  0x10
NumOfEntries: 1
' ==================

The Data VLan is 832, which is the one used by Orange. I configure Data with the 832 but I can't get a stable connection.

I have seen that if I run:

'# omcicli mib get 131
XXXXXXXXXXXXXXXXXXXXXXX
OltG
XXXXXXXXXXXXXXXXXXXXXXX
' ==================
EntityId: 0x00
OltVendorId: 0x414c434c
EquipId:
Version:
ToDInfo:
        Sequence number of GEM superframe: 0x0
        Timestamp: secs 0, nanosecs 0
' ==================

the OltVersionId is (0x414c434c) ALCL I am trying to change it to HWTC with the command:
  'omcicli mib set 131, OltG, OltVendorId, 0x48575443'
 but it doesn't save the changes.

Am I missing something to do?

I'm trying to make the OLT believe that I'm connecting to HWTC.

thanks for any help

</div>

Anime4000, 2022-07-07:
`OltVendorId` tell which OLT are you connected to, similar to Huawei MA5671a OpenWRT Firmware show far end OLT brand Id.

You only can change `PON_VENDOR_ID` to `HWTC`

you need copy complete ONT information to ODI Stick to fool your ALCL OLT...

Nokia/Alcatel (ALCL) OLT is very troublesome to get connected, their OLT can give fake `O5` making troubleshoot even harder

renatogalera, 2022-07-08:
I'm live in Brazil and use provider Vivo, which is a telefonica company. Same as Movistar.

The system is pretty much the same, VLAN 10 for Internet, 20 for IPTV and 30 for Voice.

I can't connect either, is there a way to debug and extract logs from these attempts?

The ONT is MitraStar. I try this config.

flash set GPON_PLOAM_FORMAT 0
flash set GPON_PLOAM_PASSWD $HEXPASS
flash set GPON_ONU_MODEL GPT-2741GNAC-N1
flash set GPON_SN MSTCXXXXXXXX
flash set PON_VENDOR_ID MSTC
flash set OMCI_FAKE_OK 1
flash set DEVICE_TYPE 0
flash set OMCI_OLT_MODE 1

renatogalera, 2022-07-08:
I get OLT info

OLT-G: classId:131, instanceId:0(0x0)
         Olt Vendor Id:0x414c434c
         Equipment id:0x2020202020202020202020202020202020202020
         Version:0x0408040000010000000000000000
         Time of day information:0x0000000000000000000000000000

Anime4000, 2022-07-09:
> I got wrong password ploam and status remains 5.

That's draw back when ISP using Nokia OLT...

I also under Nokia OLT, I just set `HW_HWVER` of Huawei then I got internet bu using SFU type firmware

renatogalera, 2022-07-09:
For example, my GPON SN is MTSC1087ac31g

What I find strange is that the registry is merged.

Serial Number:0x4d535441087ac31g

4d53544 = MTSC in Hexadecimal
1087ac31g = plain text

Follow my ONT configuration.

[ont-config.txt](https://github.com/Anime4000/RTL960x/files/9077329/ont-config.txt)

akw28888, 2022-07-09:
> For example, my GPON SN is MTSC1087ac31g
> 
> What I find strange is that the registry is merged.
> 
> Serial Number:0x4d535441087ac31g
> 
> 4d53544 = MTSC in Hexadecimal 1087ac31g = plain text
> 
> Follow my ONT configuration.
> 
> [ont-config.txt](https://github.com/Anime4000/RTL960x/files/9077329/ont-config.txt)
> 
> ONUG: classId:256, instanceId:0(0x0) vendorId:1297306691(0x4d535443) version:0x4750542d32373431474e41432d4e Serial Number:0x4d535441087ac31g trafficManagementOption:2(0x2) deprecated:0(0x0) batteryBackup:0(0x0) adminstrativeState:0(0x0) operationalState:0(0x0) onuSurvivalTime:0(0x0) logicalOnuId: logicalPassword: credentialsStatus:0(0x0) extendedTcLayerOptions:1(0x1)
> 
> based on that information, i tried use GPT-2741GNAC-N as HW_HWVER. But, didn't work either

Try to setup OMCI_SW_VER1 and VER2 to 100WVK0b1

renatogalera, 2022-07-09:
> Try to setup OMCI_SW_VER1 and VER2 to 100WVK0b1

Hi,

I had already tried, strange that both values are reset after reboot. All others are persistent

Anime4000, 2022-07-09:
do this before changing SW VER
`flash set OMCI_OLT_MODE 21`
this will make **segmentation fault** and manage to prevent default value override

renatogalera, 2022-07-09:
> do this before changing SW VER `flash set OMCI_OLT_MODE 21` this will make **segmentation fault** and manage to prevent default value override

Tks. Now all data is persistent.

But I can't connect, thanks for the help guys. I'll stop trying, I'll continue studying and if I find an alternative, I'll comment here.

renatogalera, 2022-07-09:
I found out on a Brazilian [forum](https://forum.adrenaline.com.br/threads/banda-larga-vivo-xdsl-vdsl-fibra-optica-hfc.594503/post-1075980405) that the GPON SN needs to be like this >

4d535441087ac31g

hex + plain text

However, the fields have a character limit, can I insert this amount of characters in GPON_SN?

tdmadam, 2022-07-10:
> I found out on a Brazilian [forum](https://forum.adrenaline.com.br/threads/banda-larga-vivo-xdsl-vdsl-fibra-optica-hfc.594503/post-1075980405) that the GPON SN needs to be like this > [...]

Verify your Vendor ID again, hex to ASCII 4d535441 is MSTA

The serial number is in the form 4 ASCII + 8 Hex
so it should be

flash set PON_VENDOR_ID MSTA
flash set GPON_SN MSTA087ac31g

Someone mentioned on this forum that after 10 failed start attempts the SFP switches firmware to the backup version?
@Anime4000 @stich86
Is this only for Ubiquiti, or does it also apply to ODI stick?

renatogalera, 2022-07-10:
> Verify your Vendor ID again, hex to ASCII 4d535441 is MSTA
>
> The serial number is in the form 4 ASCII + 8 Hex so it should be
>
> flash set PON_VENDOR_ID MSTA flash set GPON_SN MSTA087ac31g
>
> Someone [...]

Hi tdmadam, for security I change my GPON_SN.

stich86, 2022-07-10:
> Verify your Vendor ID again, hex to ASCII 4d535441 is MSTA
>
>
>
> The serial number is in the form 4 ASCII + 8 Hex
>
> so it should be
>
>
>
> flash set PON_VENDOR_ID MSTA
>
> flash set GPON_SN MSTA087ac31g [...]

Only Ubiquiti :(

renatogalera, 2022-07-17:
Hi guys.

I was able to connect with the G-010S-P and carlitoxx v1 firmware

I used the tutorial, anyone from São Paulo Brazil can use this howtow.

https://valero-torres.es/2018/09/08/sfp-gpon-de-carlitoxxpro-funcionando-sobre-ftth-de-orange/

`
/etc/init.d/sys.sh

         oem-generic)
                 uci set sys.target.name=oem-generic
                 uci set sys.features.mib_customized=1
                 uci set sys.features.dying_gasp_enabled=0
                 uci set sys.features.suppress_power_level=0
                 uci set sys.features.eeprom_sync_option='0x01'
                 uci set sys.mib.vendor_id='MTSC'
                 uci set sys.mib.ont_version='GPT-2741GNAC-N'
                 uci set sys.mib.equipment_id='MSTCXXXXXXXX\0\0\0\0'
`