Anime4000/RTL960x #51: DFP-34X-2C2 - strange multicast on ethernet

dominiaz, 2022-07-10:
I have 2Mbit/s traffic all the time on Jazztel ISP in Spain.

I have configured vlan1074 on sfp-sfpplus1 on RB4011. I am getting Internet over DHCP Client.

On sfp-sfplplus1 i see strange multicast traffic (using MT torch).

src: 10.179.0.10 dst: 239.192.250.254

What is it? How to block that traffic?

skon77, 2022-07-10:
RB4011 (Terminal):

/ip firewall raw
add action=drop chain=prerouting src-address=10.179.0.10

longthanhtran, 2023-11-09:
I think you can try with `diag gpon show ds-flow` to identify the downstream flow with multicast traffic. For example, 
```sh
RTK.0> gpon show ds-flow
============================================================
     GPON ONU MAC D/S Flow Status
Flow ID | GEM Port | Type | Multicast | AES
      0 |      704 |  ETH |           |
      1 |      705 |  ETH |           |
      2 |      706 |  ETH |           |
      3 |      707 |  ETH |           |
      4 |     4095 |  ETH |           |
      5 |     4094 |  ETH |         * |
     64 |       37 | OMCI |           |
============================================================
```
Then remove the downstream flow via its id `diag gpon del ds-flow flow-id 5` (per above result) and verify again.

In my case, the traffic of iptv gone completely so I think it will be back once the sfp reboot / reconnect.