Anime4000/RTL960x #54: Mikrotik and gpon onu

shpr0t, 2022-07-20:
Hello, help with router os mikrotik setup.
how to configure to get access 192.168.1.1 web.
Dfp 34g-2c2 gpon onu  
sprut0x81gmail.com     my mail
thank you

Anime4000, 2022-07-21:
With default mikrotik redefine config, do this

1. Go to Bridge > Ports, change `sfp-sfpplus1` to `ether1`
[image]
2. Go to Interface > Interface List, add `sfp-sfpplus1` as WAN group
[image]
3. Go to IP > Addresses, add IP Address `192.168.1.100/24` for `sfp-sfpplus1`
[image]

Now you can access `192.168.1.1` from `192.168.88.0/24` subnet via NAT magic

skon77, 2022-07-21:
> how to configure to get access on web interface?

My configuration for ROS7 on RB5009

1. Check that SFP1 is not in Bridge
2. In "IP / Interface List" add SFP1 to WAN group
3. In "IP / Address List" set "Address" as 192.168.1.2, "Network" as 192.168.1.1 and "Interface" as SFP1
4. In "IP / NAT rule" add "Chain" as <srcnat>, "dst.address" as 192.168.1.1, "Out.interface" as SFP1 and in "Action" tab set "Action" as <src-nat> and "To Addresses" as 192.168.1.2

shpr0t, 2022-07-27:
one more question:
obligatory presence of an optical cable (fiber) to the module SFP for connection to 192.168.1.1 ?

shpr0t, 2022-07-30:
received access, without optics does not connect.
thank you all

rajkosto, 2022-11-18:
yes, this is a mikrotik problem only, it will not pass packets or try to establish link if the SFP module is reporting "RX Loss", which the stick does without a fiber connected.
if you use a TP-Link media converter or something for the SFP stick, you can access it's webgui even without fiber in.

charlez79, 2023-04-07:
That can be resolved connecting the fiber to another SFP. That way i can get the module up and running anyway.