Anime4000/RTL960x #55: Performance of this onu system

zhliji2, 2022-07-21:
I have many 200M adsl, I run many of this [onu](https://item.taobao.com/item.htm?spm=a1z09.2.0.0.67002e8dCrqJJr&id=597031866488) on huawei s9306 switch, I need run 4 of 200M adsl account on one onu module, but benmark bandwith
[image]
, but this ssh to onu, cpu idel is 0
[image]

ping is lost package
[image]

How can i to performance it, if need run env, i can to give it to performace.

Anime4000, 2022-07-22:
have you try: `OMCI_TM_OPT` 0 or 1 or 2
2 is default

zhliji2, 2022-07-25:
setting OMCI_TM_OPT is 2 this bandwidth is
[image]

and package is lost
and setting OMCI_TM_OPT is 0 this bandwidth is
[image]

adn setting OMCI_TM_OPT  is 1， it faild to pppoe。

default 2 is bandwidth is better than 0， But both of them is package is losting，
on the onu system，see the cpu used
[image]

cpu idle is 0， why cpu nice is too high

Gouster4, 2022-09-23:
Did you tried flashing firmware drom this github repo? I had speed issue with firmware that came preinstalled with my device. Firmware from this repo fixed e everything.

Anime4000, 2022-09-23:
Yes, `220817` fixed most network performance
when do 2.5Gb Mode, need do `echo > /etc/config/fix_speed` to fix low upload speed