Anime4000/RTL960x #57: CIG G-97S

moriel5, 2022-07-24:
I was wondering about this tranciever for a long time, since I could find absolutely no documentation, however according to [this post](https://forum.mikrotik.com/viewtopic.php?p=771961), it would appear that it has a Realtek RTL9601CI.

Anime4000, 2022-07-24:
since it using RTL9601CI SoC, you can flash with V2801F firmware, I have tested V2801F firmware on TWCGPON657 because both are using RTL9601CI SoC

moriel5, 2022-07-24:
That would certainly be interesting.

Once I have the budget for it, I'll purchase one from our network provider (they have a flat one-time fee for any tranciever they sell, which while expensive, is cheaper than ordering a G.PON tranciever from abroad, whether that be from eBay, AliExpress or somewhere else.

The only thing cheaper is to purchase an ONT, such as the ZTE  F601.

shuher21, 2023-10-23:
I had flashed my CIG G-97S/Nokia G-010S-Q with full dump "C00R657V2801F_V1.9.0-220404.bin" from working stick post "https://github.com/Anime4000/RTL960x/issues/155"
But after flashing I got a brick.

Any chance of having a modified V2801F firmware, running on CIG G-97S/Nokia G-010S-Q ?