Anime4000/RTL960x #58: DFP-34X-2C2 uplink poor performance

dexxter00, 2022-08-02:
Hi! Another successful story, but with some problems!
I'm using DFP-34X-2C2 with 220414 firmware. The SPF is plugged into Broadcom(BCM57810S) PCI-E card and the router is running OpenWRT!
So far so good.
The problem is no matter what I do, the uplink is limited to 10Mbps and it should be 60 Mbps. Downlink is okay, running at 95Mbps as with the original ONT.
What I've tried
**OMCI_FAKE_OK** = 0 and 1 - same
**OMCI_TM_OPT** = 0 - net not working, 1 - same, 2 - same 
**OMCI_VEIP_SLOT_ID** = 255 - same, 11 - same
**PON_MODE** = 1 - same, 3 - not working
**FIBER_MODE** = 0, 1, 2, 3 - same
**LAN_SDS_MODE** = 1 - same, 3 - same

**same** = DOWN - 95Mbit, UP - 10Mbit

Anime4000, 2022-08-02:
have you tried to change `OMCC_VER` ?
default is 0x80 (128)

dexxter00, 2022-08-02:
> have you tried to change `OMCC_VER` ? default is 0x80 (128)
Yes, OMCC_VER = 0x80 and 0x81 net is working, but same result.
OMCC_VER = 0xB3 net is not working at all

What to try next? I'm in bridge mode, OMCI is provisioning 3 VLANs (NET, VOIP, TV). On Huawei ONT net working without bridge only on port 1 (OMCI is provisioning there).

Anime4000, 2022-08-02:
do you have other SFP host device to test?

dexxter00, 2022-08-02:
> do you have other SFP host device to test?

What do you mean? I have only one SFP stick and the Huawei ONT from my provider.

Or you asking about another router?
This same router was working in the same config, but from the stock ONT in bridge mode.

tdmadam, 2022-08-02:
If you have VoIP and Internet, then usually the traffic is mapped to different upstream queues.
Can you try on your router to test the Internet traffic with different 0-7 priority bits set?

dexxter00, 2022-08-02:
> Can you try on your router to test the Internet traffic with different 0-7 priority bits set?

I have tried (ingress and egress)
0:0
0:1
0:3
Not helping. Anything else? Where do I find such info? Maybe in original ONT? Or in OMCI info?

tdmadam, 2022-08-02:
What is the model number of your Huawei ONT? How many ethernet ports does it have?
Do you have another router where you could insert SFP? Just trying to eliminate SFP+BCM57810S combo as the reason.

Collect all configuration information about bridges, queues, vlans from SFP

```
for ME in 00002 00005 00006 00007 00011 00024 00045 00047 00049 00050 00052 00078 00079 00083 00084 00089 00130 00131\
00133 00134 00136 00137 00148 00157 00158 00171 00240 00244 00245 00248 00249 00250 00253 00255 00256 00257 00262 00263 \
00264 00266 00267 00268 00272 00273 00274 00277 00278 00280 00281 00284 00287 00296 00298 00307 00308 00309 00310 00311 \
00312 00321 00322 00329 00330 00334 00340 00341 65282 65294 65408 65527 65528 65529 65530 65531 74; do echo -e "\n\nMIB:" \
$ME"\n\n";omcicli mib get $ME; done;
```

If you decide to publish this, remove the serial number from ME6 and ME256.

Check the port status of the SFP
`# diag`

`RTK.0> port get status port all`

Check how many flows in upstream and downstream.
```
RTK.0> gpon show us-flow
RTK.0> gpon show ds-flow
```

dexxter00, 2022-08-02:
HG8145V5, 4 eth ports
At the moment, I don't have another router. The best is to try with Mikrotik but don't have one!
This is message I get in the log when SFP port is up
`kern.info kernel: [12581.781063] bnx2x 0000:01:00.0 eth2: NIC Link is Up, 1000 Mbps full duplex, Flow control: ON - receive & transmit`

Here is first command output: [pastebin](https://pastebin.com/6p66BPCt)

```
RTK.0> port get status port all
Port Status Speed    Duplex TX_FC RX_FC
---- ------ -----    ------ ----- -----
0    Up     1000M    Full   En    En   
2    Up     1000M    Full   Dis   Dis
```
```

RTK.0> gpon show us-flow
============================================================
    GPON ONU MAC U/S Flow Status
Flow ID | GEM Port | Type | TCont
      0 |      267 |  ETH |     0
      1 |      395 |  ETH |     1
      2 |     2699 |  ETH |     2
     64 |       11 | OMCI |    16
============================================================
```

```
RTK.0> gpon show ds-flow
============================================================
     GPON ONU MAC D/S Flow Status
Flow ID | GEM Port | Type | Multicast | AES
      0 |     4095 |  ETH |         * |    
      1 |      267 |  ETH |           |   *
      2 |      395 |  ETH |           |   *
      3 |     2699 |  ETH |           |   *
     64 |       11 | OMCI |           |   *
============================================================
```

tdmadam, 2022-08-03:
The traffic descriptor confirms your report. The three GEM ports 267, 395, and 2699 have the upstream rate limit of 1342177 Bytes/s = 10.24Mbps

``` # omcicli mib get 280 ```

```
MIB: 00280
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
TrafficDescriptor
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityId: 0x0001
CIR: 1342177
PIR: 1342177
CBS: 0
PBS: 0
ColourMode: 0
IngressColourMarking: 0
EgressColourMarking: 0
MeterType: 0
=================================
=================================
EntityId: 0x0002
CIR: 1342177
PIR: 1342177
CBS: 0
PBS: 0
ColourMode: 0
IngressColourMarking: 0
EgressColourMarking: 0
MeterType: 0
=================================
=================================
EntityId: 0x0003
CIR: 1342177
PIR: 1342177
CBS: 0
PBS: 0
ColourMode: 0
IngressColourMarking: 0
EgressColourMarking: 0
MeterType: 0
=================================
```
As a test, I tried changing the OMCI_TM_OPT on my stick from 2 (Priority and rate controlled) to 0 (Priority controlled).
It accepted the change, but the rate was still within the limits of whatever was assigned to the ONT.
Are the hardware version and equipment ID confirmed as valid strings?

dexxter00, 2022-08-03:
>Are the hardware version and equipment ID confirmed as valid strings?

They're copied from provider's ONT, but may not be sent by the stick properly, who knows? Maybe try othe SW ver?
I have tried to change HW_HWVER to some random string, I still have internet, the speed is still the same.

dexxter00, 2022-08-03:
I think it's a Huawei OLT.
I can't source MA5671A at the moment, nor I will have a chance in the near future.
The original values are configured as seen from the WebUI of the ONT. The tests are made with them. Trials are made with other strings as well as ```OMCI_OLT_MODE=1```, but everytime it gives the same results. 
What can be done, maybe @Anime4000 can have some idea?!?
<details>
  <summary> # omcicli mib get 131 </summary>

```
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
OltG
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityId: 0x00
OltVendorId: 0x48575443
EquipId:                     
Version: 10
ToDInfo:
	Sequence number of GEM superframe: 0x0
	Timestamp: secs 0, nanosecs 0
=================================
```

</details>

tdmadam, 2022-08-03:
That's the problem with web interfaces. What GUI developers display for users doesn't always match the strings used by engineers to connect.

OltVendorId: 0x48575443 so it is HWTC

> it also shows `1048568` and `4194296`

These hardcoded ONT values are in Kbps so it is 1Gbps and 4Gbps. They are used to limit the bandwidth per port. As @stich86 pointed out they cause a problem for people who have higher plans and need to be overriten to go above 1Gbps.

The other value (10.24Mbps) must come from the DBA profile assigned by the ISP.

Anime4000, 2022-08-03:
@dexxter00 try do something simple first, since you have tried doing various `flash set`...
have you tried other firmware?

dexxter00, 2022-08-03:
@Anime4000 
> @dexxter00 try do something simple first, since you have tried doing various `flash set`... have you tried other firmware?

Yes, first tried with `M114_sfp_ODI_Vlan_220414` , then tried with `M114_sfp_ODI_hybrid_220527` , now I'm running `[M114_sfp_ODI_220304`.
With 527 - can't seem to bridge port and no internet.
With 304 - internet is working, but same speed.

@tdmadam - do you know which commands I can use inside Huawei's shell so I can get real values? I have access to SSH / Telnet of stock ONT.

tdmadam, 2022-08-03:
> @tdmadam - do you know which commands I can use inside Huawei's shell so I can get real values? I have access to SSH / Telnet of stock ONT.

This command is from my Echolife. Hopefully it is also present in your model.

```WAP>omcicmd show log```

```
OLT Type: OTHER OLT

[1970-01-01 00:01:43.200614]OLT->ONT: Priority=1,SN=7, Device ID=0xA
MT=MIB Reset(15), MeID=ONT DATA(2), InstID=0
-----------------------------------------------
80 07 4f 0a 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00
```
Extract all hex lines from the log
Follow my github on how to read this in Wireshark. Use omci.lua from my repository as I started updating this file with missing MEs
If anyone wants to write a script to make it simpler, feel free to do so. Tomorrow I leave for a business trip, so my time will be limited.
The MEs you want to check first are 6, 7, 256, 257

ToTheCLI, 2022-08-04:
Had the same issue fixed it by Setting OMCC Ver to 0xB0 YMMV.

ToTheCLI, 2022-08-04:
Try
flash set OMCC_VER 176

dexxter00, 2022-08-04:
> Try flash set OMCC_VER 176

Already tried all the options in the WebGUI. Apperantly, only 0x80 and 0x81 are working and I'm not getting Internet with others.

dexxter00, 2022-08-05:
Hello again!
Today I have tried to manipulate omci provisioned data with no luck.
`omcicli mib set       
: set MIB ME attribute
    {classId, entityId, attrName, attrValue} `

Also, stock ONT data verified from CLI, Web Interface does NOT provide wrong data in my case!
When `OMCI_TM_OPT=0`, there's no MIB 280(TrafficDescriptor) provisioned and no `ds` and `us` flows but I get O5.
Tried changing `SW_SWVER` and `OMCI_SW_VER#` to something unreal and same results. So maybe OLT doesn't care about them?

@tdmadam - I have this command on my ONT. Done all the instructions and I've got some data, but in my case MIB 6 is not present and MIB 256, 257 are present, but don't get any usable data (it says change Date & Time on 256).

dexxter00, 2022-08-23:
Thank you all for your help. For some reason, my ISP have changed the OLT to ZTE and now I'm problem free.
Now I'm getting `OltVendorId: 0x5a544547`, upload speed is normal and not getting MIB 280 provisioned.

closed 2022-08-23 by dexxter00

SkateWarp, 2026-05-02:
> The traffic descriptor confirms your report. The three GEM ports 267, 395, and 2699 have the upstream rate limit of 1342177 Bytes/s = 10.24Mbps
>
> `# omcicli mib get 280`
>
> ```
> MIB: 00280 [...]

Hello!
I had the exact same issue with a Tenda HG1 (Realtek based). Upload limited to 10mbps.
Thanks to your assumption, I changed the PIR of MIB 280 to 125000000 (I had the exact same value 1342177, and OLT Huawei) and had my upload speed fixed.
Here's the command, for whoever needs it. 
`omcicli mib set 280 [entityId] PIR 125000000 `
I applied that to all the EntitiesID in my case

Please note, **this won't survive a reboot.**