Anime4000/RTL960x #60: OMCI parameters setup issue

akw28888, 2022-08-03:
Some Huawei OLTs lock third-party ONUs, the OLT will verify the Vendor ID from OMCI 256.
If the Vendor ID from ONU is not HWTC, the OLT will refuse to register ONU.

So some ISP's ONU will fake the Vendor ID to HWTC to bypass this limit, but the SN does not start with HWTC.

It seems the omci_app only reads Vendor ID from SN.
Is it possible to change that?

Also, is it possible to not set the LOID parameters?
The ISP doesn't use LOID to verify, but I got LOID error from OLT.

tdmadam, 2022-08-04:
I think this is what you are looking for.

```
 # flash set GPON_SN ALCL1234abcd
 # flash set PON_VENDOR_ID HWTC
 # reboot
```
Verify:
```
 # omcicli mib get 6
 # omcicli mib get 256
```

Anime4000, 2022-08-04:
> Also, is it possible to not set the LOID parameters?
> The ISP doesn't use LOID to verify, but I got LOID error from OLT.

I think it can't, maybe ODI can change Realtek SDK, allow blank input to disable LOID or PLOAM like:
```
flash del GPON_PLOAM_PASSWD
flash del flash set LOID
```

when `LOID` is blank and `GPON_PLOAM_PASSWD` is set, LOID is disabled,

if anyone who closer to ODI programmer, try update the source code

akw28888, 2022-08-07:
@tdmadam It's working now, thanks.
@Anime4000
There are some results:
The realtek sticks have another way to setup parameters, called ```mib set```
For PLOAM password, ODI sticks can set HEX 00000000000000000000
For LOID:
I tried ``` mib set LOID "" && mib commit``` then reboot.
The result will be: 
```
# mib get LOID

# flash get LOID
LOID=
```

But seems omci_app hardcoded default LOID value.
It still report LOID to OLT.

```
# omcicli mib get 256 |grep LogicalOnuID
LogicalOnuID: user

# omcicli mib get 65530                 
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
LoIdAuth
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityId: 0x0000
OperationId: CTC 
LoID: 0x757365720000000000000000000000000000000000000000
Password: 0x000000000000000000000000
AuthStatus: 0
=================================
```