Anime4000/RTL960x #62: Unresponsive behavior if management ip address is different

git-nerd, 2022-08-06:
My network does not use 192.168.1.0/24 so I changed the management ip.

DFP34 acts very strangely if the management ip is different than default.

If the fiber is plugged in during reboot, it never responds to the new ip. If the fiber is unplugged during reboot, then it responds much later than it responds to 192.168.1.0/24 . It then stops responding again after some time. If I need to access again, only option is power cycle the stick.

dexxter00, 2022-08-07:
It's not supposed to work like that in most cases! What router are you using? Mikrotik?
How most of us do it is to leave the stock IP addr (192.168.1.1/24) and then put the `sfp` interface into WAN firewall group.
This way, you can access your stick from every other subnet in your network (if it has access to WAN firewall group).

git-nerd, 2022-08-09:
Ended up doing something like you mentioned to have more isolation from my internal network. Regardless, I hope ODI web interface is not accessible from PON side.