Anime4000/RTL960x #64: Ufiber UF-Instant and UK FTTP from OpenReach

Pukkahq, 2022-08-13:
I have flashed my Ufiber UF-Instant with this firmware https://github.com/stich86/UF-Instant-Mod.
The ONT I have is made by nokia, has anyone in the UK got this working? 
What settings needed changing?
I dont think i am able to access the webGUI for the nokia so I am not sure how to get the GPON_PLOAM_PASSWD

Anime4000, 2022-08-13:
have you try access Nokia ONT WebGUI without fiber plugged in?
try disconnect fiber, power on Nokia ONT then try login in

Pukkahq, 2022-08-13:
I have ran an IP scan on the range 192.168.1.1 - 192.168.40.254 and i cannot even get a ping back from the device.

Anime4000, 2022-08-13:
I think you might need UART-TTL your Nokia ONT

or... broke your Nokia ONT, ask ISP to replace, during ONT replace, ask ISP Technician your PLOAM Password

I think only this way to obtain it

Pukkahq, 2022-12-28:
Just to update this. they is only 2 options that need changing 
flash set GPON_SN **YOURSN**
flash set PON_VENDOR_ID **ALCL**

closed 2022-12-29 by Pukkahq