Anime4000/RTL960x #67: Backup ODI firmware

jang5th, 2022-08-26:
Hi,

I purchased odi DFP-34X-2C2, I saw firmware version is: V1.0-220805.
I have 2 questions:
1. Where can I download ODI's firmware?
2. How can I backup firmware?

Please help.

Anime4000, 2022-08-27:
V1.0-220805 is in which sw_active ?

you can do this:
Kernel, RootFS @ sw_active 0
```
scp ubnt@192.168.1.1:/dev/mtd4 /home/user/backup/k0

scp ubnt@192.168.1.1:/dev/mtd5 /home/user/backup/r0
```
Kernel, RootFS @ sw_active 1
```
scp ubnt@192.168.1.1:/dev/mtd6 /home/user/backup/k1

scp ubnt@192.168.1.1:/dev/mtd7 /home/user/backup/r1
```

jang5th, 2022-08-27:
Step by step I do:
1. telnet 192.168.1.1
2. nv getenv sw_commit (sw_commit = 0)
3. nv setenv sw_commit 1
4. nv setenv sw_active 1
5. reboot
6. scp root@192.168.1.1:/dev/mtd4 /home/user/backup/k1
7. scp root@192.168.1.1:/dev/mtd7 /home/user/backup/r1

Or I'm wrong? I found in github project but I don't see any command to login ssh root and winscp.

Anime4000, 2022-08-27:
you can use `nc`, but you need Port Forward if you put SFP `192.168.1.1` behind NAT

tdmadam, 2022-08-27:
@jang5th 

I don't think we have the latest version V1.0-220805 here yet.

Can you post your firmware "r0" "r1" somewhere so we can test it?

jang5th, 2022-08-27:
> @jang5th
> 
> I don't think we have the latest version V1.0-220805 here yet.
> 
> Can you post your firmware "r0" "r1" somewhere so we can test it?

Hi,
How can I backup firmware?
I did: scp root@192.168.1.1:/dev/mtd7 /home/user/backup/r1 but scp not found.
[image]

jang5th, 2022-08-27:
When I type: 
nv setenv sw_commit 1
nv setenv sw_active 1

The firmware version is V1.0-220801
[image]

Anime4000, 2022-08-27:
@jang5th 
you need run that command at Linux PC

tdmadam, 2022-08-28:
@jang5th 
If scp does not work for some reason use nc with ubuntu_live
https://github.com/Anime4000/RTL960x/discussions/28

jang5th, 2022-08-28:
Sry, I'm not good linux.
I tried but failed.
[image]

Anime4000, 2022-08-28:
@jang5th
perhaps I can help you via AnyDesk, if you have LINE, add my id `anime4000`

tdmadam, 2022-08-28:
If you still need it, here are the step by step instructions for netcat:

1. Login to SFP stick
user@ubuntu:~$ ```telnet 192.168.1.1```
login: admin

     \# ```cd /tmp/```
     \# ```cat /proc/mtd```

2. Open second terminal window in Ubuntu, check the IP address
user@ubuntu:~$   ```ip a | grep inet```
    inet ```192.168.1.88```/24 brd 192.168.1.255 scope global

3. Listen for file transfer on Ubuntu on port 9999
user@ubuntu:~$   ```nc -l -p 9999 > mtd1```

4. Transfer mtd1 partition from SFP to Ubuntu
\# ```cat /dev/mtd1 | nc 192.168.1.88 9999```

5. Repeat for mtd2 mtd3 mtd4 mtd5 mtd6 mtd7
On Ubuntu:
user@ubuntu:~$   ```nc -l -p 9999 > mtd2```
On SFP
\# ```cat /dev/mtd2 | nc 192.168.1.88 9999```

6. Verify all files on Ubuntu
user@ubuntu:~$  ``` ls -l```
total 54108
-rw-rw-r-- 1 user user     8192 Aug 28 17:23 mtd1
-rw-rw-r-- 1 user user  2572288 Aug 28 17:38 mtd5

jang5th, 2022-08-28:
> If you still need it, here are the step by step instructions for netcat:
>
> 1. Login to SFP stick
>    user@ubuntu:~$ `telnet 192.168.1.1`
>    login: admin
>    # `cd /tmp/`
>    # `cat /proc/mtd`
> 2. Open [...]

Thank you for your quick response,

The pon stick was plugged into pfsense box (minipc), I tried steps that you taught but failed.
In the pfsense box interface speed: 
[image]

And these:
[image]

Anime4000, 2022-08-28:
@jang5th 
it seem NATed 192.168.1.1 > 192.168.5.33

you need do port forward 9999/tcp and 9999/udp (just in case) to 192.168.5.33

at telnet, change 192.168.5.33 to pfSense interface IP like 192.168.1.100

NATed require Port Forward from SFP to LAN

jang5th, 2022-08-28:
> @jang5th it seem NATed 192.168.1.1 > 192.168.5.33
> 
> you need do port forward 9999/tcp and 9999/udp (just in case) to 192.168.5.33
> 
> at telnet, change 192.168.5.33 to pfSense interface IP like 192.168.1.100
> 
> NATed require Port Forward from SFP to LAN

Thank you,
I download mtd1 (file size: 8.2KB) from PON stick, but I can not telnet the second times.
[image]

Anime4000, 2022-08-28:
that's very odd, it appears NAT masquerade is missing...

is SFP interface is as WAN ?

jang5th, 2022-09-03:
> 

Yes, SFP Interface is as WAN.
How can I extract firmware in ODI?
I saw 2 firmwares: 220801 & 220805.

closed 2022-09-15 by jang5th