Anime4000/RTL960x #72: DFP-34X-2C2 Speed very slow

FluorFR, 2022-09-09:
Hello,

I have ODI DFP-34X-2C2 stick.
I have configured, all value and i'm in 05 state, DHCP is OK and i get and IP.

But the speed is under 100mbit/s... (I got 2000Mbits with FS.com ONU)

Anime4000, 2022-09-09:
since you have tried OMCI_TM_OPT 0,
have you tried 1?

FluorFR, 2022-09-09:
> since you have tried OMCI_TM_OPT 0, have you tried 1?

With OMCI_TM_OPT=1 => no DHCP...

dexxter00, 2022-09-10:
I have experienced similar problem, look at my issue here - #58

FluorFR, 2022-09-10:
[image]

My TrafficDescriptor is clear... I thins this is the problem...

Anime4000, 2022-09-10:
[image]
here is mine

FluorFR, 2022-09-12:
I checked on my VSOL I don't have any TrafficDescriptor either I really don't know where to look...

FluorFR, 2022-09-12:
[image]
And i see same us/ds flow on VSOL and ODI...

Gouster4, 2022-09-15:
I had 10Mbps upload speed instead of 300Mbps, and download speed was reduced from 1000Mbps to 300Mbps.
Issue was original firmware that came with my device. I installed latest firmware from this github repo, and i think that speed is even better that with ISP's ONT.