Anime4000/RTL960x #75: SAGECOM F@st 5670 OMCI Info

Gouster4, 2022-09-14:
Hello. Im looking for SAGECOM F@st 5670 OMC Info.

Im looking for:
GPON_ONU_MODEL
GPON_SN
PON_VENDOR_ID
HW_HWVER
OMCI_SW_VER#

This is what in device info section.

Serial Number 	534D4253004360CA
Hardware Version 	2.5
Software Version 	SGFm100104
GUI Version 	1.110.64.12
UpTime 	00h11m46s
Datapump Version 	
WAN MAC Address 	64:FD:96:94:34:30

GPON Serial Number 534D4253004360CA
MTU 1492

My guess:
GPON_ONU_MODEL ?
GPON_SN 534D4253004360CA
PON_VENDOR_ID ?
HW_HWVER 2.5
OMCI_SW_VER# SGFm100104

Thanx.

Anime4000, 2022-09-15:
GPON _ONU_MODEL "F@st 5670"
the rest is correct

Gouster4, 2022-09-15:
> GPON _ONU_MODEL "F@st 5670" the rest is correct

Thank you.

closed 2022-09-15 by Gouster4

Gouster4, 2022-09-18:
I already replaced that SAGEMCOM router to HUAWEI. Its worst router but i got it working.
Just being curious...
When i did
flash set OMCI_SW_VER1 SGFm100104
flash set OMCI_SW_VER2 SGFm100104
i got both values reset to default after reboot.
Is it possible that it contains invalid characters or format? Maybe SAGECOM router also reported different string to OLT, or its just DFP-34X-2C2 firmware bug with accepting this string? thought about invalid characters, lenght limit or just invalid format.

Anime4000, 2022-09-20:
DFP-34X-2C2 need set OMCI_OLT_MODE 21
I have write about this on the doc and why.