Anime4000/RTL960x #76: [DFP-34X-2C2] Setting for systems on Openwrt

uzer00, 2022-09-14:
Hello and thank you for researches.
I need help in my conditions.
I have HG8145V5 and my params for SFP ODI 210702 is:

flash set GPON_ONU_MODEL HG8245Hv5
flash set OMCI_SW_VER1 V1R111C11S111
flash set OMCI_SW_VER2 V1R111C11S111
flash set HW_HWVER 167C.A
flash set LAN_SDS_MODE 1
flash set VLAN_CFG_TYPE 0
flash set VLAN_MANU_MODE 0
flash set GPON_SN HWTC1FA11A1D
flash set PON_VENDOR_ID HWTC
flash set DEVICE_TYPE 1
flash set OMCI_OLT_MODE 21         ( without that param speed is 100/ 10 but normal - 100/ 50 )

omcicli mib get 84

There is 4 vlan vid - 10 (internet pppoe), 20 (iptv dhcp), 40 (tr69 dhcp) ,50 (voip dhcp)
and every one has default FwdOp 0x04

In that configuration only tr69 gets ip by dhcp

But if I setup for:

vid 10  -  omcicli mib set 84 0x01 FwdOp 0x02
vid 50  -  omcicli mib set 84 0x04 FwdOp 0x02

It gets ip by dhcp for Internet pppoe and voip

Which setups should I use for OpenWRT so that vlan bridge iptv and broadcast work?

akw28888, 2022-09-18:
You can check the meaning for FwdOp (Forward Operation) on Page 69-70.
[31535950.pdf](https://github.com/Anime4000/RTL960x/files/9595283/31535950.pdf)

uzer00, 2022-09-24:
It is interesting because when I read that param on original ONU I get:
```
su>  omcicmd mib show meid 84 instid 65535
  -------------------------------------------------------
  Me(84)(VLAN Tagging Fi) has 8 instances, instance size is 28
  -------------------------------------------------------
  Instance ID = 0x0006, Instance Content:
  0a000000 00000000 00000000 00000000 00000000 00000000 1001
  Instance ID = 0x0008, Instance Content:
  14000000 00000000 00000000 00000000 00000000 00000000 1001
  Instance ID = 0x0009, Instance Content:
  28000000 00000000 00000000 00000000 00000000 00000000 1001
  Instance ID = 0x000a, Instance Content:
  32000000 00000000 00000000 00000000 00000000 00000000 1001
    -------------------------------------------------------
```
And penultimate byte show that I need used <FwdOp 0x10>, as OLT side setup that param. Isn't it?

uzer00, 2022-10-01:
Solved.
Just I need setup priority for everyone vlan interfaces and network worked fine.
And of course I used macvlan for everyone vlan interfaces. In my conditions:
vid 10 (internet pppoe) priority 2  (qos 0:2...)
vid 20 (iptv dhcp) priority 4 (qos 0:4...)
vid 40 (tr69 dhcp)  priority 0 (qos 0:0...)
vid 50 (voip dhcp)  priority 5 (qos 0:5...)

```
# cat /etc/config/network
..
config device                             
        option name 'br0.10'                       
        option type '8021q'               
        option ifname 'br0'               
        option vid '10'                            
        list egress_qos_mapping '0:2'     
        list egress_qos_mapping '1:2'     
        list egress_qos_mapping '2:2'              
        list egress_qos_mapping '3:2'     
        list egress_qos_mapping '4:2'     
        list egress_qos_mapping '5:2'              
        list egress_qos_mapping '6:2' 

```

closed 2022-10-01 by uzer00

Anime4000, 2022-10-02:
@uzer00 that command `# cat /etc/config/network` is worked in RTL960x Stick?? or Router?

uzer00, 2022-10-04:
@Anime4000  my way is best on router Mikrotik 760igs powered by OpenWRT dev fw and sfp ODI 210702. My prov offers triplay service - Internet (vlan 10 prio 2), voip (vlan 50 prio 5), iptv(vlan 20 prio 4 ) and it worked with next configuration + igmpproxy and relayd packages. All in one bridge with vlan filter and qos priority. 
```
...
config interface 'internet'
	option proto 'pppoe'
	option device 'br0.10mac0'
	option username 'XXXXX'
	option password 'XXXXX'

config interface 'iptv'
	option device 'br0.20mac0'
	option classlessroute '1'
	option proto 'dhcp'
	option defaultroute '0'

config interface 'tr69'
	option device 'br0.40mac0'
	option proto 'dhcp'
	option defaultroute '0'

config interface 'voip'
	option device 'br0.50mac0'
	option proto 'dhcp'
	option defaultroute '0'
...
config interface 'port4'
	option device 'br0.300'
	option proto 'none'
	option defaultroute '0'

config interface 'iptv_relay'
	option proto 'relay'
	list network 'iptv'
	list network 'port4'
	option defaultroute '0'
...
config device
	option type 'bridge'
	option name 'br0'
	option bridge_empty '1'
	option igmp_snooping '1'
	list ports 'lan2'
	list ports 'lan3'
	list ports 'lan4'
	list ports 'lan5'
	list ports 'sfp'
	list ports 'wan'

config bridge-vlan
	option device 'br0'
	option vlan '1'
	list ports 'sfp:u*'

config bridge-vlan
	option device 'br0'
	option vlan '10'
	list ports 'sfp:t'

config bridge-vlan
	option device 'br0'
	option vlan '20'
	list ports 'sfp:t'

config bridge-vlan
	option device 'br0'
	option vlan '40'
	list ports 'sfp:t'

config bridge-vlan
	option device 'br0'
	option vlan '50'
	list ports 'sfp:t'
...
config device
	option type 'macvlan'
	option ifname 'br0.10'
	option mode 'private'
	option name 'br0.10mac0'
	option macaddr 'XX:XX:XX:XX:XX:0A'

config device
	option type 'macvlan'
	option ifname 'br0.20'
	option mode 'private'
	option name 'br0.20mac0'
	option macaddr 'XX:XX:XX:XX:XX:0B'

config device
	option type 'macvlan'
	option ifname 'br0.40'
	option mode 'private'
	option name 'br0.40mac0'
	option macaddr 'XX:XX:XX:XX:XX:0C'

config device
	option type 'macvlan'
	option ifname 'br0.50'
	option mode 'private'
	option name 'br0.50mac0'
	option macaddr 'XX:XX:XX:XX:XX:0D'
...
config device
	option name 'br0.10'
	option type '8021q'
	option ifname 'br0'
	option vid '10'
	list egress_qos_mapping '0:2'
	list egress_qos_mapping '1:2'
	list egress_qos_mapping '2:2'
	list egress_qos_mapping '3:2'
	list egress_qos_mapping '4:2'
	list egress_qos_mapping '5:2'
	list egress_qos_mapping '6:2'
	list egress_qos_mapping '7:2'

config device
	option name 'br0.20'
	option type '8021q'
	option ifname 'br0'
	option vid '20'
	list egress_qos_mapping '0:4'
	list egress_qos_mapping '1:4'
	list egress_qos_mapping '2:4'
	list egress_qos_mapping '3:4'
	list egress_qos_mapping '4:4'
	list egress_qos_mapping '5:4'
	list egress_qos_mapping '6:4'
	list egress_qos_mapping '7:4'

config device
	option name 'br0.40'
	option type '8021q'
	option ifname 'br0'
	option vid '40'
	list egress_qos_mapping '0:0'
	list egress_qos_mapping '1:0'
	list egress_qos_mapping '2:0'
	list egress_qos_mapping '3:0'
	list egress_qos_mapping '4:0'
	list egress_qos_mapping '5:0'
	list egress_qos_mapping '6:0'
	list egress_qos_mapping '7:0'

config device
	option name 'br0.50'
	option type '8021q'
	option ifname 'br0'
	option vid '50'
	list egress_qos_mapping '0:5'
	list egress_qos_mapping '1:5'
	list egress_qos_mapping '2:5'
	list egress_qos_mapping '3:5'
	list egress_qos_mapping '4:5'
	list egress_qos_mapping '5:5'
	list egress_qos_mapping '6:5'
	list egress_qos_mapping '7:5'
...
```