Anime4000/RTL960x #77: [DFP-34X-2C2] Cannot set OMCI_SW_VER*

Gouster4, 2022-09-19:
# flash get OMCI_SW_VER1
OMCI_SW_VER1=V1.0-220817
# flash get OMCI_SW_VER2
OMCI_SW_VER2=V1.0-220817
# flash set OMCI_SW_VER1 V3R016C10S150
OMCI_SW_VER1=V3R016C10S150
# flash set OMCI_SW_VER2 V3R016C10S150
OMCI_SW_VER2=V3R016C10S150
#
# flash get OMCI_SW_VER1
OMCI_SW_VER1=V3R016C10S150
# flash get OMCI_SW_VER2
OMCI_SW_VER2=V3R016C10S150
# reboot
Connection closed by foreign host.
telnet 192.168.5.3
Trying 192.168.5.3...
Connected to 192.168.5.3.
Escape character is '^]'.
login: admin
Password:

BusyBox v1.12.4 (2022-08-17 15:13:25 CST) built-in shell (ash)
Enter 'help' for a list of built-in commands.

# flash get OMCI_SW_VER1
OMCI_SW_VER1=V1.0-220414
# flash get OMCI_SW_VER2
OMCI_SW_VER2=V1.0-220817
#

Its being reset do default after reboot.
Please fix firmware. Thanx.

TheKhack, 2022-09-19:
If using DFP-34X-2C2, one option to avoid the OMCI_SW_VER* override after reboot is to set the following value, as discussed https://github.com/Anime4000/RTL960x/issues/30#issuecomment-1146604684:

flash set OMCI_OLT_MODE 21

dexxter00, 2022-09-23:
Or, there is a firmware version that fixes this bug as well as some others - [Anime4000/RTL960x/Firmware/DFP-34X-2C2](https://github.com/Anime4000/RTL960x/tree/827f8b2c8b9a8cc1b3a050440fedd3af01788315/Firmware/DFP-34X-2C2)

Gouster4, 2022-09-23:
OMCI_OLT_MODE 21 does the trick.

closed 2022-09-23 by Gouster4

maciekb, 2023-01-18:
I have the same problem with V2801F_V1.9.0-220425.tar.