Anime4000/RTL960x #78: [DFP-34X-2C2] Access from outside of subnet

Gouster4, 2022-09-24:
Hello. i have DFP-34X-2C2 connected to OPNSense.
DFP-34X-2C2 has IP 192.168.5.3 (mask 255.255.255.0), and OPNSense has IP 192.168.5.1 on same intefrace.
Problem is that i have more subnets connected to OPNSense. (192.168.0.X; 192.168.1.X...)
I set rules to allow traffic between some subnets. Problem is that im not able to access DFP-34X-2C2 from different subnet.
Like accessing from 192.168.3.147 while DFP-34X-2C2 has IP 192.168.5.3 (mask 255.255.255.0).

Is there some way to allow access from any IP and subnet on lan (sfp) side?
Maybe trought some commands, or editing files in filesystem. But im not sure if its R/W, and if changes presist reboot.

Thanx.

Anime4000, 2022-09-24:
This mean: Route

The DFP-34X-2C2 don't have Route List to your subnet,

you need add Route List inside DFP-34X-2C2

another way is using NAT Masquerade just like WAN

Gouster4, 2022-09-24:
> This mean: Route
> 
> The DFP-34X-2C2 don't have Route List to your subnet,
> 
> you need add Route List inside DFP-34X-2C2
> 
> another way is using NAT Masquerade just like WAN

My subnet count is pretty flexible.
Is there way to add whole private (local) part of network to DFP-34X-2C2?

Anime4000, 2022-09-26:
I think doing NAT Masquerade is much simpler and safe

This what I do in my Mikrotik, allow any address able to access my xPON ONU Stick

First, I give `sfp-sfpplus1` an IP Address, such as `192.168.1.11/24`
[image]

Then, I add new `masquerade` firewall rules at second (after WAN `masquerade`)
[image]
[image]

Firewall rules should look like this:
1. WAN Masquerade
2. SFP Masquerade
3. ...

I tried this method on Ubiquiti EdgeRouter 12, also works just fine

Gouster4, 2022-09-30:
Solved by 
`route add -net 0.0.0.0/0 gw 192.168.5.1 br0`
but it didnt presist reboot. Any suggestions?

Anime4000, 2022-10-01:
to make it persistence, you need modify the firmware and add into `rc` folder, this is why I choose NAT method.

Gouster4, 2022-10-01:
> to make it persistence, you need modify the firmware and add into `rc` folder, this is why I choose NAT method.

Thanx. Will try this.

closed 2022-10-01 by Gouster4

johndoe1983, 2024-01-26:
@Anime4000 , Could I access from outside of subnet if I use a Media Converter? the IP of my ODI DFP-34X-2C2 connected to Realtek 2.5Gb Media Converter is 192.168.1.2 while my router is on 192.168.1.1 but I cannot access to 192.168.1.2 from router neither devices connected to the router

Gouster4, 2024-01-27:
> @Anime4000 , Could I access from outside of subnet if I use a Media Converter? the IP of my ODI DFP-34X-2C2 connected to Realtek 2.5Gb Media Converter is 192.168.1.2 while my router is on 192.168.1.1 but I cannot access to 192.168.1.2 from router neither devices connected to the router

Whats your netmask? Because 192.168.1.1 and 192.168.1.2 looks like in same subnet if subnet uses /24 cidr (255.255.255.0 netmask) which is default on most routers.

Anime4000, 2024-01-27:
> Whats your netmask? Because 192.168.1.1 and 192.168.1.2 looks like in same subnet if subnet uses /24 cidr (255.255.255.0 netmask) which is default on most routers.

My Mikrotik LAN subnet is 192.168.88.0/24, and I use NAT to access the stick WebGUI

johndoe1983, 2024-01-27:
> Whats your netmask? Because 192.168.1.1 and 192.168.1.2 looks like in same subnet if subnet uses /24 cidr (255.255.255.0 netmask) which is default on most routers.

I have /24 as netmask on router and on SFP ODI in Media Converter but I cannot access to Telnet neither SSH and I cannot make ping from any device connected to router to 192.168.1.2 (SFP device)

Gouster4, 2024-01-27:
> > > @Anime4000 , Could I access from outside of subnet if I use a Media Converter? the IP of my ODI DFP-34X-2C2 connected to Realtek 2.5Gb Media Converter is 192.168.1.2 while my router is on 192.168.1.1 but I cannot access to 192.168.1.2 from router neither devices connected to the router
> > 
> > 
> > Whats your netmask? Because 192.168.1.1 and 192.168.1.2 looks like in same subnet if subnet uses /24 cidr (255.255.255.0 netmask) which is default on most routers.
> 
> I have /24 as netmask on router and on SFP ODI in Media Converter but I cannot access to Telnet neither SSH and I cannot make ping from any device connected to router to 192.168.1.2 (SFP device).
> 
> The only way for to connect to SFP is connect Ethernet Cable from Media Converter to the device to connect (then all the devices wont have internet because it is the FTTH provider...)

Looks like some kind of fault...
Can be hardware issue, firmware issue or configuration issue. Hopefully its configuration issue.
Can you access reset button/pins on your device?

Gouster4, 2024-01-27:
> I have /24 as netmask on router and on SFP ODI in Media Converter but I cannot access to Telnet neither SSH and I cannot make ping from any device connected to router to 192.168.1.2 (SFP device). [...]

So, you can access your your SFP stick from pc, when pc is directly connected to ethernet port of media converter. But not trought router. Right? 

But none of your devices can see it trought router, ans router itself also desnt see it?

If so, then maybe there is special port on router for guest network or WAN. They have configured separate subnets. WAN port is mostly DHCP client, or static IP, but you are most likely looking for VLAN  assigned by your provider and pppoe on top of it to get internet working.

Anime4000, 2024-01-27:
Most router cannot allow set WAN IP Address besides the Internet/PPPoE

like in Mikrotik, I add `sfp-sfpplus1` another IP Address (192.168.1.2/24)

then NAT it from 192.168.88.0/24 to 192.168.1.2

that how I did, I can access WebGUI or Telnet from 192.168.88.160 PC to 192.168.1.1 PON Stick

for non Mikrotik, I don't know, Ubiquiti can, but consumers Router are cannot!

johndoe1983, 2024-01-27:
> Looks like some kind of fault... Can be hardware issue, firmware issue or configuration issue. Hopefully its configuration issue. Can you access reset button/pins on your device?

I'm sure that it is a configuration issue because it happened with my old stock ONU (Ufiber NanoG).

> So, you can access your your SFP stick from pc, when pc is directly connected to ethernet port of media converter. But not trought router. Right?

That it's.

> But none of your devices can see it trought router, ans router itself also desnt see it?

Router can to see the device.

> If so, then maybe there is special port on router for guest network or WAN. They have configured separate subnets. WAN port is mostly DHCP client, or static IP, but you are most likely looking for VLAN assigned by your provider and pppoe on top of it to get internet working.

> Most router cannot allow set WAN IP Address besides the Internet/PPPoE
> for non Mikrotik, I don't know, Ubiquiti can, but consumers Router are cannot!

That would be. Maybe there are any way to allow set WAN IP address for to access to SFP of Media Converter in my OpenWRT router from any device of my LAN.