Anime4000/RTL960x #81: CRS310 & GP801R 2,5 Gbps issues.

Mhrok, 2022-10-05:
Hello,
I'm tinkering around using my own SFP ONTs at home. My ISP has nothing against it.
Recently I bought CRS310 with plenty of SFP and SFP+ ports - I'm still learning how to use it.
I've got Opton GP801R flashed to newest V2801F_V1.9.0-220425 firmware, it was warking properly in SFP port. I was greedy and I set LAN_SDS_MODE 4 and now SFP does not answer my calls on its management IP. Yes, I do connect the ISP to activate the module and I tried both in SFP and SFP+ ports with and without auto-neg.
Is there an easy way to restore communication with the module or should I prepare UART to access the settings?

Mhrok, 2022-10-05:
I just bought new one. Its next day shipment, do tomorrow morning I will have a second one. Which LAN_SDS_MODE should I use to get 2,5 GbE?

Anime4000, 2022-10-06:
I not sure about CRS310 support 2.5Gb xPON
Only supported Mikrotik Switch is CRS305-1G-4S+IN, because that switch using `98PX1012`

beside using UART, you can use [2.5Gb Media Converter](https://www.aliexpress.com/item/1005004340369253.html) to restore GP801R back

ODI DFP-34X-2C2 has no this problem because it has Auto Negotiate LAN_SDS_MODE

Mhrok, 2022-10-06:
Due to unavailability of CRS305 in my country in a sane price I took the risk and will sacrifice my time and sweat for the community to do some testing with CRS310. We'll see. Just for fun I ordered ODI stick too, but its not going to show up in at least a month.
Are there any specific type of 2.5 Gb port requirements for the 2,5 Gb stick to work? Any specifics about the switch chip? I'm asking because Mikrotik on their wiki tells us pretty universal story about CRS3xx SFP+ ports, without any specifics:

```
The 2.5G link rate support is implemented since RouterOS v7.3. MikroTik devices with SFP+ and SFP28 interfaces that support 2.5G link rate require following settings to be set on both linked device interfaces.
`/interface ethernet set sfp-sfpplus1 auto-negotiation=no speed=2.5Gbps full-duplex=yes`
Devices which support 2.5G links in SFP+/SFP28 ports:
**CRS3xx series switches - All SFP+ interfaces can be used in 2.5G mode if required.**
```

itfan1, 2022-10-06:
>  Are there any specific type of 2.5 Gb port requirements for the 2,5 Gb stick to work? Any specifics about the switch chip? I'm asking because Mikrotik on their wiki tells us pretty universal story about CRS3xx SFP+ ports, without any specifics:

In the RouterOS 7.3 changelog they say the following:
``` 
sfp - added 2.5Gbps rate for SFP+ and QSFP+ interfaces on 98DXxxxx and 98PX1012 switches (requires disabled auto-negotiation);
```
I think that this should indeed cover all of the CRS3xx switches. In particular, the switch chip of the CRS305 is 98DX3236 and the switch chip of the CRS310 is 98DX226S, so both of them should work.

I have a CRS305 and it works well at 2.5G rate with every 2.5G-supporting SFP module that I tried with it. This includes three different GPON modules (Nokia G-010S-A, Nokia G-010S-Q and ODI DFP-34X-2C2) and five different types of 2.5G SFP to RJ45 modules. My understanding is that the CRS310 is supposed to be similar in this regard.

> ```
> `/interface ethernet set sfp-sfpplus1 auto-negotiation=no speed=2.5Gbps full-duplex=yes`
> ```
Indeed, these are the correct settings for my CRS305. Here is how it looks in WebFig:

[image]

Note that when one changes the settings from auto-negotiation it may be needed to disable and enable the interface before things start working. Also, one should probably not forget to upgrade the firmware after upgrading RouterOS.

Mhrok, 2022-10-10:
Many thanks for feedback!
First, I made sure ROS version is adequate:
```
> /system routerboard print 
       routerboard: yes
             model: CRS310-1G-5S-4S+
  factory-firmware: 6.49.3
  current-firmware: 7.5
  upgrade-firmware: 7.5
```

Next I tested all of my ONTs with properly configured port and with support from ISP - all of them were properly authorised from OLT side and were seen by OLT properly, issue is between SFP and CRS.

I'm getting no communication on both GP801R (one with improper LAN_SDS_MODE and one with unsuccessful firmware upgrade, apparently).
There is some hope around Sercomm FGS202 - it connects properly when in SFP+ port, but only 1G. 2,5G is a no-link situation.
```
> interface ethernet monitor sfp-sfpplus4 once
                    name: sfp-sfpplus4
                  status: link-ok
        auto-negotiation: disabled
                    rate: 1Gbps
             full-duplex: yes
         tx-flow-control: no
         rx-flow-control: no
      sfp-module-present: yes
             sfp-rx-loss: no
            sfp-tx-fault: no
                sfp-type: SFP-or-SFP+
      sfp-connector-type: SC
      sfp-link-length-sm: 20km
         sfp-vendor-name: SERCOMM
  sfp-vendor-part-number: FGS202
     sfp-vendor-revision: 0001
       sfp-vendor-serial: SCOMxxxxxxxx
  sfp-manufacturing-date: 17-08-22
          sfp-wavelength: 1310nm
         sfp-temperature: 50C
      sfp-supply-voltage: 3.299V
     sfp-tx-bias-current: 16mA
            sfp-tx-power: 1.762dBm
            sfp-rx-power: -23.372dBm
         eeprom-checksum: good
```

`/interface ethernet set sfp-sfpplus4 auto-negotiation=no speed=2.5Gbps full-duplex=yes`

```
> interface ethernet monitor sfp-sfpplus4 once
                    name: sfp-sfpplus4
                  status: no-link
        auto-negotiation: disabled
      sfp-module-present: yes
             sfp-rx-loss: no
            sfp-tx-fault: no
                sfp-type: SFP-or-SFP+
      sfp-connector-type: SC
      sfp-link-length-sm: 20km
         sfp-vendor-name: SERCOMM
  sfp-vendor-part-number: FGS202
     sfp-vendor-revision: 0001
       sfp-vendor-serial: SCOMxxxxxxxx
  sfp-manufacturing-date: 17-08-22
          sfp-wavelength: 1310nm
         sfp-temperature: 50C
      sfp-supply-voltage: 3.299V
     sfp-tx-bias-current: 16mA
            sfp-tx-power: 1.735dBm
            sfp-rx-power: -23.467dBm
         eeprom-checksum: good
```

We tried to convince ONT to boot from other firmware, but the sticks are not accepting the commands from Huawei OLT.

itfan1, 2022-10-10:
> There is some hope around Sercomm FGS202 - it connects properly when in SFP+ port, but only 1G. 2,5G is a no-link situation.

My understanding is that this module is based on an Intel/Lantiq PEB98036 chipset, has only 1MB of RAM and runs an eCos RTOS operating system. Why do you think that it should support 2.5G?

Mhrok, 2022-10-10:
> Why do you think that it should support 2.5G?
https://hack-gpon.github.io/ont-sercomm-fgs202/ lists this as 

`HSGMII | Yes`

Second, in first link mentioned there in HW Modding section there is a mention of TIM ISP from Italy prividing customers with FGS202 as a SFP ONT with their 2,5 Gbps offer.

That are my reasons to at least try - it was cheap to check, as I am going to use it as my second connection SFP, which is 100/10, thus totally ok for this SFP.

itfan1, 2022-10-10:
> > Why do you think that it should support 2.5G?
> 
> https://hack-gpon.github.io/ont-sercomm-fgs202/ lists this as
> 
> `HSGMII | Yes`
> 
> Second, in first link mentioned there in HW Modding section there is a mention of TIM ISP from Italy prividing customers with FGS202 as a SFP ONT with their 2,5 Gbps offer.

Interesting. Thanks for the info.

Mhrok, 2022-10-13:
What the most interesting thing is, it looks like Mikrotik is falsely reporting its capability to do 2.5G. While it does indicate the capability to do 2.5G, four different dongles refuse to work in SFP+ port with forced 2.5G.
I've tested GP801R, FGS202, LXT-010S-H are not working to a different degree - first two are correctly seen by OLT, LEOX was not checked yet. FGS has its issues with reboot loop every 30 seconds.

As suggested before, I ordered CRS305. It has different switch chip and maybe will act differently - I will report my findings next week.

Mhrok, 2022-11-03:
After some tinkering I tested couple of devices, sticks and switches. Results are:
- Mikrotik CRS310 and CRS305 are identical regarding the SFP ONUs. If something was working in CRS310 it was working in CRS305.
- GP801R were dead-end for 1G+ speeds and I ditched them for another vendors. To 1G it was working ok. 
- LEOX LXT-010S-H is not working with Mikrotik devices with 2,5G speed - 1G is OK.
- Sercomm FGS202 was working with 1G, not working with 2,5G. No web interface, no telnet, so no possibility to dig further.
- ODI XPON DFP-34X-C2C with SFU firmware works great, with [ @stitch86 tweaks mentioned](https://github.com/Anime4000/RTL960x/issues/17#issuecomment-1101435506) in [Device list](https://github.com/Anime4000/RTL960x/blob/main/Docs/2.5Gb.md#device-list) it goes above 1G easily and I have no issues with vlans.
Speed test during the day, so I did not have whole PON circuit free, QoS stopped me "a little" - ISP allowed 2G down 1G up. I'll test later, but no doubt - I have over 1G.
[image]

And my CRS310 details:

```
> system routerboard print 
       routerboard: yes
             model: CRS310-1G-5S-4S+
     firmware-type: dx3230L
  factory-firmware: 6.49.3
  current-firmware: 7.6
  upgrade-firmware: 7.6
```

And relevant part of the config:
```
/interface ethernet
set [ find default-name=sfp-sfpplus4 ] auto-negotiation=no speed=2.5Gbps
```

Anime4000, 2022-11-03:
I am glad that working! it's nice to have 2.5Gb Internet and 2.5Gb GPON

I in other hand, using 2.5Gb/HiSGMII PHY on my ODI and RB5009 because faster LAN_SDS_MODE sync and very stable.

before with 1Gb mode, ODI have trouble to sync with RB5009 SFP+ end up multiple Link Down, take ~5min to have internet which is not acceptable, 2.5Gb HiSGMII PHY fix it even I not using higher speed

stable connection is the priority

stich86, 2022-11-16:
@Mhrok Marcin from Leox has fixed the issue to make LEOX LXT-010S-H running in 2.5G on Mikrotik

[image]

I'm currently running it for more than a month.
If this firmware will be stable for the next couple of week it should be marked as production, so then you can ask Marcin to share and you can update your SFP

alexdelprete, 2022-11-20:
> I'm currently running it for more than a month.

I have it running at 1Gbps in the CRS305 because Marcin told me to wait for the next fw release, I suppose you're beta-testing it.

can you please share the exact config for the CRS305 to enable the 2.5Gbps and also on the stick side (LAN_SDS_MODE etc.).

Thanks a lot.