Anime4000/RTL960x #82: Flash DFP-34X-2C2 firmware via UART

qorndjs, 2022-10-12:
Hello,

I read carefully these introductions https://github.com/Anime4000/RTL960x/issues/46#issuecomment-1171070495
But I don't know how to teardown DFP-34X-2C2.
If you find time to describe the process in more detail, I would be very grateful. Or share links to information about what I need.

Thank you.

Anime4000, 2022-10-14:
teardown of newer revision of DFP-34X-2C2 is pretty straight forward, you will see some lock mechanism each side, just pull both side a bit and slide up

qorndjs, 2022-10-14:
> teardown of newer revision of DFP-34X-2C2 is pretty straight forward, you will see some lock mechanism each side, just pull both side a bit and slide up

Thank you oppa~
I saw the Soc is RTL960D.
The square icon is Tx pin.
The circle icon is Rx pin.
But I can not find the Power (3.3V) and Gnd pin on the stick.
Can you please explain the detail of the pin?

The external CMOS battery (3.3V), the uart CP2102, the ODI stick. Will I plug in below the picture. Is it right?
[image]

Anime4000, 2022-10-14:
I simply solder the pad and insert into SFP with SoC facing up