Anime4000/RTL960x #83: Alternative sources for RTL ODI DFP-34X-2C2

rajkosto, 2022-10-15:
The supplied link in README.MD has 71$ shipping to my country, and the taobao link is dead.
I've found some alternatives but they don't look the same while still saying DFP-34X-2C2
Original link: https://www.aliexpress.com/item/1005003515662920.html
Alternative 1: https://www.aliexpress.com/item/4000537015384.html (says ODI DFP-34X-2C2 but doesn't seem to have bottom heatsink?)
Alternative 2: https://www.aliexpress.com/item/1005004580254846.html (does not have ODI marking but says XPON2.... on the third line like the original link does)

Mhrok, 2022-11-04:
I bought XPON version of https://aliexpress.com/item/4000537015384.html and it works flawlessly with 2,5G. Link was dead at first, but using VPN it was easy to order it to Poland. I paid $74.43, because of taxes.

rajkosto, 2022-11-04:
You used that exact link ? That store changed it's name since i posted it hehe
Does it look like that ? with no XPON... in the serial and no bottom heatsink (little cubes pattern)
I was under the impression that those units have ZTE chipset and cannot run this customized firmware (which i need for setting HW_REV and 4-port bridge mode)

skon77, 2022-11-04:
The best solution is to buy from an official dealer - HSGQ company.
https://detail.tmall.com/item_o.htm?id=654848544398

Mhrok, 2022-11-11:
My device has a little cubes pattern - if it's really necessary I can take photos of the unit, but I'd have to disconnect it from my switch to make proper photos. It stays under 50 C under load without exceptional airflow.

rajkosto, 2022-11-11:
i ordered the 2nd link, will show what it is when it arrives
its good to know yours has the cubes pattern on the bottom (and i assume ODI sticker on the top?, you can take a picture of both the cubes and the sticker without unplugging it i think), which just means the pictures the seller posted are wrong, and it is the correct unit regardless

skon77, 2022-11-13:
LOL! So you clarify please! Do you need screws and cubes?
Or do you want the original ODI products?
I attach two screenshots from the websites of these two companies. As you can see for yourself, ODI and HSGQ are one and the same company! 

---------------
[image]
---------------
[image]
---------------

They are essentially two divisions of the same company. The production division and the distribution division.

rajkosto, 2022-11-13:
Thank you for the STICK note, it means that the one I bought from alternative 2, if it is the same as the seller pictures, would be STICK3V06. The pictures without the cooling cubes say PON STICK V1, which might not be what I want (but other poster said that he bought from that link and got the cubes version, so it is just the images that are inaccurate/show old version).

skon77, 2022-11-14:
I assume that the type of stick design may depend on the availability of components at the time of production.
If you look at the official [catalog](http://www.odi-tec.com/products_6/78.html), there are no additional cooling cubes at all.
Contact HSGQ via TMALL. Perhaps they can comment.

rajkosto, 2022-11-15:
Here's what i got from "alternative 2" link, it has the ODI logo and XPON2 serial and is a STICK3V06 so i am satisfied.
https://images.sshnuke.net/2022-11-15_09-33-16_HFa6ZcWS2.png