Anime4000/RTL960x #84: Factory reset without UART

grayver, 2022-10-16:
Hello! I've got ODI DFP-34X-2C2 from the seller you recommended. Then I tried to use the stick with my Mikrotik RB5009. I got O5 and even an IP address very quickly, however Internet hadn't been working. I was trying to play around with different stick parameters with no success. But when I set parameter LAN_SDS_MODE to 3, I lost access to the stick. Now the stick is not accessible neither via WebGUI, nor by telnet. However, Mikrotik shows module presence.

Is there a way to reset the stick to factory defaults without bringing UART converter and soldering to stick internals?

crrodriguez, 2022-10-16:
The cheapest way is connecting the stick to a media converter (TP-Link MC220L, for 19 bucks.. gigabit only..  generic 2.5gbe one for something like 30 USD) and fix the SDS mode there.

rajkosto, 2022-11-16:
Keep in mind you do not have to buy the 2.5gbps media converter in order to fix if you set LAN_SDS_MODE to 4
Somehow the ODI DFP-34X-2C2 can autonegodiate speed with Qualcomm/Realtek media converter even if you set LAN_SDS_MODE to 4, see this output with it connected to one:
```
# cat /proc/lan_sds/lan_sds_cfg
lan_sds_mode = 1(Fiber 1G)
# flash get LAN_SDS_MODE
LAN_SDS_MODE=4
```

Anime4000, 2022-11-16:
so far ODI with RTL9601D can do SFP Auto Negotiate,

for other stick like V2801F, TWCGPON657 or RTL9601CI based will lock you out, safer way is to use 2.5Gb Media Converter, no need to take apart or soldering and UART

thankful with ODI approach with SFP Auto Nego

zentavr, 2023-03-27:
I have the same question, but a different problem:
I had updated the MAC address, but did not update a key.

Anime4000, 2023-03-28:
> I have the same question, but a different problem: I had updated the MAC address, but did not update a key.

for ODI, it should allow to access WebUI, but some SFP host require fiber to present,

Like RB5009, no need fiber to be plugged can access WebUI, my self, wrong MAC Key device wont allow to connect to OLT, but able to access via Telnet or WebUI

before tear down, try plug SFP Stick to SFP media converter

Anime4000, 2023-03-28:
@zentavr SFP 
most Realtek based SFP Mediaconvert should work
https://github.com/Anime4000/RTL960x/blob/main/Docs/2.5Gb.md#device-list

unless they change device revision or something, however, it should works

zentavr, 2023-03-28:
I have Mikrotik RB4011iGS+5HacQ2HnD, and I could ping/access the stick over SSH/Web only when have a fiber cable inside of it. So when there are issues with the signal/provider - I cannot even debug the problem at the stick side.

Anime4000, 2023-03-28:
I see, I guess get an RB5009, it doesn't need fiber to plug to access Telnet/SSH/WebUI

zentavr, 2023-03-28:
Upgraded my RB4011iGS from 6.x to latest 7.8 RouterOS. Now I can see 2.5Gbps mode, but stick still wants a fiber to be plugged in.

rajkosto, 2023-03-28:
the STICK doesnt WANT anything
the STICK is reporting RX_LOS when there is no incoming PON signal, and YOUR MIKROTIK is DISABLING the ethernet on that interface while RX_LOS is active (Loss of Signal) because it thinks there wont be any packets if there is no signal (which is true on simple transcievers without SoC)

zentavr, 2023-03-28:
I was able to reset without having UART.

So:
1. In one terminal I run `ping 192.168.100.100` just to see when the stick comes online. Sometimes I can see 1 ping back, sometimes 4.
2. At the second terminal just in one line: `while true ; do sshpass -p admin-password ssh -oKexAlgorithms=+diffie-hellman-group1-sha1 -oHostKeyAlgorithms=+ssh-rsa -oCiphers=+3des-cbc -oConnectTimeout=1 -oConnectionAttempts=1 admin@192.168.100.100 -- flash_eraseall /dev/mtd3; if [ $? -eq 0 ]; then break; else sleep 1; fi; done`. Yes, long-long line. If you want - you can create a bash script. I have MacOS Ventura 13.2.1. `sshpass` installed by brew.
3. Wait. I had waited maybe 15-20 minutes just to catch 4 pings. In such case it's enough time to make a wipe. When wipe happens - you'll see something like this in the terminal: `Erasing 4 Kibyte @ 3b000 -- 98 % complete.`.
4. Maybe it enough, but I'd repeated the step above again.
5. Again, maybe it's ok just to re-plug the stick into the port/media converted. I made it in a long way... instead of that put another command and waited for 4 next pings: `while true ; do sshpass -p admin-password ssh -oKexAlgorithms=+diffie-hellman-group1-sha1 -oHostKeyAlgorithms=+ssh-rsa -oCiphers=+3des-cbc -oConnectTimeout=1 -oConnectionAttempts=1 admin@192.168.100.100 -- reboot; if [ $? -eq 0 ]; then break; else sleep 1; fi; done`
6. Finally the stick appeared at 192.168.1.1 and I was able to log in.

zentavr, 2023-03-29:
Probably we can reset by curl request as well, but I had not tried that.

The process looks like this:
```shell
# Log In
curl -v -X POST http://192.168.1.1/boaform/admin/formLogin --data-raw 'challenge=&username=admin&password=admin&save=Login&submit-url=%2Fadmin%2Flogin.asp'
# Reset
curl -v --http0.9 -X POST http://192.168.1.1/boaform/formSaveConfig --data-raw 'reset=Reset&submit-url=%2Fsaveconf.asp'
```

How you can see - no tokens/etc are required to make a second request (WTF? Security???). It's likely web server just whitelists your IP and accepts all further admin request without authentication.

rajkosto, 2023-03-29:
the only time i managed to "brick" my stick was by turning off "VLAN Aware Bridge" in the HGU webgui... after that it never responded to pings on the LAN IP again until i did the eraseall via uart.
For your method to work you must also know the LAN IP, and it must be working, which isnt the case when people want to factory reset.

zentavr, 2023-03-29:
@Anime4000 if you are OK - I can set up a pull request with the instructions which I'd mentioned here with the program method of how to reset the stick.

crrodriguez, 2023-03-29:
I believe the stick has tftp recover parameters, but unfortunately I wasn't able to make it work. it is supposedly programmed to ask tftp server at 192.168.1.7 for firmware, but it does not do so at least automatically. maybe there is a prompt for it at the serial console but I do not have it attached.

zentavr, 2023-03-29:
@crrodriguez in theory we can run firmware in qemu and test that.

Anime4000, 2023-03-30:
> @Anime4000 if you are OK - I can set up a pull request with the instructions which I'd mentioned here with the program method of how to reset the stick.

sure! no problem, improvement, documentation is very welcome

zentavr, 2023-03-30:
@Anime4000 here we go: #143