Anime4000/RTL960x #86: Incompatibility between Mikrotik CSS610-8G-2S+IN and V2801F V-SOL

nazar-pc, 2022-11-01:
Hi folks, my switch doesn't seem to recognize SFP module at all.
When I insert module into running switch, switch doesn't seem to recognize it at all.
If I reboot switch with module inserted, it prints 0x03 character (ASCII) as vendor and doesn't recognize it as anything useful otherwise.

Does this mean switch lacks drivers or did anyone experience anything similar?

Anime4000, 2022-11-01:
the switch is latest firmware?
have you tried Disable Auto Negotiate ?

nazar-pc, 2022-11-01:
Switch was at latest firmware SwOS Lite 2.14.
Auto-negotiation was enabled, Mikrotik support also suggested to disable and fix to 1G and indeed, module seems to be recognized now, will look into how to get it to actually work now, thanks!

closed 2022-11-01 by nazar-pc

nazar-pc, 2022-11-01:
If I can ask another question here: how do I get to web interface?
I have isolated SFP and one Ethernet into separate group, assigned 192.168.1.2 to the machine, but 192.192.1.1 is unreachable and SwOS doesn't see any MAC addresses on that interface either. Looks like there is another trick I'm missing.

Anime4000, 2022-11-30:
I see, for now I put CSS610-8G-2S+IN in incompatible list

nazar-pc, 2022-12-30:
I think in the end the issue is not the switch, it is OLT frm Raisecom that rejects non-Raisecom ONU and because of this ONU's Web UI doesn't even start.

Anime4000, 2022-12-30:
that troublesome, with extended OMCI check can't make PON Stick to work

nazar-pc, 2025-07-23:
Finally had success with another ONU stick on CSS610-8G-2S+IN, so the switch as such seems to be fine