Anime4000/RTL960x #87: XPON STICK ONU SFP Info (Temp, Rx power, Tx Power) can not be read by 3rd Party Devices

ToTheCLI, 2022-11-03:
The ODI XPON STICK ONU SFP Info (Temp, Rx power, Tx Power) can not be read by 3rd Party Devices but can be seen in the WebGUI, because these values are not in EEPROM(not exposed on the i2c bus) other devices can not read and display these values
[image]

Is there a way to modify firmware to make them viewable or any way to contact ODI?

ToTheCLI, 2022-11-03:
[image]
WebGUI

TheKhack, 2022-11-03:
http://www.odi-tec.com/lx_17.html

Anime4000, 2022-11-04:
ODI using Emulated, for me I don't bother about that, since it act like Ethernet SFP

rndm2, 2022-11-27:
> http://www.odi-tec.com/lx_17.html

Sent them a message :) Will see...