Anime4000/RTL960x #88: CarlitoxxPro CPGOS03-0490 v2.0

wirszews, 2022-11-14:
Hi,
Will this soft fit CarlitoxxPro CPGOS03-0490 v2.0?
Like uploading V2801F?  

thx for info.

Anime4000, 2022-11-15:
I believe it uses RTL9601CI, same hardware as V2801F
you may use V2801F firmware on that

Anime4000, 2022-11-19:
try open `192.168.1.1/upgrade.asp` to update firmware

wirszews, 2022-11-19:
The same situation
It reaches 100%, the stick resets and the original firmware (CarlitoxxPro) is loaded

crrodriguez, 2022-12-01:
I was successful doing so

flash set LAN_SDS_MODE 1
flashed https://github.com/Anime4000/RTL960x/raw/main/Firmware/V2801F/V2801F_V1.9.0-220425.tar
Was welcomed by @Anime4000  modded interface
I have, however not tested with the fiber plugged, however as far as I understand it should work...