Anime4000/RTL960x #90: SFU vs. IGD Mode

S8T8, 2022-11-15:
Hello,
the ISP provided an Huawei HG8010H ONT that I'm trying to replace with an ODI stick, the router (branded but unlocked) is connecting via PPPoE using the ISP VLAN for internet (other VLANs are provided for VoIP, IPTV etc. but not used),

After many day of tests I finally got an O5 status almost with any recent firmware version, but only using IGD version of the firmware the router is able to connect.
Based on my understanding of the excellent description, the lightweight SFU mode seems to be more indicated, there is an easy explanation for why only the IGD version with all the router functions done by the router is fitting in my case?

Also, I noticed that after the first set-up of the hybrid version, I have to delete the WAN Bridge
[image]
I'm missing something?

Thanks

Anime4000, 2022-11-15:
if you using SFU type, it bridge everything what OLT map for, you need check 2 command in telnet:

1. VLAN Filter (PON to ETH Mapping)
`omcicli mib get 84`

2. Extended VLAN Filter, useful for some OLT provision VLAN A at PON side and VLAN B at LAN side (any to any)
`omcicli mib get 171`

Anime4000, 2022-11-16:
> Your explanation " it bridge everything what OLT map" to me seems what I try to achieve using IGD type and removing the WAN Bridge.

OMCI can be weird sometime, it depend on how OLT is configured, many ISP just map 1:1 VLAN, for example PON and LAN side VLAN is same value.

I not sure why ISP doing VLAN value differently from PON and LAN,

Also, some ISP doing double VLAN, this can be more confusing, such setup can many VLAN within VLAN aslong GPON MTU not reach 2000 bytes

S8T8, 2022-11-29:
An update on this;
ISP is providing just internet (not VoIP or IPTV with home subscription), with the contract an Huawei HG8010H is assigned, router needs to use VLAN 835 for PPPoE connection.

Using ODI DFP-34X-2C2 with IGD firmware I'm able to connect using VLAN 835 on router after I delete the WAN Bridge (as explained in the first post).

Switching to SFU firmware the router is not able to connect anymore using VLAN 835 but with the VLAN 3310 found in the `Treatment Inner   : PRI 0,VID 3310, TPID 2` applied either to the SFP or router.
Now, VLAN 835 was used for what? And which firmware is better to use in this case?

@Anime4000 two more info,
- any instruction on how to use Scripts on the firmware? [https://github.com/Anime4000/RTL960x/tree/main/Tools/etc/scripts](url) 
- in the latest 2209XX firmware there is an option to Enable or Disable "Fiber Reset", someone know what is it?

THANKS

Anime4000, 2022-11-30:
1. can you show us  `omcicli mib get 84` on SFU and IGD?
2. that I not really sure, no docs about it

S8T8, 2022-11-30:
1) `omcicli mib get 84` updated on the previous post: [https://github.com/Anime4000/RTL960x/issues/90#issuecomment-1315707026](url) is the same for SFU and IGD.
Could be retaled to [https://github.com/Anime4000/RTL960x/issues/26](url) ?
2) What about the Tools/Scripts, fix.speed for example [https://github.com/Anime4000/RTL960x/blob/main/Tools/etc/scripts/fix_speed.sh](url)

skon77, 2022-11-30:
@Anime4000 ,
If we're talking about scripts...
I want to script SFP reboot automatically when Mikrotik router has been rebooted. This solution will allow me not to reconnect the SFP stick manually.
I suppose to do it by ping IP address of Mikrotik (192.168.1.2). In the absence of a scheduler (cron) can be done through an infinite loop. With a function call every 3-5 minutes.

```
while true; do
SFP_REBOOT=$(ping -c 1 -W 1 -q 192.168.1.2 | grep -c '1 packets received')

    if [[ $SFP_REBOOT -eq 0 ]] ; then
      reboot -d 30
    fi

  sleep 300
done
```

I will implement this script in FW and autostart via RC35. Is that a good idea?