Anime4000/RTL960x #91: Which version to use when ISP provided router is Hybrid

rajkosto, 2022-11-15:
They provide me with a Nokia wifi5 router that is operating in Hybrid mode (saw in ritool)
Fake O5 is given unless SN/Device/SwRev/HwRev match.
Their router has 2 WAN connections set up in Route mode, and they cannot be set to bridge mode, this is on VLAN 100 for PPPoE and 600 for TR069. I will need the ODI stick to give me the VLAN 100 somehow (tagged)
It also has 2 of it's ports set to BRIDGED mode separately, and this cannot be controlled what they are bridged to, this is done via OMCI on AUTH. This is for IPTV, and it never gives me a VLAN to see (traffic on the ports themselves is untagged when they come out as ethernet). I will also need this to work.

So which firmware do i use and how do I set up the VLANs ?

Anime4000, 2022-11-16:
some OLT map VLAN differ than LAN side (example: PON VLAN 209 = LAN VLAN 400), using ODI stick will bridge what PON interface got and ignore VLAN Mapping for LAN side

rajkosto, 2022-11-16:
Is this what the fix_vlan_tag.sh script does ? Is it better to use the hybrid firmware for this case then, so the "correct" vlan ids are output instead of the unchanged ones ?

Anime4000, 2022-11-17:
fix_vlan_tag.sh do what OMCI sends,
you need try both firmware at newer one

rajkosto, 2022-11-18:
i have indeed confirmed that the PON-side VLAN tag is the same for MULTIPLE flows/queues that I have, so transparent mode is not enough
my ONU is true Hybrid mode and ISP uses both modes via OMCI:
Internal HGU VLAN map for running pppoe and dhcp and sip clients:
PON VLAN 2713 -> HGU VLAN 100 for PPPoE with 500mbps flow/queue limit
PON VLAN 9 -> HGU VLAN 600 for TR069 DHCP

LAN1 port is set to HGU mode, so thats where the ONT DHCP server is accessible, routing, etc
the other ports, LAN2-LAN4 are set to SFU mode, so they are available for configuration via OMCI, and are done like this:
LAN2 -> bind to PON VLAN 1313 with 10mbps flow/queue limit
LAN3 -> bind to PON VLAN 1313 with 10mbps flow/queue limit (but this is independent of previous queue, so its 10+10mbps!)
LAN4 -> bind to PON VLAN 2713 with 500mbps flow/queue limit (again, independent of the previous 2713 binding, so its 500+500mbps total!)

since DFP in transparent SFU mode ONLY exposes PON VLAN ids, it means there are DOUBLE ds/us flows on the 2713 and 1313 VLANs! But i can only use one ! so i can only get 10mbps on 1313 and 500mbps on 2713 while on original ONU i could get 10+10 (two iptv STBs supported), 500+500 with two pppoe clients (one internal in the router, one via LAN4). It also creates some weird things like much more traffic received when using only one stb than normal (maybe packets are being doubled before being sent from stick to my router?).

i must figure out a way to get the DFP to actually map the VLANs according to 
1. HGU map (so the first PPPoE vlan appears as VID 100, instead of 2713)
2. Port map (so the 1313,1313,2713 vlans appear as some "fake" vlans, like 1000 + portNum, so the independent traffic limits/queues/flows can be utilized at the same time)

I have omci logs, dumps, and various other root ssh executed logs from my original ONT, if anyone wants to check them in private.