Anime4000/RTL960x #92: issues with DFP-34X-2C2 and Intel I210IS ?

panachoi, 2022-11-16:
I have a router (PCEngines APU6, [https://www.pcengines.ch/apu6b4.htm](url) where I'm unable to reach the stick when plugged into the SFP port. It would appear that the SFP is detected correctly, as I can see it with ethtool:

```
root@apu6:~# ethtool enp1s0                                                     
Settings for enp1s0:                                                            
        Supported ports: [ FIBRE ]                                              
        Supported link modes:   1000baseKX/Full                                 
        Supported pause frame use: Symmetric                                    
        Supports auto-negotiation: Yes                                          
        Supported FEC modes: Not reported                                       
        Advertised link modes:  1000baseKX/Full                                 
        Advertised pause frame use: Symmetric                                   
        Advertised auto-negotiation: Yes                                        
        Advertised FEC modes: Not reported                                      
        Speed: Unknown!                                                         
        Duplex: Unknown! (255)                                                  
        Auto-negotiation: on                                                    
        Port: FIBRE                                                             
        PHYAD: 0                                                                
        Transceiver: internal                                                   
        Supports Wake-on: pumbg                                                 
        Wake-on: g                                                              
        Current message level: 0x00000007 (7)                                   
                               drv probe link                                   
        Link detected: no                                                      

```
```
root@apu6:~# ethtool --module-info enp1s0                                       
        Identifier                                : 0x03 (SFP)                  
        Extended identifier                       : 0x04 (GBIC/SFP defined by 2)
        Connector                                 : 0x01 (SC)                   
        Transceiver codes                         : 0x00 0x00 0x00 0x02 0x22 0x0
        Transceiver type                          : Ethernet: 1000BASE-LX       
        Transceiver type                          : FC: intermediate distance ()
        Transceiver type                          : FC: Longwave laser (LC)     
        Transceiver type                          : FC: Single Mode (SM)        
        Encoding                                  : 0x01 (8B/10B)               
        BR, Nominal                               : 1300MBd                     
        Rate identifier                           : 0x00 (unspecified)          
        Length (SMF,km)                           : 20km                        
        Length (SMF)                              : 20000m                      
        Length (50um)                             : 0m                          
        Length (62.5um)                           : 0m                          
        Length (Copper)                           : 0m                          
        Length (OM3)                              : 0m                          
        Laser wavelength                          : 1310nm                      
        Vendor name                               : ODI                         
        Vendor OUI                                : 00:00:00                    
        Vendor PN                                 : DFP-34X-2C2                 
        Vendor rev                                :                             
        Option values                             : 0x00 0x1a                   
        Option                                    : RX_LOS implemented          
        Option                                    : TX_FAULT implemented        
        Option                                    : TX_DISABLE implemented      
        BR margin, max                            : 0%                          
        BR margin, min                            : 0%                          
        Vendor SN                                 : XPON22100654                
        Date code                                 : 221009                    
```

Configuring an ip address on the interface does not change the behavior, the admin interface at 192.168.1.1 is not reachable with any protocol.

Plugging the SFP back into a SFP port on a switch works without issue, and then I can reach the admin interface

I've noticed that contrary to (or perhaps I've misread?) the documentation, LAN_SDS_MODE is set to 0, and not 3. I'm hesistant to change this, as I would have no easy way to recover if I get locked out.

Anime4000, 2022-11-16:
ODI DFP-34X-2C2 has auto nego SFP, try leave it around 5 to 8 minute....

mine is take around 3 minute for DFP-34X-2C2 with RB5009 (<ROS 7.6) fully negotiate, ODI firmware will keep trying...

also, I think it's possible to change mode from `1000baseKX/Full` to `1000baseX/Full`, that what RTL906x understands

panachoi, 2022-11-16:
> ODI DFP-34X-2C2 has auto nego SFP, try leave it around 5 to 8 minute....
> mine is take around 3 minute for DFP-34X-2C2 with RB5009 (<ROS 7.6) fully negotiate, ODI firmware will keep trying...

This is not the issue. I can leave it for hours and am still not able to connect. Again, plugging it into a switch makes in available again, usually within a couple of minutes.

> also, I think it's possible to change mode from `1000baseKX/Full` to `1000baseX/Full`, that what RTL906x understands

By changing LAN_SDS_MODE ?

skon77, 2022-11-16:
> mine is take around 3 minute for DFP-34X-2C2 with RB5009 (<ROS 7.6) fully negotiate, ODI firmware will keep trying...

Does DFP-34X-2C2 work successfully with the RB5009 (ROS 7.7b) at 2.5Gbps?
If set LAN_SDS_MODE=4 the connection with stick will not be lost?

Anime4000, 2022-11-17:
> > mine is take around 3 minute for DFP-34X-2C2 with RB5009 (<ROS 7.6) fully negotiate, ODI firmware will keep trying...
> 
> Does DFP-34X-2C2 work successfully with RB5009 (ROS 7.7b) at 2.5Gbps? If set LAN_SDS_MODE=4 the connection with stick will not be lost?

7.7b4 do support 2.5Gb ODI stick no problem, and i set LAN_SDS_MODE 4,
ODI do have SFP auto nego, setting LAN_SDS_MODE will not lock you out