Anime4000/RTL960x #93: Strange permanent incoming traffic from DFP-34X-2C2

r00tGER, 2022-11-16:
Internet works great, but I see permanent incoming traffic on SFP interface from ONU Stick, ~3...8Mbps
[image]
It's not a download traffic from LAN 

Some samples from ONU Stick (via SSH connection):
```
# ifconfig -a
br0       Link encap:Ethernet  HWaddr 38:3A:21:27:E4:60
          inet addr:192.168.1.1  Bcast:192.168.1.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:5305 errors:0 dropped:0 overruns:0 frame:0
          TX packets:4087 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:869079 (848.7 KiB)  TX bytes:3531796 (3.3 MiB)

eth0      Link encap:Ethernet  HWaddr 38:3A:21:27:E4:60
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:5305 errors:0 dropped:0 overruns:0 frame:0
          TX packets:43263 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:869079 (848.7 KiB)  TX bytes:5412178 (5.1 MiB)
          Interrupt:26 Base address:0x2000
```
```
# flash get OMCI_OLT_MODE
OMCI_OLT_MODE=0 (Default Mode)
# flash get VLAN_CFG_TYPE
VLAN_CFG_TYPE=1 (Manual)
# flash get VLAN_MANU_MODE
VLAN_MANU_MODE=1 (Tagging Mode | VLAN Stacking / Q-in-Q)
# flash get DIRECT_BRIDGE_MODE
DIRECT_BRIDGE_MODE=1
# flash get PON_MODE
PON_MODE=1 (GPON)
# flash get PON_VENDOR_ID
PON_VENDOR_ID=OEM
# flash get FIBER_MODE
FIBER_MODE=0
# flash get DEVICE_TYPE
DEVICE_TYPE=0 (bridge)
# flash get OMCI_VEIP_SLOT_ID
OMCI_VEIP_SLOT_ID=255
```

Some screenshots from WEB UI:
[image]
[image]
[image]
[image]

Anime4000, 2022-11-16:
it could be IPTV broadcast?

skon77, 2022-11-16:
It looks like broadcast IPTV traffic.
Is your provider MGTS?
Welcome to [ru-board](https://forum.ru-board.com/topic.cgi?forum=8&topic=80480&start=1700) forum.

rajkosto, 2022-11-27:
if you dont do vlan filtering you will see multicast IPTV traffic from everyone on your node (on the multicast vlan id)... mine totaled up to like 90mbps ! but with vlan filtering and/or igmp snooping it doesnt get passed to other interfaces unless needed

Anime4000, 2022-11-27:
what I understand IPTV traffic Multi-cast, is like TV Antenna broadcast but over wire right?

skon77, 2022-11-28:
> what I understand IPTV traffic Multi-cast, is like TV Antenna broadcast but over wire right?

But in general, if ISP hardware is properly configured, multicast traffic is not permanent. It must appear on request from the client. And of course, if now your neighbor has requested multicast, it will appear for you as well.

longthanhtran, 2023-11-10:
looks like this case same as #51 and my choice was to [remove](https://github.com/Anime4000/RTL960x/issues/51#issuecomment-1803704448) the downstream traffic

chitz852, 2024-03-14:
> looks like this case same as #51 and my choice was to [remove](https://github.com/Anime4000/RTL960x/issues/51#issuecomment-1803704448) the downstream traffic

Is there a way to remove it with the config,  to survive a restart ? In my case for example is default gem port 4095 https://www.cisco.com/c/en/us/td/docs/switches/lan/catalyst_pon/software/configuration_guide/mcast/b-gpon-config-multicast/configuring_multicast_in_gpon_network.html

```
# diag gpon show ds-flow
gpon show ds-flow
============================================================
     GPON ONU MAC D/S Flow Status
Flow ID | GEM Port | Type | Multicast | AES
      0 |     4095 |  ETH |           |
      1 |      355 |  ETH |           |   *
      2 |     1379 |  ETH |           |   *
      3 |     1507 |  ETH |           |   *
      4 |     1251 |  ETH |           |   *
      5 |     1123 |  ETH |           |   *
      6 |      995 |  ETH |           |   *
      7 |      867 |  ETH |           |   *
      8 |      483 |  ETH |           |   *
      9 |      611 |  ETH |           |   *
     10 |      739 |  ETH |           |   *
     64 |       99 | OMCI |           |
============================================================
```