Anime4000/RTL960x #96: UF-Instant WEB interface crashes when attempting to login

FiberExplorer, 2022-11-21:
I have modded the UF-Instant firmware "4.3.1 or 4.4.2 doesn't change the outcome" but when I try log in the WebGui 
crash and tell connection refused and no pages are available.

Could someone give me some tips?

Many Thanks!

Anime4000, 2022-11-22:
is that has been flashed with V2801F ?

FiberExplorer, 2022-11-22:
> is that has been flashed with V2801F ?

I have flashed with this procedure and relative firmware suggested in the guide at https://github.com/stich86/UF-Instant-Mod.
if I need to reflash please so kindly to suggest the appropriate firmware and a procedure.

Manty Thanks

Anime4000, 2022-11-22:
for fast reply and easy help, join this discord: https://discord.gg/KUHwDgkRtR

FiberExplorer, 2022-11-22:
I've joined and I'll try to find a possible solution before post any request.   

Many Thanks!!

closed 2022-11-23 by Anime4000