Anime4000/RTL960x #97: DFP-34X-2C2 PON State O5, but no traffic

rndm2, 2022-11-24:
Hi, 

I am trying to get freshly bought DFP-34X-2C2 working with latest 220414 firmware bad no luck so far. 
I changed all parameters according to this documentation (including ELAN_MAC_ADDR and MAC_KEY) https://github.com/Anime4000/RTL960x/blob/main/Docs/FLASH_GETSET_INFO.md

Here is my list of current parameters: 
```
# flash all 
<Config Name="ROOT">
 <Dir Name="HW_MIB_TABLE">
  <Value Name="SUPER_NAME" Value="adsl"/>
  <Value Name="SUPER_PASSWORD" Value="realtek"/>
  <Value Name="TEST_MAC_ADDR" Value="realtek"/>
  <Value Name="ELAN_MAC_ADDR" Value="5876ac105330"/>
  <Value Name="WLAN_MAC_ADDR" Value="00e04c076802"/>
  <Value Name="MAC_KEY" Value="5aee44cf1d71e0a64b0d52483b68ef0d"/>
  <Value Name="WAN_PHY_PORT" Value="4"/>
  <Value Name="LASER_POLARITY_TYPE" Value="0"/>
  <Value Name="PON_DETECT_ENABLE" Value="1"/>
  <Value Name="HW_RF_TYPE" Value="7"/>
  <Value Name="HW_WLAN0_TX_POWER_CCK_A" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_CCK_B" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_HT40_1S_A" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_HT40_1S_B" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_DIFF_HT40_2S" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_DIFF_HT20" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_DIFF_OFDM" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_11N_TSSI1" Value="0"/>
  <Value Name="HW_WLAN0_11N_TSSI2" Value="0"/>
  <Value Name="HW_WLAN0_11N_THER" Value="0"/>
  <Value Name="HW_WLAN0_11N_THER_2" Value="0"/>
  <Value Name="HW_WLAN0_11N_PA_TYPE" Value="0"/>
  <Value Name="HW_WLAN0_REG_DOMAIN" Value="1"/>
  <Value Name="HW_WLAN0_11N_TRSWITCH" Value="0"/>
  <Value Name="HW_WLAN0_11N_TRSWPAPE_C9" Value="0"/>
  <Value Name="HW_WLAN0_11N_TRSWPAPE_CC" Value="0"/>
  <Value Name="HW_ANT_DIVERSITY" Value="0"/>
  <Value Name="HW_TX_ANT" Value="0"/>
  <Value Name="HW_CS_THRESHOLD" Value="11"/>
  <Value Name="HW_CCA_MODE" Value="1"/>
  <Value Name="HW_PHY_TYPE" Value="0"/>
  <Value Name="HW_WLAN_LED_TYPE" Value="7"/>
  <Value Name="BYTE" Value="255"/>
  <Value Name="WORD" Value="65535"/>
  <Value Name="DWORD" Value="4294967295"/>
  <Value Name="INT1" Value="2147483647"/>
  <Value Name="INT2" Value="-2147483648"/>
  <Value Name="WIFI_SUPPORT" Value="0"/>
  <Value Name="HW_SERIAL_NO" Value="XPON22100613"/>
  <Value Name="HW_FON_KEYWORD" Value="63,62,68,4a,75,3c,65,cd,7b,75,e5,2d,a9,80,72,8c,97,fc,bd,9b,37,51,d9,e3,82,92,68,a1,07,ae,a0,b5"/>
  <Value Name="HW_WLAN0_11N_XCAP" Value="0"/>
  <Value Name="HW_WLAN0_11N_XCAP_2" Value="0"/>
  <Value Name="HW_WLAN0_11N_TARGET_PWR" Value="0"/>
  <Value Name="HW_WLAN0_COUNTRYCODE" Value="1"/>
  <Value Name="HW_WLAN1_COUNTRYCODE" Value="1"/>
  <Value Name="HW_WLAN0_COUNTRYSTR" Value="US"/>
  <Value Name="HW_WLAN1_COUNTRYSTR" Value="US"/>
  <Value Name="HW_CWMP_MANUFACTURER" Value="ODI"/>
  <Value Name="HW_CWMP_PRODUCTCLASS" Value="DFP-34X-2C2"/>
  <Value Name="HW_HWVER" Value="V2.0"/>
  <Value Name="GPON_ONU_MODEL" Value="FG824CD"/>
  <Value Name="LAN_SDS_MODE" Value="0"/>
  <Value Name="PON_MODE" Value="1"/>
  <Value Name="PON_LED_SPEC" Value="0"/>
  <Value Name="OUI" Value="11,11,11"/>
  <Value Name="PORT_REMAPPING" Value="00,01,00,00"/>
  <Value Name="PON_VENDOR_ID" Value="SCOM"/>
  <Value Name="GPON_SN" Value="SCOM300FBFC5"/>
  <Value Name="EPON_ONU_MODEL" Value="EPON"/>
  <Value Name="EPON_EXTONU_MODEL" Value="DFP-34X-2C2"/>
[... 141 lines cut ...]
  <Value Name="VCI" Value="43"/>
 </Dir>
 <Dir Name="AUTO_PVC_SEARCH_TBL"> <!--index=6-->
  <Value Name="VPI" Value="8"/>
  <Value Name="VCI" Value="51"/>
 </Dir>
 <Dir Name="AUTO_PVC_SEARCH_TBL"> <!--index=7-->
  <Value Name="VPI" Value="8"/>
  <Value Name="VCI" Value="59"/>
 </Dir>
 <Dir Name="PPPOE_SESSION_TBL">
 </Dir>
 <Dir Name="RIP_TBL">
 </Dir>
 <Dir Name="DHCP_RESERVED_IPADDR_TBL">
 </Dir>
 <Dir Name="EPON_LLID_TBL">
 </Dir>
</Config>
# flash all
```

My initial device is Sercomm FG824CD which have `OLT Type: Nokia`. Not sure exactly what it means and what parameter I have to change from above. 

Any ideas?

Anime4000, 2022-11-25:
try upgrade to latest firmware and change additional parameter like OUI

rndm2, 2022-11-25:
Upgraded today to latest. 
What value should I put to OUI? 

I also think I would try to set OMCI_OLT_MODE=21 and set software version.

rndm2, 2022-11-25:
[Screenshot](http://random.rediron.net.ua/9694477a95ce993d2e7960fb041a29bc.Screenshot 2022-11-25 at 17.46.07.png)

Does this mean that traffic goes? I don't understand why there are no sent packets.

Mhrok, 2022-11-25:
I suppose there are VLANs on the other side of ONT, you need to send something tagged to connect to anything.
https://github.com/Anime4000/RTL960x/blob/main/Docs/OMCI_VLAN.md

rndm2, 2022-11-25:
> I suppose there are VLANs on the other side of ONT, you need to send something tagged to connect to anything.
> https://github.com/Anime4000/RTL960x/blob/main/Docs/OMCI_VLAN.md

It is Vodafone NEBA Spain. 
They use VLAN 24 for PPPoE. 

But I feel you are talking about other kind of VLANs. Thanks. Will try now.

Anime4000, 2022-11-26:
you need untag the VLAN at Router side

rollingafull, 2022-11-26:
I've configured Lowi (Vodafone) in Spain.
After configure the stick the vlan ID is not the same, you must configure it on the router, i use an RF5900 and this is my config.

1º Configure a new  VLAN interface with vlanID 57 and set it to sfp+ interface.
2º Configure a pppoe connection against vlan interface.

After this you must entablish a connection if not try with the other vlanID.

I only have internet service, if you have phone or tv service, must configure the other vlanIDs for it.

rndm2, 2022-11-26:
Hah. 
I was sure on a router I need VLAN ID 24 as in Vodafone router. 
I even started doubt in my network knowledge after this.

I created VLAN interface with id 57 and then created PPPoE against it. No traffic :( 
Also, with `diag l2-table get entry address` I got other VLAN ids 3, 6, 9, 45, 180, 182, 4005 I tried all of them. No PPPoE 🙈 
Strange thing that with this method I didn't get ID 57. 

@rollingafull could you please show me your `flash all` from your `DFP-34X-2C2`. I think I might be broke something during experiments. 
Do you have some specific settings for you Mikrotik? Like non-default MTU for SFP/VLAN

rndm2, 2022-11-26:
Success!!! 
Finally I reset my stick to factory defaults and flashed back 220414 firmware and it works now! 

One last thing. I need to acquire PPPoE password from Vodafone :)

Mhrok, 2022-11-26:
I have no experience with Vodafone, but my provider has PPPoE login and password specified in the documents I signed. Different providers in my area give the creds on demand.

rndm2, 2022-11-26:
@Mhrok my mistake was in modifying additional parameters. Some of them stopped traffic. 
I wasn't aware from the beginning that I need to use VLAN ID 57 (not 24 as in Vodafone router which apparently encrypt not only passwords but VLAN ids as well). 
And I tried to do more changes on stick at the time when it already worked. 
 
In the end, on stick I need only set `GPON_PLOAM_PASSWD` and `GPON_SN `. That's it.

closed 2022-11-26 by rndm2